aircrack-ng/rtl8812au #382: Compilation error on 5.6.4.1

terencode, 2019-06-22:
Using GCC 9.1:

```
DKMS make.log for rtl88xxau-r712.472b8b3 for kernel 5.1.12-arch1-1-ARCH (x86_64)
sam. juin 22 19:11:25 CEST 2019
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.1.12-arch1-1-ARCH/build M=/var/lib/dkms/rtl88xxau/r712.472b8b3/build  modules
make[1]: Entering directory '/usr/lib/modules/5.1.12-arch1-1-ARCH/build'
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/os_dep/linux/rtw_cfgvendor.o
/var/lib/dkms/rtl88xxau/r712.472b8b3/build/os_dep/osdep_service.c: In function ‘isFileReadable’:
/var/lib/dkms/rtl88xxau/r712.472b8b3/build/os_dep/osdep_service.c:2204:10: error: implicit declaration of function ‘get_ds’; did you mean ‘get_da’? [-Werror=implicit-function-declaration]
 2204 |   set_fs(get_ds());
      |          ^~~~~~
      |          get_da
/var/lib/dkms/rtl88xxau/r712.472b8b3/build/os_dep/osdep_service.c:2204:10: error: incompatible type for argument 1 of ‘set_fs’
 2204 |   set_fs(get_ds());
      |          ^~~~~~~~
      |          |
      |          int
In file included from ./include/linux/uaccess.h:11,
                 from ./include/linux/crypto.h:26,
                 from ./include/crypto/hash.h:16,
                 from ./include/linux/uio.h:14,
                 from ./include/linux/socket.h:8,
                 from ./include/linux/compat.h:15,
                 from ./include/linux/ethtool.h:17,
                 from ./include/linux/netdevice.h:41,
                 from /var/lib/dkms/rtl88xxau/r712.472b8b3/build/include/osdep_service_linux.h:30,
                 from /var/lib/dkms/rtl88xxau/r712.472b8b3/build/include/osdep_service.h:50,
                 from /var/lib/dkms/rtl88xxau/r712.472b8b3/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl88xxau/r712.472b8b3/build/os_dep/osdep_service.c:19:
./arch/x86/include/asm/uaccess.h:29:40: note: expected ‘mm_segment_t’ {aka ‘struct <anonymous>’} but argument is of type ‘int’
   29 | static inline void set_fs(mm_segment_t fs)
      |                           ~~~~~~~~~~~~~^~
/var/lib/dkms/rtl88xxau/r712.472b8b3/build/os_dep/osdep_service.c: In function ‘retriveFromFile’:
/var/lib/dkms/rtl88xxau/r712.472b8b3/build/os_dep/osdep_service.c:2242:11: error: incompatible type for argument 1 of ‘set_fs’
 2242 |    set_fs(get_ds());
      |           ^~~~~~~~
      |           |
      |           int
In file included from ./include/linux/uaccess.h:11,
                 from ./include/linux/crypto.h:26,
                 from ./include/crypto/hash.h:16,
                 from ./include/linux/uio.h:14,
                 from ./include/linux/socket.h:8,
                 from ./include/linux/compat.h:15,
                 from ./include/linux/ethtool.h:17,
                 from ./include/linux/netdevice.h:41,
                 from /var/lib/dkms/rtl88xxau/r712.472b8b3/build/include/osdep_service_linux.h:30,
                 from /var/lib/dkms/rtl88xxau/r712.472b8b3/build/include/osdep_service.h:50,
                 from /var/lib/dkms/rtl88xxau/r712.472b8b3/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl88xxau/r712.472b8b3/build/os_dep/osdep_service.c:19:
./arch/x86/include/asm/uaccess.h:29:40: note: expected ‘mm_segment_t’ {aka ‘struct <anonymous>’} but argument is of type ‘int’
   29 | static inline void set_fs(mm_segment_t fs)
      |                           ~~~~~~~~~~~~~^~
/var/lib/dkms/rtl88xxau/r712.472b8b3/build/os_dep/osdep_service.c: In function ‘storeToFile’:
/var/lib/dkms/rtl88xxau/r712.472b8b3/build/os_dep/osdep_service.c:2277:11: error: incompatible type for argument 1 of ‘set_fs’
 2277 |    set_fs(get_ds());
      |           ^~~~~~~~
      |           |
      |           int
In file included from ./include/linux/uaccess.h:11,
                 from ./include/linux/crypto.h:26,
                 from ./include/crypto/hash.h:16,
                 from ./include/linux/uio.h:14,
                 from ./include/linux/socket.h:8,
                 from ./include/linux/compat.h:15,
                 from ./include/linux/ethtool.h:17,
                 from ./include/linux/netdevice.h:41,
                 from /var/lib/dkms/rtl88xxau/r712.472b8b3/build/include/osdep_service_linux.h:30,
                 from /var/lib/dkms/rtl88xxau/r712.472b8b3/build/include/osdep_service.h:50,
                 from /var/lib/dkms/rtl88xxau/r712.472b8b3/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl88xxau/r712.472b8b3/build/os_dep/osdep_service.c:19:
./arch/x86/include/asm/uaccess.h:29:40: note: expected ‘mm_segment_t’ {aka ‘struct <anonymous>’} but argument is of type ‘int’
   29 | static inline void set_fs(mm_segment_t fs)
      |                           ~~~~~~~~~~~~~^~
  CC [M]  /var/lib/dkms/rtl88xxau/r712.472b8b3/build/os_dep/linux/wifi_regd.o
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:276: /var/lib/dkms/rtl88xxau/r712.472b8b3/build/os_dep/osdep_service.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1571: _module_/var/lib/dkms/rtl88xxau/r712.472b8b3/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.1.12-arch1-1-ARCH/build'
make: *** [Makefile:2343: modules] Error 2

```

kimocoder, 2019-06-22:
A kernel support patch is missing. Don't have more time this evening

kimocoder, 2019-06-22:
@terencode "git pull && make -j4 && make install && modprobe -r 88XXau && modprobe 88XXau"

Couldn't let you out to dry :+1:

terencode, 2019-06-22:
@kimcoder you rock, thanks!

closed 2019-06-22 by terencode