aircrack-ng/rtl8812au #383: CPU warning on 5.6.4.1

terencode, 2019-06-24:
Getting the following on boot with Linux 5.1.14:

```
WARNING: CPU: 4 PID: 1139 at net/wireless/nl80211.c:15732 cfg80211_ch_switch_notify+0xf0/0x100 [cfg80211]
Modules linked in: fuse mousedev joydev input_leds ccm algif_aead des_generic arc4 algif_skcipher 88XXau(O) cmac kvm_amd md4 kvm algif_hash hid_generic irqbypass zram cfg80211 snd_usb_audio snd_hda_codec_realtek af_alg snd_hda_code>
 i2c_core
CPU: 4 PID: 1139 Comm: RTW_CMD_THREAD Tainted: G           O      5.1.14-23-tkg-bmq #1
Hardware name: Gigabyte Technology Co., Ltd. X470 AORUS ULTRA GAMING/X470 AORUS ULTRA GAMING-CF, BIOS F31 05/06/2019
RIP: 0010:cfg80211_ch_switch_notify+0xf0/0x100 [cfg80211]
Code: 89 f6 e8 83 1c fd e0 48 83 3b 00 75 e4 e9 56 ff ff ff 48 8b 85 f0 00 00 00 48 85 c0 74 0c 49 8b 14 24 48 89 50 68 eb 8c 0f 0b <0f> 0b eb 86 66 66 2e 0f 1f 84 00 00 00 00 00 90 41 57 41 56 41 55
RSP: 0018:ffffc9000171bde8 EFLAGS: 00010246
RAX: 0000000000000000 RBX: 000000000000146e RCX: ffff8883f56bf418
RDX: ffff8883f4df6750 RSI: ffffc9000171be20 RDI: ffff88840844e000
RBP: ffff888405e88000 R08: ffff8883f4df60d0 R09: 0000000000000001
R10: ffff888107c07600 R11: 0000000000000008 R12: ffffc9000171be20
R13: ffff8883f56bf000 R14: ffff88840844e000 R15: ffff8883f56bf300
FS:  0000000000000000(0000) GS:ffff88840e700000(0000) knlGS:0000000000000000
CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
CR2: 00007f1fcef53b60 CR3: 00000003e8646000 CR4: 00000000003406e0
Call Trace:
 rtw_cfg80211_ch_switch_notify+0xb0/0xee [88XXau]
 ? rtw_chk_start_clnt_join+0x6e/0x6e [88XXau]
 rtw_chk_start_clnt_join+0x67/0x6e [88XXau]
 join_cmd_hdl+0x252/0x349 [88XXau]
 rtw_cmd_thread+0x290/0x3e8 [88XXau]
 kthread+0x12c/0x160
 ? rtw_stop_cmd_thread+0x34/0x34 [88XXau]
 ? kthread_park+0x80/0x80
 ret_from_fork+0x22/0x40
---[ end trace 0158fbd29bb17a74 ]---
sd 10:0:0:0: [sdc] Attached SCSI disk
hardware_rekey not supported
IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
wlan0: Gained carrier
```

satmandu, 2019-06-26:
I see something similar with 5.6.4.1:
```
[   40.175569] WARNING: CPU: 0 PID: 710 at net/wireless/nl80211.c:16025 cfg80211_ch_switch_notify+0x114/0x120 [cfg80211]
[   40.175579] Modules linked in: zram nls_iso8859_1 hci_uart btqca smsc95xx btrtl usbnet btbcm btintel bluetooth bcm2835_thermal ecdh_generic pwm_bcm2835 ecc uio_pdrv_genirq crct10dif_ce uio leds_gpio sch_fq_codel 88XXau(OE) cfg80211 raspberrypi_hwmon bcm2835_rng ip_tables x_tables autofs4 btrfs zstd_compress raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor xor_neon raid6_pq libcrc32c raid1 raid0 multipath linear f2fs sdhci_iproc sdhci_pltfm i2c_bcm2835 sdhci bcm2835 phy_generic
[   40.175700] CPU: 0 PID: 710 Comm: RTW_CMD_THREAD Tainted: G           OE     5.2.0-050200rc6-generic #201906222033
[   40.175705] Hardware name: raspberrypi rpi/rpi, BIOS 2019.01 02/12/2019
[   40.175714] pstate: 60400005 (nZCv daif +PAN -UAO)
[   40.175857] pc : cfg80211_ch_switch_notify+0x114/0x120 [cfg80211]
[   40.176085] lr : rtw_cfg80211_ch_switch_notify+0xe4/0x134 [88XXau]
[   40.176089] sp : ffff0000125ebcb0
[   40.176094] x29: ffff0000125ebcb0 x28: ffff00000917b430 
[   40.176102] x27: ffff000009089358 x26: 0000000000000003 
[   40.176110] x25: ffff80002ef25300 x24: 0000000000000001 
[   40.176117] x23: ffff8000311f3000 x22: 0000000000000002 
[   40.176125] x21: ffff80002ef25000 x20: ffff0000125ebd40 
[   40.176132] x19: ffff8000294bc400 x18: 0000000000000001 
[   40.176140] x17: 0000000000000000 x16: 0000000000000000 
[   40.176147] x15: ffff0000110268d0 x14: 04056c163d000000 
[   40.176154] x13: 0000000000000000 x12: 0000000000000000 
[   40.176162] x11: 0000000000000006 x10: fffffffffffffffe 
[   40.176169] x9 : 0000000000000008 x8 : 0000000000000006 
[   40.176176] x7 : ffaa006a0105c003 x6 : 0000000000000001 
[   40.176183] x5 : 0000000000000040 x4 : ffff80002ef26f50 
[   40.176191] x3 : 00000000000015a4 x2 : ffff80002ef26b10 
[   40.176199] x1 : 0000159a00000003 x0 : 0000000000000000 
[   40.176207] Call trace:
[   40.176345]  cfg80211_ch_switch_notify+0x114/0x120 [cfg80211]
[   40.176525]  rtw_cfg80211_ch_switch_notify+0xe4/0x134 [88XXau]
[   40.176697]  rtw_chk_start_clnt_join+0x88/0x90 [88XXau]
[   40.176867]  join_cmd_hdl+0x248/0x350 [88XXau]
[   40.177038]  rtw_cmd_thread+0x2a8/0x40c [88XXau]
[   40.177055]  kthread+0x110/0x118
[   40.177065]  ret_from_fork+0x10/0x18
[   40.177071] ---[ end trace e2a60ad239833be4 ]---
```

The interface continues to work though...

terencode, 2019-08-23:
Still persists with 5.6.4.1 master. 
Also, is it normal the mode shows debug information? Shouldn't it be disabled?

terencode, 2019-09-04:
Using latest commits on 5.3 rc7: 
```
WARNING: CPU: 3 PID: 12579 at net/wireless/nl80211.c:16092 cfg80211_ch_switch_notify.cold+0xc/0x13 [cfg80211]
Modules linked in: overlay kvm_amd joydev input_leds mousedev hid_generic 88XXau(OE) uas usb_storage usbhid hid fuse lz4 lz4_compress cfg80211 rfkill ccm algif_aead des_generic arc4 libarc4 kvm snd_hda_codec_realtek snd_hda_codec_g>
 sysfillrect sysimgblt fb_sys_fops drm agpgart
CPU: 3 PID: 12579 Comm: RTW_CMD_THREAD Tainted: G        W  OE     5.3.0-13.rc7-tkg-pds #1
Hardware name: System manufacturer System Product Name/CROSSHAIR VI HERO, BIOS 0002+ 08/23/2019
RIP: 0010:cfg80211_ch_switch_notify.cold+0xc/0x13 [cfg80211]
Code: 21 06 00 eb 0f 0b e9 c3 29 fc ff 48 c7 c7 60 0a 70 c1 e8 0e 06 00 eb 0f 0b e9 70 31 fc ff 48 c7 c7 60 0a 70 c1 e8 fb 05 00 eb <0f> 0b e9 b5 38 fc ff 48 c7 c7 60 0a 70 c1 48 89 1c 24 e8 e4 05 00
RSP: 0018:ffffa5630a88bdd8 EFLAGS: 00010246
RAX: 0000000000000024 RBX: 0000000000001446 RCX: 0000000000000000
RDX: 0000000000000000 RSI: ffff9f14ce8d7648 RDI: ffff9f14ce8d7648
RBP: ffff9f14c86f9c00 R08: ffff9f14ce8d7648 R09: 0000000000000633
R10: 0000000000000003 R11: 000000000001e8d4 R12: ffffa5630a88be10
R13: ffff9f1497404000 R14: ffff9f149333f800 R15: ffff9f149333fb00
FS:  0000000000000000(0000) GS:ffff9f14ce8c0000(0000) knlGS:0000000000000000
CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
CR2: 000005f1793312d0 CR3: 00000003d7574000 CR4: 0000000000340ee0
Call Trace:
 rtw_cfg80211_ch_switch_notify+0xf4/0x146 [88XXau]
 ? rtw_chk_start_clnt_join+0x8c/0x8c [88XXau]
 rtw_chk_start_clnt_join+0x5d/0x8c [88XXau]
 join_cmd_hdl+0x27b/0x387 [88XXau]
 rtw_cmd_thread+0x253/0x3a9 [88XXau]
 kthread+0x131/0x170
 ? rtw_stop_cmd_thread+0x39/0x39 [88XXau]
 ? kthread_park+0x80/0x80
 ret_from_fork+0x22/0x40
```

terencode, 2019-09-06:
After resuming from suspend the adapter will not work. The above warning is printed and I also see:
```
sept. 06 11:19:51 terence-arch iwd[645]: Unexpected connection related event -- is another supplicant running?
seèpt. 06 11:19:51 terence-arch iwd[645]: Received Deauthentication event, reason: 6, from_ap: true
``

kimocoder, 2019-09-06:
Confirmed. Adding kernel v5.3 fixes in some hours, I'll leave a note

terencode, 2019-09-06:
Thank you very much!

terencode, 2019-09-08:
@kimocoder can't compile with your latest commits:
```
/var/lib/dkms/rtl88xxau/r822.e35068c/build/os_dep/linux/rtw_cfgvendor.c:2167:4: error: ‘const struct wiphy_vendor_command’ has no member named ‘policy’
 2167 |   .policy = VENDOR_CMD_RAW_DATA,
      |    ^~~~~~
/var/lib/dkms/rtl88xxau/r822.e35068c/build/os_dep/linux/rtw_cfgvendor.c:2167:13: error: ‘VENDOR_CMD_RAW_DATA’ undeclared here (not in a function); did you mean ‘VENDOR_CMD_MAX_DATA_LEN’?
 2167 |   .policy = VENDOR_CMD_RAW_DATA,
      |             ^~~~~~~~~~~~~~~~~~~
      |             VENDOR_CMD_MAX_DATA_LEN
/var/lib/dkms/rtl88xxau/r822.e35068c/build/os_dep/linux/rtw_cfgvendor.c:2181:4: error: ‘const struct wiphy_vendor_command’ has no member named ‘policy’
 2181 |   .policy = VENDOR_CMD_RAW_DATA,
      |    ^~~~~~
make[2]: *** [scripts/Makefile.build:279: /var/lib/dkms/rtl88xxau/r822.e35068c/build/os_dep/linux/rtw_cfgvendor.o] Error 1
```

kimocoder, 2019-09-08:
I know, will finish it up soon :+1:

terencode, 2019-09-08:
Ok cool 😎

kimocoder, 2019-09-08:
@terencode update and report back. remember to reload module of course :+1:

terencode, 2019-09-09:
@kimocoder thanks but unfortunately I'm still getting the warning `WARNING: CPU: 2 PID: 1346 at net/wireless/nl80211.c:16023 cfg80211_ch_switch_notify.cold+0xc/0x13 [cfg80211]`
on kernel 5.2.13. It appears connected to the network but no internet.

terencode, 2019-09-09:
Ok the internet problem is a systemd issue that occurs at boot even with 5.2.20 : 
```
sept. 09 16:36:32 terence-arch systemd-networkd[595]: wlan0: Failed
sept. 09 16:36:32 terence-arch systemd-networkd[595]: wlan0: DHCP4 CLIENT: Failed to set IAID+DUID: Device or resource busy
sept. 09 16:36:32 terence-arch systemd-networkd[595]: wlan0: IPv6 successfully enabled
sept. 09 16:36:32 terence-arch systemd-udevd[529]: wlan0: Failed to process device, ignoring: Device or resource busy
sept. 09 16:36:32 terence-arch systemd-udevd[529]: wlan0: Failed to rename network interface 3 from 'wlan0' to 'wlp1s0f0u6': Device or resource busy
```

Workaround is to reload the module.

kimocoder, 2019-09-09:
what device/chipset is this?

terencode, 2019-09-09:
Archer T4U/rtl8812au

terencode, 2019-09-28:
@kimocoder Update on the situation:  trying with latest commits from 5.6.4.1 branch, it still does not work with both kernel 5.2.16 and 5.3.1 the `BUG: kernel NULL pointer dereference, address: 00000000000008c0` is in both kernels, full trace in my previous comment.

terencode, 2019-09-29:
With #424 (thanks @u-ra!) it's usable again. I just have this that has been reported already when restarting iwd:
```
------------[ cut here ]------------
WARNING: CPU: 2 PID: 34040 at net/wireless/nl80211.c:16023 cfg80211_ch_switch_notify.cold+0xc/0x13 [cfg80211]
Modules linked in: 88XXau(OE) cfg80211 snd_hrtimer snd_seq_midi snd_seq_dummy snd_seq_midi_event snd_seq overlay kvm_amd wacom joydev mousedev input_leds hid_logitech_hidpp hid_generic uas usbhid usb_storage hid igb dca fuse lz4 lz4_compress rfkill ccm algif_aead des_generic arc4 algif_skcipher cmac md4 kvm algif_hash snd_hda_codec_realtek irqbypass snd_hda_codec_generic snd_usb_audio af_alg zram ledtrig_audio crct10dif_pclmul crc32_pclmul snd_hda_codec_hdmi ghash_clmulni_intel snd_hda_intel(OE) snd_usbmidi_lib snd_rawmidi snd_hda_codec(OE) snd_seq_device media snd_hda_core aesni_intel snd_hwdep snd_pcm aes_x86_64 crypto_simd snd_timer cryptd glue_helper ccp snd pcspkr zenpower(OE) soundcore rng_core pinctrl_amd gpio_amdpt evdev pcc_cpufreq mac_hid acpi_cpufreq wmi_bmof mxm_wmi vboxnetflt(OE) vboxnetadp(OE) vboxpci(OE) nls_iso8859_1 nls_cp437 vfat fat vboxdrv(OE) pkcs8_key_parser i2c_dev i2c_piix4(OE) sg crypto_user asus_wmi_sensors(OE) wmi ip_tables x_tables ext4
 crc32c_generic crc16 mbcache jbd2 sd_mod ahci libahci libata crc32c_intel xhci_pci scsi_mod xhci_hcd amdgpu amd_iommu_v2 gpu_sched i2c_algo_bit ttm drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops drm agpgart [last unloaded: cfg80211]
CPU: 2 PID: 34040 Comm: RTW_CMD_THREAD Tainted: G           OE     5.2.16-32-ck1-tkg-MuQSS #1
Hardware name: System manufacturer System Product Name/CROSSHAIR VI HERO, BIOS 7501 09/23/2019
RIP: 0010:cfg80211_ch_switch_notify.cold+0xc/0x13 [cfg80211]
Code: 09 2a 2d d5 0f 0b e9 7f 1c fc ff 48 c7 c7 e0 19 23 c1 e8 f6 29 2d d5 0f 0b e9 ec 23 fc ff 48 c7 c7 e0 19 23 c1 e8 e3 29 2d d5 <0f> 0b e9 ad 2b fc ff 48 c7 c7 e0 19 23 c1 48 89 1c 24 e8 cc 29 2d
RSP: 0018:ffffb012c26e3dd8 EFLAGS: 00010246
RAX: 0000000000000024 RBX: 000000000000146e RCX: 0000000000000000
RDX: 0000000000000000 RSI: ffff9dbf4e897568 RDI: 00000000ffffffff
RBP: ffff9dbf49f87800 R08: 0000000000000c27 R09: 0000000000000003
R10: 0000000000000001 R11: 0000000000011b80 R12: ffffb012c26e3e10
R13: ffff9dbf4083a000 R14: ffff9dbed44c3800 R15: ffff9dbed44c3b00
FS:  0000000000000000(0000) GS:ffff9dbf4e880000(0000) knlGS:0000000000000000
CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
CR2: 000055950d613018 CR3: 000000036cc82000 CR4: 0000000000340ee0
Call Trace:
 rtw_cfg80211_ch_switch_notify+0xf4/0x146 [88XXau]
 ? rtw_chk_start_clnt_join+0x8c/0x8c [88XXau]
 rtw_chk_start_clnt_join+0x5d/0x8c [88XXau]
 join_cmd_hdl+0x27b/0x387 [88XXau]
 rtw_cmd_thread+0x253/0x3a9 [88XXau]
 kthread+0x131/0x170
 ? rtw_stop_cmd_thread+0x39/0x39 [88XXau]
 ? kthread_park+0x80/0x80
 ret_from_fork+0x22/0x40
---[ end trace 598a0dd8d68a58ad ]---
```

kimocoder, 2019-11-11:
@terencode 

Update to branch v5.6.4.2, Realtek relesed a new base with Android 10 support + the P2P NoA buffer issue and I've added lots on top of that again.

The CPU warning should be resolved, none here at all :+1: 
However, if you'd encounter any more issues, feel free to open new tickets. Thanks!

closed 2019-11-11 by kimocoder

terencode, 2019-11-11:
@kimocoder Thank you for your hard work, testing it soon :+1:

mikegleasonjr, 2020-05-07:
I still have this in Ubuntu 20.04 kernel 5.4.0-29-generic

satmandu, 2020-05-08:
Still getting these errors with the Experimental v5.7.0 driver, on a custom kernel with Ubuntu 20.04 but they appear to be non-fatal WARNINGs only, so I guess that's ok?

```
[   32.400997] ------------[ cut here ]------------
[   32.401154] WARNING: CPU: 1 PID: 1838 at net/wireless/nl80211.c:3159 nl80211_send_chandef+0x158/0x168 [cfg80211]
[   32.401156] Modules linked in: iptable_nat nf_nat nf_conntrack nf_defrag_ipv4 iptable_mangle iptable_filter bluetooth ecdh_generic ecc dm_multipath bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) videobuf2_dma_contig v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 raspberrypi_hwmon brcmfmac videobuf2_common hwmon videodev vc_sm_cma(C) brcmutil mc rpivid_mem uio_pdrv_genirq uio sch_fq_codel i2c_dev ip_tables x_tables ipv6 nf_defrag_ipv6 btrfs blake2b_generic zstd_decompress zstd_compress xxhash raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor xor_neon raid6_pq raid1 raid0 linear md_mod dm_mirror dm_region_hash dm_log dm_mod zfs(PO) zunicode(PO) zlua(PO) zcommon(PO) znvpair(PO) zavl(PO) icp(PO) spl(O) zlib_deflate 88XXau(O) cfg80211 rfkill v3d gpu_sched vc4 cec drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops drm drm_panel_orientation_quirks backlight snd_soc_core snd_pcm_dmaengine snd_pcm sdhci_iproc snd_timer snd
[   32.401227] CPU: 1 PID: 1838 Comm: RTW_CMD_THREAD Tainted: P         C O      5.6.10-gf44c50c86-raspi2 #1
[   32.401229] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[   32.401232] pstate: 40000005 (nZcv daif -PAN -UAO)
[   32.401276] pc : nl80211_send_chandef+0x158/0x168 [cfg80211]
[   32.401311] lr : nl80211_send_chandef+0x38/0x168 [cfg80211]
[   32.401314] sp : ffffffc019883c10
[   32.401316] x29: ffffffc019883c10 x28: ffffffe75c093430 
[   32.401319] x27: ffffffe75bedb1d0 x26: 0000000000000000 
[   32.401322] x25: ffffffc019883d58 x24: ffffff80f0221000 
[   32.401325] x23: ffffff80f58b3100 x22: ffffff80cc3d5880 
[   32.401328] x21: ffffff80ca236014 x20: ffffff80dc1cf800 
[   32.401331] x19: ffffffc019883d58 x18: 0000000000000000 
[   32.401333] x17: 0000000000000000 x16: ffffffe7819d9178 
[   32.401336] x15: 0000000000000000 x14: 0000000000000000 
[   32.401339] x13: ffffffff00000000 x12: ffffffffffffffff 
[   32.401341] x11: 0000000000000004 x10: 0000000000000004 
[   32.401344] x9 : ffffffe75be6f190 x8 : ffffff80ca23601c 
[   32.401346] x7 : 0000000000000000 x6 : ffffff80ca23601c 
[   32.401349] x5 : 0000000000000000 x4 : 00000000ffff32a1 
[   32.401351] x3 : 0000000000002e68 x2 : 00000000000000be 
[   32.401354] x1 : 0000000019883d90 x0 : 0000000000000000 
[   32.401357] Call trace:
[   32.401391]  nl80211_send_chandef+0x158/0x168 [cfg80211]
[   32.401424]  nl80211_ch_switch_notify.isra.0.constprop.0+0xec/0x180 [cfg80211]
[   32.401455]  cfg80211_ch_switch_started_notify+0x54/0xd8 [cfg80211]
[   32.401562]  rtw_cfg80211_ch_switch_notify+0x14c/0x158 [88XXau]
[   32.401613]  rtw_chk_start_clnt_join+0x78/0x84 [88XXau]
[   32.401662]  join_cmd_hdl+0x248/0x354 [88XXau]
[   32.401717]  rtw_cmd_thread+0x2f4/0x45c [88XXau]
[   32.401725]  kthread+0xfc/0x128
[   32.401730]  ret_from_fork+0x10/0x1c
[   32.401734] ---[ end trace 54a9ca2fbdb5d811 ]---
```

mikegleasonjr, 2020-05-08:
For me I experience a full computer lock-up

kimocoder, 2020-05-10:
v5.6.4.1 should not be used.. it was released, but the v5.6.4.2 came down right after, so it hasn't been patched much.

stefanofornari, 2020-08-24:
Hello, I have the same issues with 5.4.51+:

Aug 23 18:23:20 0x0002 kernel: [  123.514438] WARNING: CPU: 0 PID: 337 at net/wireless```
/nl80211.c:3157 nl80211_send_chandef+0x174/0x18c [cfg80211]
Aug 23 18:23:20 0x0002 kernel: [  123.514566] Modules linked in: 8021q garp stp llc 8192eu(O) hid_logitech_hidpp joydev evdev sha256_generic libsha256 cfg80211 hid_logitech_dj rfkill raspberrypi_hwmon bcm2835_codec(C) v4l2_mem2mem bcm2835_v4l2(C) snd_bcm2835(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common snd_pcm snd_timer snd videodev mc vc_sm_cma(C) uio_pdrv_genirq uio fixed i2c_dev ip_tables x_tables ipv6 nf_defrag_ipv6
Aug 23 18:23:20 0x0002 kernel: [  123.514811] CPU: 0 PID: 337 Comm: RTW_CMD_THREAD Tainted: G        WC O      5.4.51+ #1333
Aug 23 18:23:20 0x0002 kernel: [  123.514825] Hardware name: BCM2835
Aug 23 18:23:20 0x0002 kernel: [  123.514837] Backtrace: 
Aug 23 18:23:20 0x0002 kernel: [  123.514884] [<c0014cb4>] (dump_backtrace) from [<c0014fa4>] (show_stack+0x20/0x24)
Aug 23 18:23:20 0x0002 kernel: [  123.514967]  r6:00000009 r5:00000000 r4:00000000 r3:a184da11
Aug 23 18:23:20 0x0002 kernel: [  123.515002] [<c0014f84>] (show_stack) from [<c074d81c>] (dump_stack+0x20/0x28)
Aug 23 18:23:20 0x0002 kernel: [  123.515082] [<c074d7fc>] (dump_stack) from [<c0023b7c>] (__warn+0xdc/0x100)
Aug 23 18:23:20 0x0002 kernel: [  123.515111] [<c0023aa0>] (__warn) from [<c0023c50>] (warn_slowpath_fmt+0xb0/0xc0)
Aug 23 18:23:20 0x0002 kernel: [  123.515140]  r9:bf1ade74 r8:00000c55 r7:bf161998 r6:00000009 r5:c0a47028 r4:00000000
Aug 23 18:23:20 0x0002 kernel: [  123.516557] [<c0023ba4>] (warn_slowpath_fmt) from [<bf161998>] (nl80211_send_chandef+0x174/0x18c [cfg80211])
Aug 23 18:23:20 0x0002 kernel: [  123.516596]  r9:d25b9000 r8:d11e3e84 r7:c0a47028 r6:d461b0c0 r5:d11e3e84 r4:d461b0c0
Aug 23 18:23:20 0x0002 kernel: [  123.519348] [<bf161824>] (nl80211_send_chandef [cfg80211]) from [<bf16b23c>] (nl80211_ch_switch_notify.constprop.22+0xd0/0x198 [cfg80211])
Aug 23 18:23:20 0x0002 kernel: [  123.519474]  r7:d11e3e10 r6:00000058 r5:c0a47028 r4:d461b0c0
Aug 23 18:23:20 0x0002 kernel: [  123.522460] [<bf16b16c>] (nl80211_ch_switch_notify.constprop.22 [cfg80211]) from [<bf16e3e8>] (cfg80211_ch_switch_notify+0x98/0x114 [cfg80211])
Aug 23 18:23:20 0x0002 kernel: [  123.522866]  r10:00000985 r9:d11e3e94 r8:d45e0800 r7:d25b9000 r6:d25b8800 r5:d11e3e84
Aug 23 18:23:20 0x0002 kernel: [  123.522964]  r4:d45e08b4
Aug 23 18:23:20 0x0002 kernel: [  123.526821] [<bf16e350>] (cfg80211_ch_switch_notify [cfg80211]) from [<bf3d1240>] (rtw_cfg80211_ch_switch_notify+0xf8/0x13c [8192eu])
Aug 23 18:23:20 0x0002 kernel: [  123.526972]  r9:d25b91a0 r8:00000006 r7:0000098f r6:00000001 r5:dc990000 r4:c0a47028
Aug 23 18:23:20 0x0002 kernel: [  123.530968] [<bf3d1148>] (rtw_cfg80211_ch_switch_notify [8192eu]) from [<bf391684>] (rtw_chk_start_clnt_join+0x70/0xc8 [8192eu])
Aug 23 18:23:20 0x0002 kernel: [  123.531091]  r10:dc98db6a r9:dc98d661 r8:dc98d9a0 r7:c0a47028 r6:dc98daca r5:00000001
Aug 23 18:23:20 0x0002 kernel: [  123.531107]  r4:00000006
Aug 23 18:23:20 0x0002 kernel: [  123.535091] [<bf391614>] (rtw_chk_start_clnt_join [8192eu]) from [<bf39192c>] (join_cmd_hdl+0x250/0x384 [8192eu])
Aug 23 18:23:20 0x0002 kernel: [  123.535340]  r5:dc98d662 r4:dc98d000
Aug 23 18:23:20 0x0002 kernel: [  123.539859] [<bf3916dc>] (join_cmd_hdl [8192eu]) from [<bf364374>] (rtw_cmd_thread+0x3b0/0x400 [8192eu])
Aug 23 18:23:20 0x0002 kernel: [  123.540073]  r10:bf451000 r9:ffffbb08 r8:d11e2000 r7:dc98ded0 r6:dc98dee8 r5:d3de25c0
Aug 23 18:23:20 0x0002 kernel: [  123.540093]  r4:dc98d000
Aug 23 18:23:20 0x0002 kernel: [  123.542871] [<bf363fc4>] (rtw_cmd_thread [8192eu]) from [<c0044134>] (kthread+0x11c/0x154)
Aug 23 18:23:20 0x0002 kernel: [  123.543236]  r10:d11c3118 r9:d11e7c28 r8:bf363fc4 r7:dc98d000 r6:00000000 r5:d3509f20
Aug 23 18:23:20 0x0002 kernel: [  123.543307]  r4:d11c3100
Aug 23 18:23:20 0x0002 kernel: [  123.543350] [<c0044018>] (kthread) from [<c00090ac>] (ret_from_fork+0x14/0x28)
Aug 23 18:23:20 0x0002 kernel: [  123.543374] Exception stack(0xd11e3fb0 to 0xd11e3ff8)
Aug 23 18:23:20 0x0002 kernel: [  123.543397] 3fa0:                                     00000000 00000000 00000000 00000000
Aug 23 18:23:20 0x0002 kernel: [  123.543420] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Aug 23 18:23:20 0x0002 kernel: [  123.543440] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000
Aug 23 18:23:20 0x0002 kernel: [  123.543466]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c0044018
Aug 23 18:23:20 0x0002 kernel: [  123.543484]  r4:d3509f20 r3:ffffffff
```