aircrack-ng/rtl8812au #384: CONFIG_RTL8821C not working

windy32, 2019-06-24:
There are a lot of errors when compiling RTL8821C and they cannot be simply fixed.
A working version can be found at this repo:
https://github.com/xfor19120101/rtl8821cu-5.4.1-kernel-5.0

Hurricos, 2019-07-30:
Sorry, @windy32 - you're missing some background here.  Realtek has a habit of splitting off a number of drivers from a common tree, and in the version you tried to compile, the RTL8821C compiler flag is still present. This does not mean a driver can be compiled from these sources that supports the RTL8821CU.

Now that Realtek is moving ahead past its older hardware, the `rtw88` project is intended to work with multiple different RTL USB modules.

kimocoder, 2019-07-31:
8821C is not supported in this driver

closed 2019-07-31 by kimocoder

NoahAndrews, 2019-07-31:
What about #388?

Hurricos, 2019-07-31:
CU is not AU, and I believe that refers to either 11 or 12. The 8821AU driver runs the RTL8811AU chip; probably a typo as those patches look to be for MIMO