aircrack-ng/rtl8812au #385: Open.HD RTL8812AU TXpower uncertainty.

htcohio, 2019-06-26:
Hello Friends,

Our Open.HD FPV digital DIY system has been using the v5.2.20 branch as it was the most stable.

Part of our wish list is to have consistent transmit power.  I realize that these newer cards may have Hardware locks or restrictions but I just wanted to share some of our TX power readings with you.

The first screenshot is an example of the comparison between 5700 and 5180 mhz. 

The second is just a screenshot of a chart I found on Google.  You will notice 

The 3rd is a screenshot of the 5.8 frequency with DFS/TPC (Dynamic frequency switching and transmission power control) requirements.....
-- we've had several users on 5.8 that have had pretty good success but some say that shortly end of the flights the air suddenly stops transmitting or can no longer receive anything on the ground... 

I am curious if maybe DFS is being triggered for some reason forcing the air to switch frequencies even though it is in packet injection mode?  this is just a guess but I would appreciate any feedback you may have.

Since Aircrack, and Open.hd is made up of a group of Hackers for the most part obviously the hope is that we can completely bypass some of the limitations just for consistency with the system.

these test results have been confirmed on the following WiFi cards with the RTL8812AU Chipset:

   • Edimax AC1200
   • ASUS AC56 
*this is actually A dual antenna module but the actual transmitting antenna is an internal PCB type.  so the single SMA antenna on the Asus ac56 is only for receiving while operating in 802.11 BG mode.
   • ALFA AWUS036ACH
   • ALFA AWUS036AC

*FYI we are injecting frames for the video transmission and operating in monitor mode for the receiver while 1024bit frame sizes.

P.S. We regularly build new images with updates that appear to be added to the V5.2.20 driver and we still have the same 2.4Ghz significant packet loss while in monitor mode..... 5.8ghz appears to perform for much better in comparison.  But, in order to achieve the goal of a legitimate DIY alternative to DJI lightbridge type of system we really need to get 2.4 working.

it would be awesome if one of you developers that are very knowledgeable about the Realtek cards would be willing to take on testing the system out and putting together some optimizations geared towards the 802.11b/g rates that we utilize...

Ultimately, our hope is to have an 8812au dual band driver that is comparable to the atheros ar9271 ATH9k driver that works perfectly but it is becoming more difficult to find. 

our project has a multitude of pre-built images so if anybody has:

 2 x raspberry pis 
 2 x RTL8812AU cards
 1 x Pi camera 

Flashing our image directly onto SD cards gets you up and running right away. 

Here is the link to our GitHub Wiki for anyone that's interested. 
https://github.com/HD-Fpv/Open.HD/wiki

[image]
[image]

htcohio, 2019-06-28:
Hi,
I have more power meter results that you might be interested in and here are the results.

FYI all of these tests were done on 802.11b rates at 11mbps in order to maximize the measured power output on ALFA AWUS036ACH with RTL8812AU Chipset.

A few observations.

1. As you can see, it looks like it's automatically limiting the transmit power on DFS radar channels even though the maximum limit is 1000mw. 
my understanding is that DFS/TPC is disabled in the source code but the test results seem to show otherwise.

2. Some users have claimed that DFS channels suddenly stopped transmitting from the Drone. do you have any suggestions on how we might be able to ground test to see if some radar is forcing a channel switch that is terrible for us because we use fixed data rates and channels for our systems? 

TXpower Measurements
5180=500mw
5220=460mw
5240=350mw
5260=420mw
5280=420mw
5300=420mw
5320=very poor performance

5500=140mw DFS CHANNEL
5520=140mw DFS CHANNEL
5540=140mw DFS CHANNEL
5560=140mw DFS CHANNEL
5580=140mw DFS CHANNEL
5600=160mw DFS CHANNEL

5620
5640
5660
5680
5700=180mw

5745=210mw
5765=250mw
5785=250mw
5805=very poor performance
5825=230mw

htcohio, 2019-06-30:
Bump... Poke... ? 

;)

kimocoder, 2019-06-30:
I'm a bit busy these days, but branch of v5.6.4 and above has updated PHYDM which fixed the range issues. Unfortunately we haven't added 8814au and 8821au support to it yet. Hopefully I got more time soon. Cheers

htcohio, 2019-07-01:
thank you for the reply.

it's just a little more complicated for us then changing the driver.

you see several people some of which are no longer involved with the project implemented various patches and fixes for various reasons.

myself and the other developers/testers who work with it have tried spending time but we need to reverse engineer and look back to see what was implemented. 

I've been starting to dig a little bit and I'm starting to see the complexity of these drivers compared to the ones from the past. 

because of our unique use case, consistent higher power output with the least amount of dropped packets is obviously everybody's gold but, by the way our transmission is handled it is even more critical.

honestly, myself and others really are not concerned with 8814/21 drivers because more antennas on the air add weight. Also, there do not seem to be very many RTL8821AU cards that are of the high power type. 

We have discovered cards with the Sky85601 LNA PA control module give the best performance for some reason (Like the ASUS AC56). 

in the short to medium-term I'd like to compile a wish list specific to our needs and also investigate these drivers further. 

thank you so much for the response.

kimocoder, 2019-08-17:
@htcohio Hi there! there's a problem with txpower (and range of course) on the older versions.. but now that v5.6.4.1 is released, the problem has been fixed (Realtek provided a upgrade on the phydm). So if your using the 8812au chipset (specific, NOT 8811au, 8814au and 8821au) you could actually move over to the newest branch and see if that fixes some issues.. please report back, and sorry for taking such long time :+1: