aircrack-ng/rtl8812au #387: 8812au issues on hcxdumptool

veganinside, 2019-06-30:
After installing on Raspbian and tested on aircrack, once I try hcxdumptool after setting the interface on monitor mode I cant capture any package. anyone using this driver and hcxdumptools without issues?

dontdeadopeninside, 2019-07-19:
I too have this issue.

kimocoder, 2019-08-16:
`root@kali:~/Desktop/wifite2# hcxdumptool -i wlan0
initialization...

start capturing (stop with ctrl+c)
INTERFACE................: wlan0
ERRORMAX.................: 100 errors
FILTERLIST...............: 0 entries
MAC CLIENT...............: b025aae5b02b
MAC ACCESS POINT.........: 002419143b86 (incremented on every new client)
EAPOL TIMEOUT............: 150000
REPLAYCOUNT..............: 62460
ANONCE...................: c00a01ef25502fc534b523730059bd801e55fd48f55719240cfa2e3d5dfcc41e

INFO: cha=11, rx=595, rx(dropped)=0, tx=180, powned=0, err=0`

kimocoder, 2019-08-16:
@veganinside @dontdeadopeninside 

I do.. which branch are you guys using?

ZerBea, 2019-08-17:
I can confirm @kimocoder:
5.2.20 is working fine out of the box (monitor mode and full packet injection):
https://github.com/aircrack-ng/rtl8812au/issues/376#issuecomment-521911820

ZerBea, 2019-08-17:
If you're running the driver on a system running kernel >= 4.20 it is possible that you are affected by this xhci issue:
https://bugzilla.kernel.org/show_bug.cgi?id=202541
If the patch
https://bugzilla.kernel.org/show_bug.cgi?id=202541#c20
is not reverted, dmesg will show you this warning:
[ 9488.948365] xhci_hcd 0000:03:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
after the device is plugged-in:
[ 9322.177857] usb 1-10.4: New USB device found, idVendor=7392, idProduct=a812, bcdDevice= 2.00
[ 9322.177860] usb 1-10.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 9322.177861] usb 1-10.4: Product: Edimax AC600 USB
[ 9322.177863] usb 1-10.4: Manufacturer: Realtek
[ 9322.177864] usb 1-10.4: SerialNumber: 00e04c000001
[ 9326.878219] rtl88xxau 1-10.4:1.0 wlp3s0f0u10u4: renamed from wlan0

In that case packet injection will not work, because the xhci host failed:
$ sudo hcxdumptool -i wlp3s0f0u10u4 --do_rcascan
INFO: cha=1, rx=177, rx(dropped)=0, tx=26, err=0, aps=8 aps=0 (0 in range, WARNING: packet injection possible not working)

It is also possible that monitor mode is not working ,too:
$ sudo hcxdumptool -i wlp3s0f0u10u4 --do_rcascan
INFO: cha=6, rx=0, rx(dropped)=0, tx=12, err=0, aps=0 aps=0 (0 in range, WARNING: packet injection possible not working)

That isn't a rtl8812au driver issue and there is nothing @ kimocoder can do here. The issue is related to the xhci host and other drivers are affected, too:
https://bugzilla.kernel.org/show_bug.cgi?id=202541#c47

kimocoder, 2019-08-17:
I'll force it down to USB 2.0 again, the USBmodeswitch code and related is very hackish (dirty) anyway

kimocoder, 2019-08-17:
But what does @veganinside actually have of chipset? I need more information, because on the CHIPSET 8812au (the drivers named that too, but we have several in use)

I need to know which device/chipset & which branch you are trying to accomplish this with :+1:

kimocoder, 2019-11-11:
Closing issue as there's nothing to do here, for now at least.. Thanks for participating!

closed 2019-11-11 by kimocoder