aircrack-ng/rtl8812au #389: 5.6.4.1 doesnt work correctly on 8811

davidbepo, 2019-07-17:
since support for other chipsets was added to 5.6.4.1 i decided to try it, but even tough the device is recognized correctly it doesnt show any networks to connect to

if you want any more details please let me know

kimocoder, 2019-07-17:
Ok. That's 8821au driver, we'll dive into it shortly

davidbepo, 2019-07-17:
wow, thanks for the quick reply

davidbepo, 2019-08-23:
should i retest with the newest commits?

kimocoder, 2019-08-23:
8814au works... 8821au (8811au) needs some testing/fixes propably.

davidbepo, 2019-08-23:
just retested it and the same happens

also a thing that i forgot to note is that after reverting to older driver and rebooting the same happens(no networks shown) and it requires to unplug and plug the usb to work (this also happened on the original report)

fariouche, 2019-08-31:
Just pulled the updates from 5.6.4.1 and it seems there is a regression regarding the support of the 8812 chipset.
Looking at the code, the issue is because of some cleanup
in includes/hal_intf.h, the enum CHIP_TYPE must remain untouched.
This is because this enum is then used somewhere to retrieve the chip type from an array.
Changing the enums here will lead the driver to not correctly detecting the chip.
In the case of the 8812, it was initializing it as a 8188E, and then a big mess happen (incorrect functions called, crashes etc)
Just revert the enum back to its original content is enough.

I believe the problem happens for all the chipsets

kimocoder, 2019-09-01:
Tested 8812au and 8814au, they are working.. 8821au had issues before i made the cleanup commits though. I cannot seem to find my 8821au adapter, but I'll look harder when I'm back again

fariouche, 2019-09-01:
forgot to say that I got the issue when testing in AP mode. I didn't looked closer

kimocoder, 2019-09-08:
@davidbepo update and compile/build/install and report back :+1:

davidbepo, 2019-09-08:
i did test it and the results were... interesting
even tough i was still unable to connect, this time gnome-shell freaked out, the settings didnt open and weird things happened
so different behavior for sure
if you want any more details please ask

KenMacD, 2019-09-25:
@kimocoder any update on this? Can I help in testing anything?

Testing on a Kali image they're using `5.2.20`, which works, and `5.6.4.1` does not. I tried to `bisect` where it stops working, but there's a lot of non-working builds on newer kernels in between. One `dmesg` line that looks interesting in some of the `5.6.4` stuff is `rtl8812au_hal_init: Download Firmware failed!!`

DrSchottky, 2019-10-08:
Tested my RTL8811AU (0bda:a811) and isn't working with 5.6.4.1.
No scan results, no frames received in monitor mode

davidbepo, 2019-10-31:
retested with latest commit and no network shows up

kimocoder, 2019-11-09:
Move over to v5.6.4.2 instead of v5.6.4.1

closed 2019-11-09 by kimocoder

davidbepo, 2019-11-09:
@kimocoder just tried 5.6.4.2 and still no networks show up :(

KenMacD, 2019-11-11:
@kimocoder same here, still broken. Did you want to re-open this bug, or should a new one be opened?

KenMacD, 2019-11-18:
@davidbepo I just tried the latest commits on the `v5.6.4.2` branch (https://github.com/aircrack-ng/rtl8812au/commit/329baeeef7e3b2fca940d8cb9a197aba17fc8319) and a scan worked.

davidbepo, 2019-11-18:
okay, let me test real quick

davidbepo, 2019-11-18:
DKMS build error :(

davidbepo, 2019-11-18:
its the same error on the CI btw

kimocoder, 2019-11-18:
@davidbepo update and build again. just fixed it

davidbepo, 2019-11-18:
its WORKING!

davidbepo, 2019-11-18:
and i should note that in my brief testing its faster and more stable, but of course more time is required