aircrack-ng/rtl8812au #391: gcc error, unrecognized command '-mgeneral-regs-only' (rpi 3b+ raspbian stretch)

dinolunchbox, 2019-07-20:
Clean install of raspbarian stretch (tried it with buster too). Updated system. 
Cloned v5.3.4 into rtl8812au and cd rtl*
ran sudo apt-get install bc raspberrypi-kernel-headers
did sed changes for makefile (arm64 because I have a 3b+)
typed make and it then fails with a gcc error saying unknown command -mgeneral-regs-only

I actually tried it firstly with the newer raspbian buster release but had the same issue. I reverted back to stretch because this used to work for me, I'm not sure which version of the driver I used before though but I had it installed last year around November last year, but was a bit put out to find it failing with the same issue. No idea what I'm doing wrong, if anything. Googling about it, seems like maybe this is an issue with gcc and arm64 cross compiling or something (it's all dutch to me).

I also tried to install with dkms but that failed too (I remembered I couldn't get it to work last year either so I gave up with it).

kimocoder, 2019-07-20:
Use branch v5.6.4.1 instead. Report back

dinolunchbox, 2019-07-20:
Same issue with 5.6.4.1.

kimocoder, 2019-07-20:
Ok, I'll check soon, just got a shipment with RPI stuff

mcabain, 2019-07-23:
I saw the same error in v5.6.4.1 when using 64bit mode, but I was able to get it to compile in 32bit mode on my 3 B+. The module loads, I can enable monitor mode and did a test capture with wireshark, but haven't tested anything more extensive than that. I got the idea from some of the comments in issue  #377 

eg: 
`CONFIG_PLATFORM_ARM_RPI = y`
`CONFIG_PLATFORM_ARM64_RPI = n`

dinolunchbox, 2019-08-04:
Ive just tried using arm instead of arm64 and it built ok so thanks for that.

ipatch, 2019-08-14:
are there any advantages of compiling this driver for arm64 instead of arm?

kimocoder, 2019-08-16:
Guys, you should check this links out.. the issue seems related to the compiler. Please read and fix or whatever.. and report back :+1: 

https://github.com/raspberrypi/linux/issues/2682
https://www.raspberrypi.org/forums/viewtopic.php?t=158127

ipatch, 2019-08-20:
Been doing some reading up on this, and I'm pretty sure the version of GCC that ships with Raspbian 10 ie. Buster does not provide the,

```shell
-mgeneral-regs-only
```

which would explain why the 64bit version of the driver would not compile.  Granted I don't know if the Raspberry Pi core dev team maintains different version of GCC for different versions of the Pi, ie. **3,3b,3b+,4**  Presently I have a **TP-Link** Archer T9UH plugged into a Raspberry Pi 3B+ and running the below command with a `apt install build-essential` installs and sets ups **gcc (Raspbian 8.3.0-6+rpi1) 8.3.0** on **Raspbian 10 Buster**

> _for future me_, i'll edit this reply with future finds when they arise.

cheers 🍩
Chris

kimocoder, 2019-11-11:
This issue should be resolved in the v5.6.4.2 with many others issues over v5.6.4.1.
If any more problems should appear, please open a new issue report and report :1st_place_medal:

closed 2019-11-11 by kimocoder