aircrack-ng/rtl8812au #393: Beacon capturing stops after being in monitor mode for 60 seconds

ViacheslavKlepach, 2019-07-24:
Hi All,

For some reason, my WiFi sniffer(Tenda U12 AC1300) stops capturing beacons after 60 seconds.
In order to enable beacons capturing for next 60 seconds I have to rerun the command:
sudo ifconfig wlan0 down && sudo iwconfig wlan0 mode monitor && sudo ifconfig wlan0 up

Found a note here: https://www.kismetwireless.net/docs/readme/datasources_wifi/
"The rtl8812 and rtl8814 drivers (available at https://github.com/aircrack-ng/rtl8812au.git) support monitor mode, however they do not properly implement the mac80211 control layer; while they support creating VIFs for monitor mode, they do not actually provide packets."

**My setup:**
**uname:**
4.15.0-55-generic #60-Ubuntu x86_64 GNU/Linux
**dkms output:**
rtl8812au, 5.6.4.1, 4.15.0-55-generic, x86_64: installed
**Wi-Fi adapter:** Tenda U12 AC1300
**iwconfig:**
wlan0     IEEE 802.11b  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Monitor  Frequency:5.5 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
**ifconfig:**
wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        unspec 50-2B-73-C0-09-99-30-3A-00-00-00-00-00-00-00-00  txqueuelen 1000  (UNSPEC)
        RX packets 1887348  bytes 0 (0.0 B)
        RX errors 0  dropped 1069878  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
**iw dev:**
phy#0
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr 50:2b:73:c0:09:99
		type monitor
		txpower 12.00 dBm

**tshark output:**
17373 59.005796842 ff:ff:ff:ff:ff:ff Beacon frame, SN=0, FN=0, Flags=........C, BI=100, SSID=vk1234 802.11 100 6  802.11a 
17390 59.108212131 ff:ff:ff:ff:ff:ff Beacon frame, SN=0, FN=0, Flags=........C, BI=100, SSID=vk1234 802.11 100 6  802.11a 
17407 59.210597559 ff:ff:ff:ff:ff:ff Beacon frame, SN=0, FN=0, Flags=........C, BI=100, SSID=vk1234 802.11 100 6  802.11a 
17430 59.313010622 ff:ff:ff:ff:ff:ff Beacon frame, SN=0, FN=0, Flags=........C, BI=100, SSID=vk1234 802.11 100 6  802.11a 
17446 59.415413675 ff:ff:ff:ff:ff:ff Beacon frame, SN=0, FN=0, Flags=........C, BI=100, SSID=vk1234 802.11 100 6  802.11a 
17478 59.517805037 ff:ff:ff:ff:ff:ff Beacon frame, SN=0, FN=0, Flags=........C, BI=100, SSID=vk1234 802.11 100 6  802.11a 
17503 59.620207741 ff:ff:ff:ff:ff:ff Beacon frame, SN=0, FN=0, Flags=........C, BI=100, SSID=vk1234 802.11 100 6  802.11a 
17516 59.722089904 ff:ff:ff:ff:ff:ff Beacon frame, SN=0, FN=0, Flags=........C, BI=100, SSID=vk1234 802.11 100 6  802.11a 
17542 59.824998077 ff:ff:ff:ff:ff:ff Beacon frame, SN=0, FN=0, Flags=........C, BI=100, SSID=vk1234 802.11 100 6  802.11a 
17552 59.873609033 Probe Response, SN=3334, FN=0, Flags=........C, BI=100, SSID=vk1234 802.11 100 6  802.11a 
17553 59.873610566  Acknowledgement, Flags=........C 802.11 100 6  802.11a 
17573 59.927331602 ff:ff:ff:ff:ff:ff Beacon frame, SN=0, FN=0, Flags=........C, BI=100, SSID=vk1234 802.11 100 6  802.11a 
18210 64.943925293  Acknowledgement, Flags=........C 802.11 100 6  802.11a 
18564 66.475847250  Acknowledgement, Flags=........C 802.11 100 6  802.11a 
18565 66.475848042  Acknowledgement, Flags=........C 802.11 100 6  802.11a 
18684 67.403587962  Acknowledgement, Flags=........C 802.11 100 6  802.11a 
19137 69.913690958  Acknowledgement, Flags=........C 802.11 100 6  802.11a 
19607 72.424756644  Acknowledgement, Flags=........C 802.11 100 6  802.11a 
19620 72.474747045  Acknowledgement, Flags=........C 802.11 100 6  802.11a

NoahAndrews, 2019-07-24:
I can reproduce this issue, though I got 533 seconds of successful sniffing.

underscrew, 2019-08-10:
Same issue here!

kimocoder, 2019-11-14:
@ViacheslavKlepach @NoahAndrews @v-borges 

Move over to v5.6.4.2, Realtek did some work to it, so hopefully it'll be better.
Frame injection and better IOCTL handling is also packed

vladinko0, 2020-04-08:
Issue still persists.