aircrack-ng/rtl8812au #394: Does not work on kernel 5.3-rc1

terencode, 2019-07-24:
Produces the following trace when booting linux 5.3.0-rc1 and the adapter doesn't show.
```
WARNING: CPU: 4 PID: 487 at net/wireless/core.c:868 wiphy_register.cold+0x54/0x435 [cfg80211]
Modules linked in: 88XXau(O+) kvm_amd hid_generic(+) kvm snd_hda_codec_realtek irqbypass cfg80211 snd_hda_codec_generic ledtrig_audio snd_hda_codec_hdmi crct10dif_pclmul crc32_pclmul crc32c_intel ghash_clmulni_intel snd_hda_intel usbhid rfkill snd_hda_codec snd_hwdep aesni_intel snd_hda_core aes_x86_64 glue_helper snd_pcm crypto_simd cryptd wmi_bmof mxm_wmi igb snd_timer ccp snd pcspkr k10temp sp5100_tco i2c_piix4 soundcore rng_core dca gpio_amdpt wmi gpio_generic evdev acpi_cpufreq nls_iso8859_1 nls_cp437 vfat fat it87(O) hwmon_vid pkcs8_key_parser sg crypto_user ip_tables x_tables sd_mod uas usb_storage ahci libahci libata xhci_pci scsi_mod xhci_hcd amdgpu amd_iommu_v2 gpu_sched i2c_algo_bit ttm drm_kms_helper drm i2c_core
CPU: 4 PID: 487 Comm: systemd-udevd Tainted: G           O      5.3.0-2.rc1-tkg-pds #1
Hardware name: Gigabyte Technology Co., Ltd. X470 AORUS ULTRA GAMING/X470 AORUS ULTRA GAMING-CF, BIOS F41 07/22/2019
RIP: 0010:wiphy_register.cold+0x54/0x435 [cfg80211]
Code: ff e9 bf 38 fa ff 48 c7 c7 d0 c0 ac a0 e8 d0 09 62 e0 0f 0b b8 ea ff ff ff e9 a7 38 fa ff 48 c7 c7 d0 c0 ac a0 e8 b8 09 62 e0 <0f> 0b b8 ea ff ff ff e9 8f 38 fa ff 48 c7 c7 d0 c0 ac a0 e8 a0 09
RSP: 0018:ffffc90001267a60 EFLAGS: 00010282
RAX: 0000000000000024 RBX: 000000000000034e RCX: 0000000000000000
RDX: ffff88840e726dc0 RSI: ffff88840e7174c8 RDI: ffff88840e7174c8
RBP: ffffc90001267ac8 R08: ffff88840e7174c8 R09: 0000000000000003
R10: 0000000000000001 R11: 0000000000027988 R12: ffff88840a33db00
R13: 0000000000000000 R14: ffffc90001090448 R15: ffffffffa0fb83a8
FS:  00007fcf0a26a840(0000) GS:ffff88840e700000(0000) knlGS:0000000000000000
CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
CR2: 000055e84755e000 CR3: 000000040a602000 CR4: 00000000003406e0
Call Trace:
 rtw_wiphy_register+0x1e/0x20 [88XXau]
 rtw_cfg80211_ndev_res_register+0x11/0x17 [88XXau]
 rtw_os_ndev_register+0xe3/0x14c [88XXau]
 rtw_os_ndevs_register+0x9a/0xfb [88XXau]
 rtw_os_ndevs_init+0x24/0x38 [88XXau]
 rtw_drv_init+0x29a/0x337 [88XXau]
 usb_probe_interface+0x20c/0x3f0
 really_probe+0x140/0x6a0
 driver_probe_device+0xb6/0x100
 __driver_attach+0x118/0x260
 ? __device_attach_driver+0x180/0x180
 ? __device_attach_driver+0x180/0x180
 bus_for_each_dev+0x64/0x90
 bus_add_driver+0x14a/0x1e0
 driver_register+0x8b/0xd0
 usb_register_driver+0x76/0x110
 ? 0xffffffffa094f000
 rtw_drv_entry+0x56/0x1000 [88XXau]
 do_one_initcall+0x36/0x1f4
 ? __vunmap+0x2d2/0x2f0
 ? kmem_cache_alloc_trace+0x16e/0x210
 ? do_init_module+0x23/0x250
 do_init_module+0x5c/0x250
 load_module+0x2add/0x2c60
 ? __se_sys_init_module+0x149/0x170
 __se_sys_init_module+0x149/0x170
 do_syscall_64+0x55/0x120
 entry_SYSCALL_64_after_hwframe+0x44/0xa9
RIP: 0033:0x7fcf0ba53b3e
Code: 48 8b 0d 55 f3 0b 00 f7 d8 64 89 01 48 83 c8 ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 49 89 ca b8 af 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 22 f3 0b 00 f7 d8 64 89 01 48
RSP: 002b:00007fffecdc8a58 EFLAGS: 00000246 ORIG_RAX: 00000000000000af
RAX: ffffffffffffffda RBX: 000055e847421b50 RCX: 00007fcf0ba53b3e
RDX: 00007fcf0b6c384d RSI: 00000000002d7820 RDI: 000055e847e282f0
RBP: 00007fcf0b6c384d R08: 000000000000005f R09: 000055e847464d30
R10: 000055e847413010 R11: 0000000000000246 R12: 000055e847e282f0
R13: 000055e847425410 R14: 0000000000020000 R15: 000055e847421b50
---[ end trace 16efc43938a2b44d ]---
------------[ cut here ]------------
WARNING: CPU: 4 PID: 487 at /var/lib/dkms/rtl88xxau/r708.3da4742/build/os_dep/linux/os_intfs.c:1711 rtw_os_ndev_register+0xaa/0x14c [88XXau]
Modules linked in: 88XXau(O+) kvm_amd hid_generic(+) kvm snd_hda_codec_realtek irqbypass cfg80211 snd_hda_codec_generic ledtrig_audio snd_hda_codec_hdmi crct10dif_pclmul crc32_pclmul crc32c_intel ghash_clmulni_intel snd_hda_intel usbhid rfkill snd_hda_codec snd_hwdep aesni_intel snd_hda_core aes_x86_64 glue_helper snd_pcm crypto_simd cryptd wmi_bmof mxm_wmi igb snd_timer ccp snd pcspkr k10temp sp5100_tco i2c_piix4 soundcore rng_core dca gpio_amdpt wmi gpio_generic evdev acpi_cpufreq nls_iso8859_1 nls_cp437 vfat fat it87(O) hwmon_vid pkcs8_key_parser sg crypto_user ip_tables x_tables sd_mod uas usb_storage ahci libahci libata xhci_pci scsi_mod xhci_hcd amdgpu amd_iommu_v2 gpu_sched i2c_algo_bit ttm drm_kms_helper drm i2c_core
CPU: 4 PID: 487 Comm: systemd-udevd Tainted: G        W  O      5.3.0-2.rc1-tkg-pds #1
Hardware name: Gigabyte Technology Co., Ltd. X470 AORUS ULTRA GAMING/X470 AORUS ULTRA GAMING-CF, BIOS F41 07/22/2019
RIP: 0010:rtw_os_ndev_register+0xaa/0x14c [88XXau]
Code: 00 00 e8 76 df ff ff 4d 39 fd 75 7f 48 89 ef e8 ed 92 85 e0 85 c0 0f 94 c0 0f b6 c0 eb 7d 48 c7 c7 68 3c fb a0 e8 15 6d 1a e0 <0f> 0b e9 83 00 00 00 c6 83 58 45 00 00 01 e9 84 00 00 00 4c 8d b7
RSP: 0018:ffffc90001267af0 EFLAGS: 00010286
RAX: 0000000000000024 RBX: ffffc9000108d000 RCX: 0000000000000006
RDX: 0000000000000007 RSI: 0000000000000082 RDI: ffff88840e7174c0
RBP: ffff888403f40000 R08: 00000000000004b2 R09: 0000000000000003
R10: 0000000000000001 R11: 0000000000026854 R12: ffffc900010903a2
R13: 0000000000000000 R14: ffffc90001090448 R15: ffffffffa0fb83a8
FS:  00007fcf0a26a840(0000) GS:ffff88840e700000(0000) knlGS:0000000000000000
CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[... 62 lines cut ...]
 bus_add_driver+0x14a/0x1e0
 driver_register+0x8b/0xd0
 usb_register_driver+0x76/0x110
 ? 0xffffffffa094f000
 rtw_drv_entry+0x56/0x1000 [88XXau]
 do_one_initcall+0x36/0x1f4
 ? __vunmap+0x2d2/0x2f0
 ? kmem_cache_alloc_trace+0x16e/0x210
 ? do_init_module+0x23/0x250
 do_init_module+0x5c/0x250
 load_module+0x2add/0x2c60
 ? __se_sys_init_module+0x149/0x170
 __se_sys_init_module+0x149/0x170
 do_syscall_64+0x55/0x120
 entry_SYSCALL_64_after_hwframe+0x44/0xa9
RIP: 0033:0x7fcf0ba53b3e
Code: 48 8b 0d 55 f3 0b 00 f7 d8 64 89 01 48 83 c8 ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 49 89 ca b8 af 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 22 f3 0b 00 f7 d8 64 89 01 48
RSP: 002b:00007fffecdc8a58 EFLAGS: 00000246 ORIG_RAX: 00000000000000af
RAX: ffffffffffffffda RBX: 000055e847421b50 RCX: 00007fcf0ba53b3e
RDX: 00007fcf0b6c384d RSI: 00000000002d7820 RDI: 000055e847e282f0
RBP: 00007fcf0b6c384d R08: 000000000000005f R09: 000055e847464d30
R10: 000055e847413010 R11: 0000000000000246 R12: 000055e847e282f0
R13: 000055e847425410 R14: 0000000000020000 R15: 000055e847421b50
---[ end trace 16efc43938a2b44f ]---
```

terencode, 2019-07-24:
More verbose backtrace on the v5.6.4.1 branch:

<details><summary>Click to show</summary>
<p>

```
RTW: module init start
RTW: rtl88xxau v5.6.4.1_33916.20190619
RTW: build time: Jul 24 2019 21:30:40
RTW: rtw_inetaddr_notifier_register
RTW: 
usb_endpoint_descriptor(0):
RTW: bLength=7
RTW: bDescriptorType=5
RTW: bEndpointAddress=81
RTW: wMaxPacketSize=512
RTW: bInterval=0
RTW: RT_usb_endpoint_is_bulk_in = 1
RTW: 
usb_endpoint_descriptor(1):
RTW: bLength=7
RTW: bDescriptorType=5
RTW: bEndpointAddress=2
RTW: wMaxPacketSize=512
RTW: bInterval=0
RTW: RT_usb_endpoint_is_bulk_out = 2
RTW: 
usb_endpoint_descriptor(2):
RTW: bLength=7
RTW: bDescriptorType=5
RTW: bEndpointAddress=3
RTW: wMaxPacketSize=512
RTW: bInterval=0
RTW: RT_usb_endpoint_is_bulk_out = 3
RTW: 
usb_endpoint_descriptor(3):
RTW: bLength=7
RTW: bDescriptorType=5
RTW: bEndpointAddress=4
RTW: wMaxPacketSize=512
RTW: bInterval=0
RTW: RT_usb_endpoint_is_bulk_out = 4
RTW: 
usb_endpoint_descriptor(4):
RTW: bLength=7
RTW: bDescriptorType=5
RTW: bEndpointAddress=85
RTW: wMaxPacketSize=64
RTW: bInterval=1
RTW: RT_usb_endpoint_is_int_in = 5, Interval = 1
RTW: nr_endpoint=5, in_num=2, out_num=3

RTW: USB_SPEED_HIGH
RTW: CHIP TYPE: RTL8812
RTW: read_chip_version_8812a SYS_CFG(0xF0)=0x04411135
RTW: rtw_hal_config_rftype RF_Type is 2 TotalTxPath is 2
RTW: Chip Version Info: CHIP_8812_Normal_Chip_TSMC_C_CUT_2T2R_RomVer(0)
RTW: _ConfigChipOutEP_8812 OutEpQueueSel(0x07), OutEpNumber(3)
RTW: Boot from EFUSE, Autoload OK !
input: DREVO.Inc BladeMaster TE 88K Consumer Control as /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-4/1-4:1.0/0003:1A2C:B58F.0004/input/input24
input: DREVO.Inc BladeMaster TE 88K Mouse as /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-4/1-4:1.0/0003:1A2C:B58F.0004/input/input25
hid-generic 0003:1A2C:B58F.0004: input,hiddev98,hidraw3: USB HID v1.10 Keyboard [DREVO.Inc BladeMaster TE 88K] on usb-0000:01:00.0-4/input0
RTW: Hal_EfuseReadEFuse8812A: eFuse_Addr offset(0x141) !!
RTW: HW EFUSE
RTW: 0x000: 29 81 0A 86  0C 00 00 00  02 00 00 00  00 00 00 00  
RTW: 0x010: 24 24 24 24  24 24 29 29  29 29 29 F2  EE EE FF FF  
RTW: 0x020: FF FF 16 15  14 15 1C 1B  1B 1A 1A 1A  1C 1D 1D 1D  
RTW: 0x030: 06 EE FF FF  EE FF 0E EE  FF FF 25 25  25 25 25 25  
RTW: 0x040: 2A 2A 2A 29  29 F2 EE EE  FF FF FF FF  24 23 21 23  
RTW: 0x050: 2A 29 28 27  27 26 2A 2A  2A 2A 06 EE  FF FF EE FF  
[... 218 lines cut ...]
 load_module+0x2add/0x2c60
 ? __se_sys_init_module+0x149/0x170
 __se_sys_init_module+0x149/0x170
 do_syscall_64+0x55/0x120
 entry_SYSCALL_64_after_hwframe+0x44/0xa9
RIP: 0033:0x7f1932443b3e
Code: 48 8b 0d 55 f3 0b 00 f7 d8 64 89 01 48 83 c8 ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 49 89 ca b8 af 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 22 f3 0b 00 f7 d8 64 89 01 48
RSP: 002b:00007ffd991db2f8 EFLAGS: 00000246 ORIG_RAX: 00000000000000af
RAX: ffffffffffffffda RBX: 000055b22064ab50 RCX: 00007f1932443b3e
RDX: 00007f19320b384d RSI: 00000000005e5368 RDI: 000055b2210512f0
RBP: 00007f19320b384d R08: 000000000000005f R09: 000055b220652a80
R10: 000055b22063c010 R11: 0000000000000246 R12: 000055b2210512f0
R13: 000055b220649200 R14: 0000000000020000 R15: 000055b22064ab50
---[ end trace 3a41d129a2e4d951 ]---
RTW: rtw_wdev_free(wdev=00000000d385f5ad)
RTW: rtw_wiphy_free(phy0)
RTW: rtw_usb_primary_adapter_deinit((null))
RTW: rtw_dev_unload: bup==_FALSE
RTW: +r871xu_dev_remove, hw_init_completed=0
RTW: WARN free_recv_skb_queue not empty, 8
RTW: usb attached..., try to reset usb device
```

</p>
</details>

slvrshot, 2019-08-02:
Same issue any resolution?

terencode, 2019-08-02:
> Same issue any resolution?

Gently wait for the developer to have some free time :wink:

kimocoder, 2019-08-02:
It will not be too long.. upcoming:

* Kernel v5.0+ support
* Add some more "iw" output

And a lot of testing on all chipsets since 8811au, 8814au and 8821au was newly added

slvrshot, 2019-08-02:
I actually think I have 8821bu...not sure if this makes a difference but I'll just be patient

youling257, 2019-08-04:
same problem, rtl8723bs v5.2.17.1_26955.20180307_COEX20180201-6f52

kimocoder, 2019-08-16:
@slvrshot @youling257 

Your devices isn't supported by this driver.. this is only for 8812au, 8814au and 8821au :-1:

youling257, 2019-08-16:
@kimocoder 
https://github.com/search?q=wiphy_register.cold&type=Issues
if you can solved 88xxau on kernel 5.3 problem, and push commit patch, the patch also can use for my rtl8723bs v5.2.17.

RIP: 0010:wiphy_register.cold+0x54/0x435 [cfg80211]
RIP: 0010:rtw_os_ndev_register+0xaa/0x14c [88XXau]
RIP: 0010:rtw_os_ndevs_register+0xd1/0xfb [88XXau]

RIP: 0010:wiphy_register.cold.7+0x187/0x222 [cfg80211]
RIP: 0010:rtw_os_ndev_register+0xb1/0x186 [8723bs]
RIP: 0010:rtw_os_ndevs_register+0xf1/0x11b [8723bs]

kimocoder, 2019-08-16:
I've done lots only the last 9 hours.. you could try yourself

https://github.com/aircrack-ng/rtl8812au/tree/v5.6.4.1

kimocoder, 2019-08-16:
@terencode @slvrshot @youling257 

Here you go, boys! Enjoy kernel v5.3 support
https://github.com/aircrack-ng/rtl8812au/commit/c2fd51ad82af369529c9c3960f7d56c6f949793a

closed 2019-08-16 by kimocoder

terencode, 2019-08-16:
Great work @kimcoder, will test on monday.

slvrshot, 2019-09-12:
Awesome thank you!!