aircrack-ng/rtl8812au #395: MACS changing every reboot

SquirrelAssassin, 2019-07-24:
The mac is changing every reboot . why man ... why, so many other ways to do this.  As per your documentation its not working on a raspi, there is no /etc/NetworkManager

### NetworkManager

Newer versions of NetworkManager switches to random MAC address. Some users would prefer to use a fixed address.
Simply add these lines below
```
[device]
wifi.scan-rand-mac-address=no
```
at the end of file /etc/NetworkManager/NetworkManager.conf and restart NetworkManager with the command:
```
sudo service NetworkManager restart

kimocoder, 2019-07-24:
Well, this is "Mac randomization" a security festure built in newer driver an æd devices as a security percution/enhancement. But as Google statens, it's a flop more or less.

We disabled it in v5.1.5 or v5.2.20, but it doesn't mess with the injection and monitor mode and penetration testing festure, what's. what we're after.. so we did let it go for now.

SquirrelAssassin, 2019-07-24:
I can assure you in 5.3.4 it’s working and It cannot be disabled by the directions in the readme.  if someone is not smart enough to change there MAC address before doing something stupid they deserve to be caught.

closed 2019-07-25 by kimocoder