aircrack-ng/rtl8812au #396: Problems with using Hostapd or WPA supplicant on several WiFi adapters

AlexTeos225, 2019-08-02:
Hi all! 

I'm having trouble using multiple AWUS036ACH(Realtek RTL8812AU) together with hostapd or wpa_supplicant. In the beginning I had 6 AWUS051NH(Ralink rt3572). Everything worked perfectly: 6 hostapd running simultaneously, 6 connections through wpa _supplicant, monitoring mode, injection, de-authentication. Then I used one AWUS036ACH. And everything also worked without problems.

But now I started using several AWUS036ACH. When using more than two, functions such as monitoring mode and injection work well, but hostapd and wpa _supplicant start to work very unstable. Adapters randomly stop connecting to access points, with the following errors:

```
wpa_supplicant[2165]: ioctl[SIOCSIWAP]: Operation not permitted
wpa_supplicant[2165]: ioctl[SIOCSIWENCODEEXT]: Invalid argument
```
or
```
wpa_supplicant[2165]: wlan2: Association request to the driver failed
```
or
```
wpa_supplicant[1317]: wlan2: Invalid HW-addr family 0x0323
```

Hostapd starts without errors, but in fact does not create an access point. The behavior is unpredictable and not stable.

Driver version is 5.6.4.1. Installed on RPi 3 B+ (4.19.57-v7+) with "`sed -i flags - 's/CONFIG_PLATFORM_ARM_RPI = n/CONFIG_PLATFORM_ARM_RPI = y/g'`" due "-mgeneral-regs-only" error. Hostapd version is 2.6 and wpa_supplicant is 2.6.

What can cause problems when using multiple devices?

ipatch, 2019-08-14:
did you try compiling using instructions for the older gen rpi, ie. rpi 3 using arm instead of arm64.

I've been unable to get recent versions of this driver to compile on a b+ using the arm64 instructions found in the README but the non arm64 instructions work for the 3b+ as well.

AlexTeos225, 2019-08-15:
> did you try compiling using instructions for the older gen rpi, ie. rpi 3 using arm instead of arm64.

This is the only way to compile the latest drivers. I tried drivers from version 5.2.20. But everywhere it was not possible to achieve stable work.

kimocoder, 2019-08-17:
@AlexTeos225 @ipatch 

I actually think I stumbled upon the solution!
Check this out.. https://github.com/aircrack-ng/rtl8812au/blob/v5.6.4.1/include/ieee80211.h#L31

CONFIG_AP_MODE is a switch, but we don't have it in the Makefile even!

Try adding "CONFIG_AP_MODE=y" in Makefile, compile, build, reload module (rmmod 88XXau && modprobe 88XXau) and test. I'll to some tests soon myself too, lets get this going

kimocoder, 2019-08-17:
`root@kali:~/Desktop# hostapd -i wlan0 hostapd.conf 
Configuration file: hostapd.conf
wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
wlan0: interface state COUNTRY_UPDATE->HT_SCAN
Using interface wlan0 with hwaddr 26:95:ec:3f:8e:fd and ssid "test"
wlan0: interface state HT_SCAN->ENABLED
wlan0: AP-ENABLED `

got hostapd working without adding the switch too now. but i run the latest v5.6.4.1 branch on the 8812au chipset (newest HAL).

configuration used:
`### hostapd configuration file
### Operates at 135 Mbps in 802.11n/ht mode. iperf measures ~8.00 MBytes/sec  67.1 Mbits/sec
### I was not able to get 802.11ac/vht mode with additional side-channels configured with the driver.

ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
interface=wlan0
driver=nl80211
bridge=br0

### IEEE 802.11
ssid=test
hw_mode=a
channel=36
max_num_sta=128
auth_algs=1

### DFS
country_code=DE
ieee80211d=1
ieee80211h=1

### IEEE 802.11n
ieee80211n=1
ht_capab=[HT20][HT40+][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40]

### IEEE 802.11ac
ieee80211ac=1
#vht_oper_chwidth=1
#vht_capab=[HT40-]
#vht_oper_centr_freq_seg0_idx=46

### IEEE 802.11i
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_passphrase=<PASSPHRASE>
rsn_pairwise=CCMP

### WMM
wmm_enabled=1
uapsd_advertisement_enabled=1
wmm_ac_bk_cwmin=4
wmm_ac_bk_cwmax=10
wmm_ac_bk_aifs=7
wmm_ac_bk_txop_limit=0
wmm_ac_bk_acm=0
wmm_ac_be_aifs=3
wmm_ac_be_cwmin=4
wmm_ac_be_cwmax=10
wmm_ac_be_txop_limit=0
wmm_ac_be_acm=0
wmm_ac_vi_aifs=2
wmm_ac_vi_cwmin=3
wmm_ac_vi_cwmax=4
wmm_ac_vi_txop_limit=94
wmm_ac_vi_acm=0
wmm_ac_vo_aifs=2
wmm_ac_vo_cwmin=2
wmm_ac_vo_cwmax=3
wmm_ac_vo_txop_limit=47
wmm_ac_vo_acm=0

### TX queue parameters
tx_queue_data3_aifs=7
tx_queue_data3_cwmin=15
tx_queue_data3_cwmax=1023
tx_queue_data3_burst=0
tx_queue_data2_aifs=3
tx_queue_data2_cwmin=15
tx_queue_data2_cwmax=63
tx_queue_data2_burst=0
tx_queue_data1_aifs=1
tx_queue_data1_cwmin=7
tx_queue_data1_cwmax=15
tx_queue_data1_burst=3.0
tx_queue_data0_aifs=1
tx_queue_data0_cwmin=3
tx_queue_data0_cwmax=7
tx_queue_data0_burst=1.5`

closed 2019-08-23 by AlexTeos225