aircrack-ng/rtl8812au #398: device does not turns off after computer shuts down

logan001, 2019-08-07:
Adapter: Asus USB AC-68
Driver: [5.1.5](https://github.com/aircrack-ng/rtl8812au/commit/3df255bf2987afeb85c4d819503912abb1688b55)
If i'm connected to router or if i have it set up as AP the adapter is not powered off after computer shutdown.
I don't know if its a driver issue or kernel bug.
the strange think is that for example the adapter was set as AP before shutdown then the AP name is still broadcast even after the computer is turned off. i have it set as ap before over the night. the computer was set to shutdown at 2:00. in the morning computer was off. the adapter was on. the phone was showing the AP from my computer. of course no connection was posible but the SSID was still on. if unplug the computer everything works.
or start in windows and shutdown the computer from windows. ( a new method to turn off computer :D)
so to make it short: the adapter does not power off after computer shutsdown

kimocoder, 2019-08-17:
So, what you say is that the device actually runs without power? That's a new one..
I bet you have this option in your "bios" enabled, that deliveres usb power even when it's turned off?

logan001, 2019-08-17:
i didnt say that. i said that the device stays on even after shutting down the computer. the usb device is still powered on and seems to keep the settings with it. yes i know its strnage that the AP still shows on the phone even though the computer turned off. 
the strange thing would had been if the phone could connect to it :)

the good part is that 5.6.4.1 branch does turn the device off. but the speed of this branch is too slow for my device

about bios: even with that option on in biod (i have no clue how it is) the device still powers off from windows (or the 5.6.4.1 branch of this git). so i doubt its a problem of bios. how do i know if the device is powered off? simple: its not hot anymore. that is the simplest and accurate method to check if device is still powered cause the led might be turned off and the devices is still on.

so i'm guessing that the driver from 5.1.5 has some bug about power management and it would be nice to be fixed since its the only branch that actually works with good speeds for both AP or client modes. this 5.6.4.1 branch client speed is only 2-3 Mbytes/s compare to 10Mbytes from windows. well thats how much this old router can do. my other router broke so i can test to see how 5.6.4.1 behave on 5ghz

kimocoder, 2019-08-17:
v5.1.5 is loaded with shitty code.. loaded.. you better move a step up or something. v5.2.20(.2) is stable and decent, says the reports though

logan001, 2019-08-17:
you actually pointed me to [5.1.5](https://github.com/aircrack-ng/rtl8812au/issues/335#issuecomment-502806720).. i just hope you can fix it somehow :)

kimocoder, 2019-08-17:
@logan001 What chipset do you use? is it 8811, 8812, 8814, 8821 ...

logan001, 2019-08-17:
8814

kimocoder, 2019-08-17:
ouch.. the worst one. The 8814au HAL is really dirty & old.
On all our branches, we ripped it out from v5.1.5, because it hasn't got updated :-1:

logan001, 2019-08-17:
well that 5.1.5 branch works very nice (at least for me) except that it fails to turn off de device when computer shuts down.
any chance you can do something about it?

logan001, 2019-08-19:
oh i just rember. can you also add more cores to make? it seems that 5.1.5 branch use 1 core in dkms script

kimocoder, 2019-08-19:
Yes, just use "make -j <cores>" e.g 2..4..

0m3rta13, 2019-08-19:
> well that 5.1.5 branch works very nice (at least for me) except that it fails to turn off de device when computer shuts down.
> any chance you can do something about it?

Which 8814 device do you have?
When you say it works very nice are you referring to just general use as a WiFi adapter or for all modes/functions (Monitor, packet injection, hotspot etc)???

logan001, 2019-08-19:
I have Asus USB AC-68 and im referring to client and AP mode only.

cedricbambooza, 2021-05-11:
pls consider closing the issue, when it's solved by now :)