aircrack-ng/rtl8812au #399: 5.3.4 Having random system freezes/hangs and reboots using recent kernels, 5.1+.

joder666, 2019-08-16:
My main system is opensuse tumbleweed, i already reported the issue over there along with needed information.
Bug:
[https://bugzilla.suse.com/show_bug.cgi?id=1145133
](url)
The device/chip:
`ID 0bda:a811 Realtek Semiconductor Corp. RTL8811AU 802.11a/b/g/n/ac WLAN Adapter`

In short, as the title says as well as getting  this frequently:
`CPU: 5 PID: 2245 at drivers/usb/core/urb.c:362 usb_submit_urb+0x4d1/0x5d0 [usbcore]`

kimocoder, 2019-08-16:
Well, v5.3.4 isn't recommended.. we haven't really made much work on it when newer drivers was released. You should either stick to v5.2.20 which is more or less considered finished (by us at least), our you could be a pionér, an explorer to branch v5.6.4.1 which I just pushed lots of patches too.. but v5.3.4.. no would not recommend it

joder666, 2019-08-16:
Oook i went by the numbers and thought 5.3.4 was a better one. 
I did try 5.6.4.1 before reporting this but it did not work for the device. I'll switch to 5.2.20 as well as give it 5.6.4.1 another try and see how it goes.

kimocoder, 2019-08-16:
Just started the work on it.. will take some time to go through all chipsets, finishing up the base first. v5.2.20 is stable on all 3 chipsets supported, more or less

kimocoder, 2019-11-11:
@joder666 

Recommending everyone to move over to the new branch, v5.6.4.2. Other branches will be left behind on bug fixes and kernel support. The freeze issue should be resolved in latest branch too, it also has kernel 5.1,5.2,5.3 support and lots of other fixes.

If any more problems should appear, please open a new issue report. Thanks!

closed 2019-11-11 by kimocoder