aircrack-ng/rtl8812au #402: [5.6.4.1] implicit declaration of function 'rtw_get_wait_hiq_empty_ms'

ZeroChaos-, 2019-08-19:
Unable to build against kernel 5.2.8
```
git checkout --quiet v5.6.4.1
GIT update -->
   repository:               https://github.com/aircrack-ng/rtl8812au.git
   at the commit:            fd796b8bc11cefa7ad0e542d24d2d0ff087afdca
>>> Source unpacked in /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work
>>> Preparing source in /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999 ...
>>> Source prepared.
>>> Configuring source in /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999 ...
>>> Source configured.
>>> Compiling source in /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999 ...
ln: failed to create symbolic link 'Module.symvers': File exists
 * Preparing 88XXau module
make -j1 HOSTCC=x86_64-pc-linux-gnu-gcc CROSS_COMPILE=x86_64-pc-linux-gnu- 'LDFLAGS=-m elf_x86_64' KVER=5.2.8-pentoo-r1 KSRC=/usr/src/linux RTL8814=1 V=1 clean modules 
#make -C /usr/src/linux M=/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999 clean
cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko
cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko
cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions
make ARCH=x86 CROSS_COMPILE=x86_64-pc-linux-gnu- -C /usr/src/linux M=/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999  modules
make[1]: Entering directory '/usr/src/linux-5.2.8-pentoo-r1'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)
mkdir -p /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/.tmp_versions ; rm -f /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/.tmp_versions/*
make -f ./scripts/Makefile.build obj=/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999
  x86_64-pc-linux-gnu-gcc -Wp,-MD,/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/core/.rtw_cmd.o.d  -nostdinc -isystem /usr/lib/gcc/x86_64-pc-linux-gnu/8.3.0/include -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Wno-format-security -std=gnu89 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -DCONFIG_X86_X32_ABI -DCONFIG_AS_CFI=1 -DCONFIG_AS_CFI_SIGNAL_FRAME=1 -DCONFIG_AS_CFI_SECTIONS=1 -DCONFIG_AS_SSSE3=1 -DCONFIG_AS_AVX=1 -DCONFIG_AS_AVX2=1 -DCONFIG_AS_AVX512=1 -DCONFIG_AS_SHA1_NI=1 -DCONFIG_AS_SHA256_NI=1 -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -fno-jump-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -O2 --param=allow-store-data-races=0 -fplugin=./scripts/gcc-plugins/structleak_plugin.so -fplugin=./scripts/gcc-plugins/randomize_layout_plugin.so -fplugin=./scripts/gcc-plugins/stackleak_plugin.so -fplugin-arg-structleak_plugin-byref-all -DSTRUCTLEAK_PLUGIN -DRANDSTRUCT_PLUGIN -DSTACKLEAK_PLUGIN -fplugin-arg-stackleak_plugin-track-min-size=100 -Wframe-larger-than=1024 -fstack-protector-strong -Wno-unused-but-set-variable -Wno-unused-const-variable -fomit-frame-pointer -fno-var-tracking-assignments -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wno-stringop-truncation -fno-strict-overflow -fno-merge-all-constants -fmerge-constants -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -fmacro-prefix-map=./= -Wno-packed-not-aligned -O1 -Wall -Wno-switch -Wno-date-time -Wno-unused-value -Wno-unused-label -Wno-cast-function-type -Wno-unused-variable -Wno-unused -Wno-unused-parameter -Wno-unused-function -Wno-cast-function-type -Wno-misleading-indentation -Wno-uninitialized -Wno-missing-field-initializers -Wno-vla -g -I/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/include -Wno-date-time -DCONFIG_DISABLE_REGD_C -DDRV_NAME=\"rtl88xxau\" -I/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/platform -I/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/hal/btc -DCONFIG_RTL8812A -DCONFIG_RTL8821A -DCONFIG_MP_VHT_HW_TX_MODE -DCONFIG_RTL8814A -DCONFIG_MP_INCLUDED -DCONFIG_EFUSE_CONFIG_FILE -DEFUSE_MAP_PATH=\"/system/etc/wifi/wifi_efuse_8814au.map\" -DWIFIMAC_PATH=\"/data/wifimac.txt\" -DCONFIG_LOAD_PHY_PARA_FROM_FILE -DREALTEK_CONFIG_PATH=\"/lib/firmware/\" -DCONFIG_TXPWR_BY_RATE=1 -DCONFIG_TXPWR_BY_RATE_EN=0 -DCONFIG_TXPWR_LIMIT=1 -DCONFIG_TXPWR_LIMIT_EN=0 -DCONFIG_RTW_ADAPTIVITY_EN=0 -DCONFIG_RTW_ADAPTIVITY_MODE=0 -DCONFIG_IEEE80211W -DCONFIG_RTW_80211R -DHIGH_ACTIVE_HST2DEV=0 -DCONFIG_BR_EXT '-DCONFIG_BR_EXT_BRNAME="'br0'"' -DCONFIG_WIFI_MONITOR -DCONFIG_RTW_NAPI -DCONFIG_RTW_GRO -DCONFIG_RTW_NETIF_SG -DCONFIG_MP_VHT_HW_TX_MODE -mhard-float -DMARK_KERNEL_PFU -DCONFIG_RTW_DEBUG -DRTW_LOG_LEVEL=3 -DDM_ODM_SUPPORT_TYPE=0x04 -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT -I/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/hal/phydm  -DMODULE  -DKBUILD_BASENAME='"rtw_cmd"' -DKBUILD_MODNAME='"88XXau"' -c -o /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/core/rtw_cmd.o /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/core/rtw_cmd.c
/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/core/rtw_cmd.c: In function ‘rtw_chk_hi_queue_hdl’:
/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/core/rtw_cmd.c:4106:61: error: implicit declaration of function ‘rtw_get_wait_hiq_empty_ms’; did you mean ‘rtw_get_time_interval_ms’? [-Werror=implicit-function-declaration]
  while (_FALSE == empty && rtw_get_passing_time_ms(start) < rtw_get_wait_hiq_empty_ms()) {
                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~
                                                             rtw_get_time_interval_ms
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:279: /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1598: _module_/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-9999/work/rtl8812au_aircrack-ng-9999] Error 2
make[1]: Leaving directory '/usr/src/linux-5.2.8-pentoo-r1'
make: *** [Makefile:1603: modules] Error 2
```

kimocoder, 2019-08-19:
I think it's unable to build on all kernels, did something horrible wrong yesterday, and then I travelled to work at sea, let's see I'll revert the autoconf.h commit shortly from my phone

kimocoder, 2019-08-19:
Changed back the autoconf.h commit. "git pull" and build

ZeroChaos-, 2019-08-19:
Updated to df1b6776c7eebb869957bc0ba3344a959edf51f1 and same output

kimocoder, 2019-08-19:
Ooooh shit, that one I'm not familiar with actually.. and I'm far from my computer, jesus

kimocoder, 2019-08-19:
I'll take a look tomorrow, don't have any chance right away, sorry. Last tested commit was the Wi-Fi direct private ioctl commit som I know which commits to look at.

ZerBea, 2019-08-20:
@kimocoder 
attached rtw_debug. h will fix some of them
[rtw_debug.h.zip](https://github.com/aircrack-ng/rtl8812au/files/3520596/rtw_debug.h.zip)

Unfortunately there are some more issues inside the code.
attached  os_intfs.c will fix this issue
[os_intfs.c.zip](https://github.com/aircrack-ng/rtl8812au/files/3520599/os_intfs.c.zip)
static int rtw_ndev_notifier_call(struct notifier_block *nb, unsigned long state, void *ptr)
{
	struct net_device *ndev;
	_adapter *adapter

	if (ptr == NULL)
		return NOTIFY_DONE;

unfortunately there are some more issues realted to this typedef in data_types.h
typedef struct _ADAPTER _adapter, ADAPTER, *PADAPTER;
rtl8812au/os_dep/linux/ioctl_cfg80211.c:10310:56
_rtw_memcpy(wiphy->perm_addr, adapter_mac_addr(padapter)
For thios one, I have to dive deeper in that adapter padapter stuff

ZerBea, 2019-08-20:
rtw_debug.h is only a quick and dirty hack to let gcc compile without running into errors. So we can test if there are more (related) issues.
Major problem is how driver_types.h (includes rtw_debug.h) is included by rtw_cmd.c and rtw_mlme_ext.c

ZeroChaos-, 2019-08-20:
@ZerBea please attach a diff next time if possible.  it makes reviewing significantly easier for all involved.  If you are feeling it, you can even open a pr which makes review and merging as simple as possible.

ZerBea, 2019-08-20:
@ZeroChaos- good idea. I'll do a diff next time.
Unfortunately, the above hack doesn't justify a PR as it is really a dirty hack.

visbs, 2019-08-21:
adapter_mac_addr (adapter) (adapter->ETH_ALEN) is there any update for the same?

ZerBea, 2019-08-21:
Not yet. First I would like to talk with @kimocoder about that issue because this function is new in latest drivers).
This is the old function:
int rtw_wiphy_register(struct wiphy *wiphy)
{
	RTW_INFO(FUNC_WIPHY_FMT"\n", FUNC_WIPHY_ARG(wiphy));
}
I have no idea why it need the mac, now. Maybe it is a simple copy and past error - I really don't know it at this moment.

ZerBea, 2019-08-21:
Attached 2 patches. They should solve this issues:
missing ;
[os_intfs.c.patch.zip](https://github.com/aircrack-ng/rtl8812au/files/3524952/os_intfs.c.patch.zip)
adapter_mac_addr(padapter)
[ioctl_cfg80211.patch.zip](https://github.com/aircrack-ng/rtl8812au/files/3524955/ioctl_cfg80211.patch.zip)

Now we are running again into implicit declaration of function issues:
rtw_fwdl_test_trigger_chksum_fail
rtw_fwdl_test_trigger_wintint_rdy_fail

At this time we do not want to make another hack, like this one here
https://github.com/aircrack-ng/rtl8812au/issues/402#issuecomment-522994091
because it dosn't solve the issue (-Werror=implicit-function-declaration). It only let gcc compile without errors:
inside rtw_debug.c and rtw_debug.h are tons of #ifdef and #endif and probably one of them doesn't include the function prototypes so we received many "-Werror=implicit-function-declaration". They all have the same cause - and we have to find it.

@kimocoder good news: today I will get my rtl8188eus devices.

ZerBea, 2019-08-21:
@kimocoder not sure about your emoji, because I can't test it here (you know, only 8811 device). So, ioctl_cfg80211.patch needs to be tested.

terencode, 2019-08-21:
This was not @kimocoder emoji but mine just as a way to say "thank you for the work".

ZerBea, 2019-08-21:
@terencode oh, sorry, I didn't noticed that, because I addressed @kimocoder

visbs, 2019-08-21:
@ZerBea  error: implicit declaration of function ‘rtw_fwdl_test_trigger_chksum_fail’ [-Werror=implicit-function-declaration]
  if (rtw_fwdl_test_trigger_chksum_fail())

visbs, 2019-08-22:
[patches.zip](https://github.com/aircrack-ng/rtl8812au/files/3528274/patches.zip)
 Just replace these files , It will work like a charm.
PS: This is just a hack as of now @kimocoder, not sure about its consistency.

kimocoder, 2019-08-23:
@ ZeroChaos- @ZerBea @visbs

Problem has been resolved. "git pull" and build :1st_place_medal:

ZerBea, 2019-08-23:
@kimocoder compiling like a charm. No warnings, no errors using gcc 9.1
Good job, Christian - thanks.

closed 2019-08-23 by kimocoder

ZeroChaos-, 2019-08-23:
confirmed, it is building. thanks!