aircrack-ng/rtl8812au #403: How to check the driver is installed prorperly

minimal1, 2019-08-20:
Hello, 
First, thank you for making this driver. 

I installed this driver in ubuntu 18.04 for Jetson Nano board.
And I followed how to use monitor mode from Readme.md of this repo.
Then the wireless adapter worked well, but I got the problem when I use IBSS mode.

2 Jetson Nano boards use same wireless adapter and driver. Even they are in same cell of IBSS mode. But they can't communicate.

I guess I missed something when installing.
Is there any way to check the driver is installed properly?
or
Like #311 , Is the communication problem from not using 'mac80211' ?

kimocoder, 2019-08-23:
I have to read on this matter, and perform some tests of myself. I will come back to you.

minimal1, 2019-08-23:
Ok, thanks. I will wait for your good news :)

minimal1, 2019-08-29:
@kimocoder
How is the test going ??? 
I hope you have good news :)

kimocoder, 2019-09-08:
Haven't had any time to do some tests yet, no..

Some questions, which branch are you using?
and what chipset? and on which kernel?

minimal1, 2019-09-09:
Oh.. I see. I can wait.

I used 'v.5.3.4' brach, and the chipset is realtek 8812au, the usb device id is 2357:0122 in the command 'lsusb'
and my kernel is 4.9.140-tegra, ubuntu 18.04.3 LTS.

kimocoder, 2019-10-30:
I'm working on several issues,  the MESH support Will have to wait until I got more time

kimocoder, 2019-11-11:
@minimal1 

You should move/upgrade to branch v5.6.4.2, which got lots of fixes by both Realtek and me afterwards. Test it, and if the problem persist, re-open this issue report! Sorry for the long haul btw, have to much to do for some time :100:

closed 2019-11-11 by kimocoder

minimal1, 2019-11-12:
> @minimal1
> 
> You should move/upgrade to branch v5.6.4.2, which got lots of fixes by both Realtek and me afterwards. Test it, and if the problem persist, re-open this issue report! Sorry for the long haul btw, have to much to do for some time 💯

Okay, I will let you know if it works well. 
Thank you very much!!