aircrack-ng/rtl8812au #404: concurrent mode not working in 5.3.4

visbs, 2019-08-21:
Hi,
in driver if i wanted to enable p2p-dev-wlan0(p2p_device interface), which can be achieved with enabling the RTW_P2P_GROUP_INTERFACE 1 in ioctl_cfg80211.h file . But the problem is it is not creating any autonomous group (P2P_GO),
 The error which i get is 
```
Failed to create group interface p2p-wlan0-0 -19 (No such device found)
```
where is it is able to create groups for wlan0 

So how to achieve the same in P2P interface?.

Thank you
Vishal

visbs, 2019-08-23:
@kimocoder  what I wanted to know is whether this driver supports concurrent mode i.e (AP+STA) if yes how to achieve the same.

kimocoder, 2019-08-23:
I may only refer to https://github.com/aircrack-ng/rtl8812au/blob/v5.6.4.1/docs/Quick_Start_Guide_for_wpa_supplicant_WiFi_P2P_test.pdf and https://github.com/aircrack-ng/rtl8812au/blob/v5.6.4.1/docs/Realtek_WiFi_concurrent_mode_Introduction.pdf at the moment. I'll go through P2P tests when I'm back again :+1:

visbs, 2019-08-23:
what ever the document says if I enable it then it gives me a error saying 
```
 error: ‘struct _ADAPTER’ has no member named ‘iface_type’;
```

visbs, 2019-08-26:
@kimocoder  I have made some changes to the code and was able to get the p2p-dev-wlan0 working , but when using p2p wifi is not working and when using wifi(internet) devices are not able to connect to a autonomous group created by me. The thing is how can I use both 
Things which I have already done 
1)Enable concurrent_mode 
2)Enable rtw_p2p_group_interface
3)Enable CONFIG_IOCTL_CFG80211

kimocoder, 2019-08-26:
I need exact steps in order to simplify it for others, possibly by adding a switch to it :+1:

visbs, 2019-08-26:
1) set RTW_P2P_GROUP_INTERFACE 1 for (p2p interface to be created for p2p operations)
2) uncomment the #define CONFIG_IOCTL_CFG80211 1  flag in autoconf.h,
 and later fix all the errors which arises by their respective calls.

kimocoder, 2019-08-26:
Thanks, I'll go through it and simplify the steps in Makefile :1st_place_medal:

visbs, 2019-08-26:
Always here to help you :+1:

visbs, 2019-08-27:
@kimocoder I was able to create two interfaces one for wifi and other for p2p now using the concurrent mode and p2p_group_interface for p2p operations in wpa_supplicant, but whenever I start a autonomous p2p_group it shows both i.e wlan0 and p2p-wlan-0 in the top left drop down list .
```
wifi symbol : wlan0(Not connected)
wifi symbol : p2p-wlan0-0(not connected) 
```
-p2p-wlan0-0 is the autonomous group which i created
and suppose if I connect to some network (Ex: abcd)
```
wifi symbol : abcd
wifi symbol : p2p-dev-waln0(not connected)
```
I just wanted to know is it possible to hide only p2p-dev-wlan0 (because other intel and qualcom drivers are not showing that group in the list). which part of the code can actually hide that p2p-dev-wlan0 from the top-right drop down.

My goal: Basically I just want to show only one interface for wifi in that drop down .

kimocoder, 2019-11-09:
Move over to v5.6.4.2 instead of v5.6.4.1

closed 2019-11-09 by kimocoder