aircrack-ng/rtl8812au #408: Unable to build on Raspberry Pi 3+ with Buster - unrecognized command line option ‘-mgeneral-regs-only’

victorhooi, 2019-08-29:
I am attempting to install this driver on a Raspberry Pi 3+, running Buster.

I've installed the prerequisite tools:
```
sudo apt install dkms
sudo apt install bc raspberrypi-kernel-headers
```
I've cloned the git repo:
```
git clone https://github.com/aircrack-ng/rtl8812au.git
```
I then edit the `MAKEFILE`, and run `./dkms-install.sh`:

```
$ cd rtl8812au/
$ sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
$ sed -i 's/CONFIG_PLATFORM_ARM64_RPI = n/CONFIG_PLATFORM_ARM64_RPI = y/g' Makefile
$ sudo ./dkms-install.sh
About to run dkms install steps...

Creating symlink /var/lib/dkms/rtl8812au/5.3.4/source ->
                 /usr/src/rtl8812au-5.3.4

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=4.19.66-v7+ KSRC=/lib/modules/4.19.66-v7+/build.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.66-v7+ (armv7l)
Consult /var/lib/dkms/rtl8812au/5.3.4/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=4.19.66-v7+ KSRC=/lib/modules/4.19.66-v7+/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.66-v7+ (armv7l)
Consult /var/lib/dkms/rtl8812au/5.3.4/build/make.log for more information.
Finished running dkms install steps.
```

Here are the contents of `/var/lib/dkms/rtl8812au/5.3.4/build/make.log`:
```
$ cd /var/lib/dkms/rtl8812au/5.3.4/build/
$ cat make.log
DKMS make.log for rtl8812au-5.3.4 for kernel 4.19.66-v7+ (armv7l)
Wed 28 Aug 20:39:08 UTC 2019
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/4.19.66-v7+/build M=/var/lib/dkms/rtl8812au/5.3.4/build O="" modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.66-v7+'
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_io.o
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
make[2]: *** [scripts/Makefile.build:303: /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_cmd.o] Error 1
make[2]: *** Waiting for unfinished jobs....
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
make[2]: *** [scripts/Makefile.build:303: /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_security.o] Error 1
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
make[2]: *** [scripts/Makefile.build:303: /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_debug.o] Error 1
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
make[2]: *** [scripts/Makefile.build:303: /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_io.o] Error 1
make[1]: *** [Makefile:1519: _module_/var/lib/dkms/rtl8812au/5.3.4/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.66-v7+'
make: *** [Makefile:1723: modules] Error 2
```

0unknwn, 2019-09-02:
Run 
```
sed -i 's/^dkms build/ARCH=arm dkms build/' dkms-install.sh
sed -i 's/^MAKE="/MAKE="ARCH=arm\ /' dkms.conf
```
and try it again.

victorhooi, 2019-09-04:
I can confirm that I now seem to be able to build the driver!

(Does this mean there is a typo in the README? What is the actual issue here?)

I had to remove the partially built driver first:
```
$ sudo ./dkms-remove.sh
About to run dkms removal steps...

-------- Uninstall Beginning --------
Module:  rtl8812au
Version: 5.3.4
Kernel:  4.19.66-v7+ (armv7l)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

88XXau.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.19.66-v7+/updates/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...

DKMS: uninstall completed.

------------------------------
Deleting module version: 5.3.4
completely from the DKMS tree.
------------------------------
Done.
Finished running dkms removal steps.
```
Then I run the two lines from [above](https://github.com/aircrack-ng/rtl8812au/issues/408#issuecomment-527212525)
```
sed -i 's/^dkms build/ARCH=arm dkms build/' dkms-install.sh
sed -i 's/^MAKE="/MAKE="ARCH=arm\ /' dkms.conf
```
Then I build the driver:
```
$ sudo ./dkms-install.sh
About to run dkms install steps...

Creating symlink /var/lib/dkms/rtl8812au/5.3.4/source ->
                 /usr/src/rtl8812au-5.3.4

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
ARCH=arm 'make' -j4 KVER=4.19.66-v7+ KSRC=/lib/modules/4.19.66-v7+/build.............................................................
cleaning build area...

DKMS: build completed.

88XXau.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.66-v7+/updates/

depmod...

DKMS: install completed.
Finished running dkms install steps.
```
I run `iwconfig` again to check:d
```
pi@rp3-ang:~/code/rtl8812au $ iwconfig
eth0      no wireless extensions.

lo        no wireless extensions.

wlan0     IEEE 802.11  ESSID:off/any
          Mode:Managed  Access Point: Not-Associated   Tx-Power=31 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on

wlan1     unassociated  Nickname:"<WIFI@REALTEK>"
ddfxv                                                                                                                                                                              v          Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

zt0       no wireless extensions.

```
I am hitting a separate issue in Bettercap - https://github.com/bettercap/bettercap/issues/614 - but that may be separate

zaneclaes, 2019-10-16:
Great solution, thank you @0unknwn 
Suggestion to maintainers: add this to the README 🍻

kimocoder, 2019-11-11:
The issue should (is reported) to be solved in branch v5.6.4.2, with lots of other issues.
Recommending to move over from branches below, as they will be more or less abandoned as we cary on :+1:

closed 2019-11-11 by kimocoder