aircrack-ng/rtl8812au #409: 5.6.4.1 problem with Ubuntu 18.04

ggerla, 2019-09-02:
Hi all
I'm trying to use this version with Ubuntu 18.04 (kernel 4.18 or 5.0). The build and installation is ok, but after the reboot Ubuntu is not able to start.
I have AWUS1900 and AWUS036AC connected to my PC.
I tried using 

sudo ./dkms-install.sh

and also manual dkms commands.

LeeTiK, 2019-09-06:
same problem, Ubuntu 16.04, driver 5.6.4.1, AWUS036ACH

`[  186.764150] usb 1-5: new high-speed USB device number 6 using xhci_hcd
[  186.892220] usb 1-5: device descriptor read/64, error -71
[  187.148783] usb 1-5: New USB device found, idVendor=0bda, idProduct=8812
[  187.148789] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  187.148794] usb 1-5: Product: 802.11n NIC
[  187.148797] usb 1-5: Manufacturer: Realtek
[  187.148801] usb 1-5: SerialNumber: 123456
[  188.630090] PKCS#7 signature not signed with a trusted key
[  188.630114] 88XXau: loading out-of-tree module taints kernel.
[  188.630491] 88XXau: module verification failed: signature and/or required key missing - tainting kernel
[  188.746214] usb 1-5: USB disconnect, device number 6
[  188.746461] usbcore: registered new interface driver rtl88xxau
[  189.076227] usb 2-1: new SuperSpeed USB device number 2 using xhci_hcd
[  189.096350] usb 2-1: Int endpoint with wBytesPerInterval of 512 in config 1 interface 0 altsetting 0 ep 133: setting to 64
[  189.096507] usb 2-1: New USB device found, idVendor=0bda, idProduct=8812
[  189.096512] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  189.096516] usb 2-1: Product: 802.11n NIC
[  189.096520] usb 2-1: Manufacturer: Realtek
[  189.096523] usb 2-1: SerialNumber: 123456
[  190.227622] rtl88xxau 2-1:1.0 wlx00c0caa8a0a2: renamed from wlan0
[  190.240335] BUG: unable to handle kernel NULL pointer dereference at 00000000000008c0
[  190.240417] IP: rtw_ndev_notifier_call+0x2e/0x5a [88XXau]
[  190.240421] PGD 0 P4D 0
[  190.240428] Oops: 0000 [#1] SMP PTI
[  190.240432] Modules linked in: 88XXau(OE) rfcomm bnep nls_iso8859_1 arc4 iwlmvm mac80211 snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic uvcvideo snd_hda_intel snd_hda_codec snd_hda_core videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core videodev media mei_me snd_hwdep iwlwifi intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm cfg80211 irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel pcbc snd_pcm snd_seq_midi aesni_intel snd_seq_midi_event aes_x86_64 mei wmi_bmof msi_wmi input_leds snd_rawmidi intel_pch_thermal crypto_simd glue_helper cryptd snd_seq intel_cstate intel_rapl_perf snd_seq_device joydev snd_timer serio_raw snd btusb sparse_keymap btrtl btbcm intel_wmi_thunderbolt btintel bluetooth soundcore ecdh_generic shpchp mac_hid acpi_pad
[  190.240507]  parport_pc ppdev lp parport autofs4 hid_generic usbhid hid nouveau i915 mxm_wmi ttm i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt nvme fb_sys_fops ahci sdhci_pci alx psmouse mdio nvme_core sdhci libahci drm wmi video
[  190.240542] CPU: 1 PID: 2323 Comm: systemd-udevd Tainted: G           OE    4.15.0-58-generic #64~16.04.1-Ubuntu
[  190.240545] Hardware name: Micro-Star International Co., Ltd. GS73 Stealth 8RF/MS-17B7, BIOS E17B7IMS.10F 10/03/2018
[  190.240612] RIP: 0010:rtw_ndev_notifier_call+0x2e/0x5a [88XXau]
[  190.240616] RSP: 0018:ffffa10fc3e87818 EFLAGS: 00010246
[  190.240621] RAX: 0000000000000001 RBX: 000000000000000a RCX: ffffffffc0ac1ef8
[  190.240624] RDX: 0000000000000001 RSI: 000000000000000a RDI: ffff92ae615af000
[  190.240627] RBP: ffffa10fc3e87828 R08: 0000000000000300 R09: 0000000000000000
[  190.240630] R10: ffffa10fc3e87838 R11: 0000000000000040 R12: ffff92ae615af000
[  190.240633] R13: 000000000000000a R14: ffffa10fc3e878b8 R15: 0000000000000000
[  190.240637] FS:  00007f55504648c0(0000) GS:ffff92ae6d240000(0000) knlGS:0000000000000000
[  190.240641] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  190.240644] CR2: 00000000000008c0 CR3: 0000000448d20003 CR4: 00000000003606e0
[  190.240647] Call Trace:
[  190.240657]  notifier_call_chain+0x4c/0x70
[  190.240663]  raw_notifier_call_chain+0x16/0x20
[  190.240670]  call_netdevice_notifiers_info+0x2d/0x60
[  190.240676]  dev_change_name+0x21c/0x300
[  190.240682]  do_setlink+0x4e7/0xd30
[  190.240687]  ? do_setlink+0x4e7/0xd30
[  190.240695]  ? __d_alloc+0x29/0x260
[  190.240701]  ? ___slab_alloc+0x223/0x4e0
[  190.240708]  ? validate_nla+0xb3/0x240
[  190.240714]  ? nla_parse+0xa7/0x120
[  190.240719]  rtnl_setlink+0x102/0x140
[  190.240728]  rtnetlink_rcv_msg+0x207/0x280
[  190.240734]  ? _cond_resched+0x1a/0x50
[  190.240740]  ? __kmalloc_node_track_caller+0x222/0x2d0
[  190.240745]  ? rtnl_calcit.isra.28+0x100/0x100
[  190.240751]  netlink_rcv_skb+0xd9/0x110
[  190.240756]  rtnetlink_rcv+0x15/0x20
[  190.240761]  netlink_unicast+0x198/0x260
[  190.240767]  netlink_sendmsg+0x2e2/0x3d0
[  190.240774]  sock_sendmsg+0x3e/0x50
[  190.240780]  SYSC_sendto+0x101/0x190
[  190.240787]  ? __wake_up+0x13/0x20
[  190.240792]  ? netlink_getsockopt+0x1fa/0x230
[  190.240797]  SyS_sendto+0xe/0x10
[  190.240803]  do_syscall_64+0x73/0x130
[  190.240808]  entry_SYSCALL_64_after_hwframe+0x3d/0xa2
[  190.240813] RIP: 0033:0x7f554f5cbab3
[  190.240816] RSP: 002b:00007fff96197958 EFLAGS: 00000246 ORIG_RAX: 000000000000002c
[  190.240820] RAX: ffffffffffffffda RBX: 00005566404a7550 RCX: 00007f554f5cbab3
[  190.240823] RDX: 0000000000000034 RSI: 000055664047bbe0 RDI: 000000000000000f
[  190.240826] RBP: 0000000000000913 R08: 00007fff96197a10 R09: 0000000000000010
[  190.240829] R10: 0000000000000000 R11: 0000000000000246 R12: 0000000000000004
[  190.240832] R13: 000055664047adb0 R14: 000000000000000e R15: 000055664047b400
[  190.240836] Code: 00 00 48 85 d2 74 4a 55 48 89 e5 41 54 53 4c 8b 22 4d 85 e4 74 31 48 89 f3 4c 89 e7 e8 76 fe ff ff 84 c0 74 22 48 83 fb 0a 75 1c <48> 8b 04 25 c0 08 00 00 48 8d b8 88 44 00 00 ba 10 00 00 00 4c
[  190.240973] RIP: rtw_ndev_notifier_call+0x2e/0x5a [88XXau] RSP: ffffa10fc3e87818
[  190.240976] CR2: 00000000000008c0
[  190.240981] ---[ end trace 157c6b97dfb93d98 ]---
`

kimocoder, 2019-09-08:
First off.. could you update to latest patches?

1. OS
2. KERNEL VERSION
3. DRIVER BRANCH (VERSION)
4. WHICH DEVICE
5. WHICH CHIPSET
6. LOGS, DEBUG, DMESG

🙂

kimocoder, 2019-09-08:
Also.. you may try to disable this one https://github.com/aircrack-ng/rtl8812au/blob/v5.6.4.1/os_dep/linux/os_intfs.c#L1450 by adding "//" in front and compile. report back please

LeeTiK, 2019-09-09:
> Also.. you may try to disable this one https://github.com/aircrack-ng/rtl8812au/blob/v5.6.4.1/os_dep/linux/os_intfs.c#L1450 by adding "//" in front and compile. report back please

unfortunately nothing helped (
Everything remains as well, tell me what other logs can help you?
The make command works fine, everything compiles.
The line you indicated to comment on, did everything, reassembled, but everything remained the same

Just in case I say the driver version 5.6.4 works

1. 16.04.1-Ubuntu x86_64 GNU/Linux
2. 4.15.0-58-generic
3. 5.6.4.1
4. AWUS036ACH
5. RTL 8812AU
6.

- ubuntu@ubuntu-GS73-Stealth-8RF:~$ dkms status
- rtl8812au, 5.6.4.1, 4.15.0-58-generic, x86_64: installed
- ubuntu@ubuntu-GS73-Stealth-8RF:~$ ^C

- [ 2039.698827] usb 2-1: new SuperSpeed USB device number 2 using xhci_hcd
- [ 2039.721013] usb 2-1: New USB device found, idVendor=2109, idProduct=0817
- [ 2039.721019] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
- [ 2039.721023] usb 2-1: Product: USB3.0 Hub
- [ 2039.721027] usb 2-1: Manufacturer: VIA Labs, Inc.
- [ 2039.723352] hub 2-1:1.0: USB hub found
- [ 2039.723689] hub 2-1:1.0: 4 ports detected
- [ 2039.846594] usb 1-5: new high-speed USB device number 6 using xhci_hcd
- [ 2039.996480] usb 1-5: New USB device found, idVendor=2109, idProduct=2817
- [ 2039.996486] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
- [ 2039.996489] usb 1-5: Product: USB2.0 Hub
- [ 2039.996493] usb 1-5: Manufacturer: VIA Labs, Inc.
- [ 2039.997959] hub 1-5:1.0: USB hub found
- [ 2039.998156] hub 1-5:1.0: 4 ports detected
- [ 2040.710607] usb 1-5.1: new high-speed USB device number 7 using xhci_hcd
- [ 2040.859224] usb 1-5.1: New USB device found, idVendor=0bda, idProduct=8812
- [ 2040.859231] usb 1-5.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
- [ 2040.859235] usb 1-5.1: Product: 802.11n NIC
- [ 2040.859239] usb 1-5.1: Manufacturer: Realtek
- [ 2040.859243] usb 1-5.1: SerialNumber: 123456
- [ 2042.122683] PKCS#7 signature not signed with a trusted key
- [ 2042.122708] 88XXau: loading out-of-tree module taints kernel.
- [ 2042.123080] 88XXau: module verification failed: signature and/or required key missing - tainting kernel
- [ 2042.126686] BUG: unable to handle kernel NULL pointer dereference at 0000000000000000
- [ 2042.126688] IP: (null)
- [ 2042.126689] PGD 0 P4D 0
- [ 2042.126691] Oops: 0010 [#1] SMP PTI
- [ 2042.126693] Modules linked in: 88XXau(OE+) ccm xfrm_user xfrm4_tunnel tunnel4 ipcomp xfrm_ipcomp esp4 ah4 af_key xfrm_algo rfcomm bnep nls_iso8859_1 uvcvideo videobuf2_vmalloc btusb btrtl btbcm btintel videobuf2_memops videobuf2_v4l2 videobuf2_core videodev snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic media snd_hda_intel snd_hda_codec arc4 snd_hda_core snd_hwdep iwlmvm snd_pcm bluetooth ecdh_generic mac80211 mei_me mei iwlwifi cfg80211 snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq snd_seq_device joydev msi_wmi intel_wmi_thunderbolt intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel snd_timer kvm irqbypass snd crct10dif_pclmul crc32_pclmul soundcore ghash_clmulni_intel pcbc aesni_intel aes_x86_64 crypto_simd glue_helper cryptd intel_cstate intel_rapl_perf intel_pch_thermal
- [ 2042.126715] input_leds wmi_bmof serio_raw sparse_keymap shpchp acpi_pad mac_hid parport_pc ppdev lp parport autofs4 hid_generic usbhid hid i915 nouveau psmouse mxm_wmi i2c_algo_bit ttm drm_kms_helper alx nvme mdio syscopyarea nvme_core sdhci_pci sysfillrect ahci sysimgblt sdhci fb_sys_fops libahci drm wmi video
- [ 2042.126728] CPU: 5 PID: 3710 Comm: systemd-udevd Tainted: G OE 4.15.0-58-generic #64~16.04.1-Ubuntu
- [ 2042.126729] Hardware name: Micro-Star International Co., Ltd. GS73 Stealth 8RF/MS-17B7, BIOS E17B7IMS.10F 10/03/2018
- [ 2042.126730] RIP: 0010: (null)
- [ 2042.126731] RSP: 0018:ffffb0bac5db7bf0 EFLAGS: 00010202
- [ 2042.126732] RAX: 0000000000000000 RBX: ffffffffc1146890 RCX: 0000000000000000
- [ 2042.126733] RDX: ffffb0bac5db7c00 RSI: 0000000000000005 RDI: ffffffffc1146890
- [ 2042.126734] RBP: ffffb0bac5db7c40 R08: ffff9ac9ad367080 R09: ffff9ac9acc03980
- [ 2042.126735] R10: ffffd385105f4b00 R11: ffffffffb0a5f001 R12: ffffffffc0d62000
- [ 2042.126735] R13: ffffffffb09e5690 R14: ffff9ac9aa1d5000 R15: ffffffffb09e5580
- [ 2042.126737] FS: 00007fc082ff88c0(0000) GS:ffff9ac9ad340000(0000) knlGS:0000000000000000
- [ 2042.126738] CS: 0010 DS: 0000 ES: 0000 CR0: 0000000080050033
- [ 2042.126739] CR2: 0000000000000000 CR3: 000000040726a002 CR4: 00000000003606e0
- [ 2042.126740] Call Trace:
- [ 2042.126744] ? register_netdevice_notifier+0xa7/0x230
- [ 2042.126746] ? 0xffffffffc0d62000
- [ 2042.126768] rtw_ndev_notifier_register+0x15/0x17 [88XXau]
- [ 2042.126778] rtw_drv_entry+0x28/0x1000 [88XXau]
- [ 2042.126781] do_one_initcall+0x55/0x1ac
- [ 2042.126783] ? __vunmap+0x8e/0xc0
- [ 2042.126785] ? _cond_resched+0x1a/0x50
- [ 2042.126787] ? kmem_cache_alloc_trace+0xa6/0x1d0
- [ 2042.126790] do_init_module+0x5f/0x219
- [ 2042.126791] load_module+0x1937/0x1d80
- [ 2042.126794] ? ima_post_read_file+0x83/0xa0
- [ 2042.126796] SYSC_finit_module+0xe5/0x120
- [ 2042.126797] ? SYSC_finit_module+0xe5/0x120
- [ 2042.126799] SyS_finit_module+0xe/0x10
- [ 2042.126801] do_syscall_64+0x73/0x130
- [ 2042.126802] entry_SYSCALL_64_after_hwframe+0x3d/0xa2
- [ 2042.126803] RIP: 0033:0x7fc081e864d9
- [ 2042.126804] RSP: 002b:00007ffe7b74cf68 EFLAGS: 00000202 ORIG_RAX: 0000000000000139
[... cut ...]
- [ 2042.126807] RDX: 0000000000000000 RSI: 00007fc08237ce23 RDI: 0000000000000007
- [ 2042.126807] RBP: 00007fc08237ce23 R08: 0000000000000000 R09: 0000000000000000
- [ 2042.126808] R10: 0000000000000007 R11: 0000000000000202 R12: 0000000000000000
- [ 2042.126809] R13: 0000559ac99a2e70 R14: 0000000000020000 R15: 000000000aba9500
- [ 2042.126810] Code: Bad RIP value.
- [ 2042.126814] RIP: (null) RSP: ffffb0bac5db7bf0
- [ 2042.126814] CR2: 0000000000000000
- [ 2042.126816] ---[ end trace b8445ca562a31e1f ]---

kimocoder, 2019-09-09:
Did you "git pull" i see your using USB3, but I turned off USB3 support due to issues

LeeTiK, 2019-09-09:
> Did you "git pull" i see your using USB3, but I turned off USB3 support due to issues

I tried hub 2.0, the problem remains. Perhaps this is not the case. (All the rest of your driver versions worked)

Which way is it worth looking? I know how to program, though I am new to writing drivers, but I have a great desire to learn

- [   75.240271] usb 1-5: new high-speed USB device number 6 using xhci_hcd
- [   75.388415] usb 1-5: New USB device found, idVendor=1a40, idProduct=0101
- [   75.388416] usb 1-5: New USB device strings: Mfr=0, Product=1, SerialNumber=0
- [   75.388417] usb 1-5: Product: USB 2.0 Hub
- [   75.389082] hub 1-5:1.0: USB hub found
- [   75.389099] hub 1-5:1.0: 4 ports detected
- [   75.775125] usb 1-5.4: new high-speed USB device number 7 using xhci_hcd
- [   75.975059] usb 1-5.4: New USB device found, idVendor=0bda, idProduct=8812
- [   75.975061] usb 1-5.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
- [   75.975062] usb 1-5.4: Product: 802.11n NIC
- [   75.975063] usb 1-5.4: Manufacturer: Realtek
- [   75.975064] usb 1-5.4: SerialNumber: 123456
- [   77.777523] PKCS#7 signature not signed with a trusted key
- [   77.777548] 88XXau: loading out-of-tree module taints kernel.
- [   77.777930] 88XXau: module verification failed: signature and/or required key missing - tainting kernel
- [   77.782564] BUG: unable to handle kernel NULL pointer dereference at 0000000000000000
- [   77.782567] IP:           (null)
- [   77.782568] PGD 0 P4D 0 
- [   77.782570] Oops: 0010 [#1] SMP PTI
- [   77.782572] Modules linked in: 88XXau(OE+) ccm xfrm_user xfrm4_tunnel tunnel4 ipcomp xfrm_ipcomp esp4 ah4 af_key xfrm_algo rfcomm bnep nls_iso8859_1 snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic uvcvideo snd_hda_intel snd_hda_codec videobuf2_vmalloc snd_hda_core arc4 btusb btrtl btbcm btintel videobuf2_memops videobuf2_v4l2 iwlmvm videobuf2_core videodev media snd_hwdep mac80211 snd_pcm snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq snd_seq_device joydev bluetooth ecdh_generic iwlwifi mei_me cfg80211 mei intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel intel_wmi_thunderbolt msi_wmi kvm irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel pcbc aesni_intel aes_x86_64 crypto_simd glue_helper snd_timer cryptd intel_cstate snd input_leds intel_pch_thermal
- [   77.782593]  intel_rapl_perf wmi_bmof serio_raw sparse_keymap soundcore shpchp mac_hid acpi_pad parport_pc ppdev lp parport autofs4 hid_generic usbhid hid nouveau i915 psmouse mxm_wmi ttm i2c_algo_bit drm_kms_helper nvme syscopyarea alx sysfillrect sdhci_pci mdio nvme_core sysimgblt fb_sys_fops sdhci ahci libahci drm wmi video
- [   77.782606] CPU: 9 PID: 2616 Comm: systemd-udevd Tainted: G           OE    4.15.0-58-generic #64~16.04.1-Ubuntu
- [   77.782607] Hardware name: Micro-Star International Co., Ltd. GS73 Stealth 8RF/MS-17B7, BIOS E17B7IMS.10F 10/03/2018
- [   77.782608] RIP: 0010:          (null)
- [   77.782609] RSP: 0018:ffff9912850afbf0 EFLAGS: 00010202
- [   77.782610] RAX: 0000000000000000 RBX: ffffffffc10a9890 RCX: 0000000000000000
- [   77.782611] RDX: ffff9912850afc00 RSI: 0000000000000005 RDI: ffffffffc10a9890
- [   77.782612] RBP: ffff9912850afc40 R08: ffff891f6d467080 R09: ffff891f6cc03980
- [   77.782613] R10: ffff9912850afcb8 R11: 0000000000000000 R12: ffffffffc10df000
- [   77.782614] R13: ffffffffb7fe5690 R14: ffff891f6a207000 R15: ffffffffb7fe5580
- [   77.782615] FS:  00007fc20473e8c0(0000) GS:ffff891f6d440000(0000) knlGS:0000000000000000
- [   77.782616] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
- [   77.782617] CR2: 0000000000000000 CR3: 000000043c4b8003 CR4: 00000000003606e0
- [   77.782618] Call Trace:
- [   77.782622]  ? register_netdevice_notifier+0xa7/0x230
- [   77.782624]  ? 0xffffffffc10df000
- [   77.782645]  rtw_ndev_notifier_register+0x15/0x17 [88XXau]
- [   77.782659]  rtw_drv_entry+0x28/0x1000 [88XXau]
- [   77.782661]  do_one_initcall+0x55/0x1ac
- [   77.782664]  ? __vunmap+0x8e/0xc0
- [   77.782666]  ? _cond_resched+0x1a/0x50
- [   77.782668]  ? kmem_cache_alloc_trace+0xa6/0x1d0
- [   77.782670]  do_init_module+0x5f/0x219
- [   77.782672]  load_module+0x1937/0x1d80
- [   77.782674]  ? ima_post_read_file+0x83/0xa0
- [   77.782676]  SYSC_finit_module+0xe5/0x120
- [   77.782678]  ? SYSC_finit_module+0xe5/0x120
- [   77.782680]  SyS_finit_module+0xe/0x10
- [   77.782681]  do_syscall_64+0x73/0x130
- [   77.782683]  entry_SYSCALL_64_after_hwframe+0x3d/0xa2
- [   77.782684] RIP: 0033:0x7fc2035cc4d9
- [   77.782685] RSP: 002b:00007ffd673c7028 EFLAGS: 00000206 ORIG_RAX: 0000000000000139
- [   77.782686] RAX: ffffffffffffffda RBX: 000055b18ee958e0 RCX: 00007fc2035cc4d9
- [   77.782687] RDX: 0000000000000000 RSI: 00007fc203ac2e23 RDI: 0000000000000007
- [   77.782688] RBP: 00007fc203ac2e23 R08: 0000000000000000 R09: 0000000000000000
- [   77.782689] R10: 0000000000000007 R11: 0000000000000206 R12: 0000000000000000
- [   77.782690] R13: 000055b18eec9720 R14: 0000000000020000 R15: 000000000aba9500
- [   77.782691] Code:  Bad RIP value.
- [   77.782694] RIP:           (null) RSP: ffff9912850afbf0
- [   77.782695] CR2: 0000000000000000
- [   77.782696] ---[ end trace 8dcc9da38969cb66 ]---

kimocoder, 2019-09-09:
Ubuntu want some sort of signature.. do you know what this is, @fariouche ?

fariouche, 2019-09-09:
This is because of trusted boot option that is enabled.
Either disable trusted boot or sign the module.
https://unix.stackexchange.com/questions/455189/pkcs7-signature-not-signed-with-a-trusted-key

amicuchu, 2019-09-09:
Same problems in Gentoo Linux with kernel 5.2.11. Relevant parts of the dmesg a part of the same traceback as previous messages.

aht systemd-networkd[202]: wlan0: Interface name change detected, wlan0 has been renamed to wifi-rtl8812au.
aht kernel: BUG: kernel NULL pointer dereference, address: 0000000000000880
aht kernel: #PF: supervisor read access in kernel mode
aht kernel: #PF: error_code(0x0000) - not-present page
aht kernel: PGD 0 P4D 0 
aht kernel: Oops: 0000 [#1] SMP NOPTI
...........
aht kernel: systemd-udevd (285) used greatest stack depth: 12592 bytes left
aht systemd-udevd[198]: Worker [285] terminated by signal 9 (KILL)
aht systemd-udevd[198]: wlan0: Worker [285] failed

amicuchu, 2019-09-09:
I applied the diff of kimocoder plus patched the functions that register the struct and seems working

TugaPowerROM, 2019-09-22:
> I applied the diff of kimocoder plus patched the functions that register the struct and seems working

having the same thing on Mint 19.2 and kernel 5.3.1 , build and install trough DKMS works just fine, no errors, but syste freeze when assigning name.

@amicuchu can you point me to the diff of @kimocoder and the patched function to apply on my side ?

novasharper, 2019-09-22:
> Also.. you may try to disable this one https://github.com/aircrack-ng/rtl8812au/blob/v5.6.4.1/os_dep/linux/os_intfs.c#L1450 by adding "//" in front and compile. report back please

I think that was the change?

bh, 2019-09-22:
Same as for Archlinux with Linux 5.3.1 and installed via DKMS (https://aur.archlinux.org/packages/rtl88xxau-aircrack-dkms-git/)

amicuchu, 2019-09-24:
That's the patch I applied for make driver works

[disable-ndev-notif.txt](https://github.com/aircrack-ng/rtl8812au/files/3648260/disable-ndev-notif.txt)

TugaPowerROM, 2019-09-24:
> That's the patch I applied for make driver works
> 
> [disable-ndev-notif.txt](https://github.com/aircrack-ng/rtl8812au/files/3648260/disable-ndev-notif.txt)

That is for the hang when system is resolving the name of the device, did fixed that by passing net.ifnames=0 biosdevname=0 over kernel command line inside grub. Altough this approach is better for sure. Woks fine on Mint 19.2 with kernel 5.3.1 witout the grub hack. Having another issue, driver is working, but is not seing all the routers/hotspots. It seems is searching only for 5Ghz wifi. Also Monitor mode cannot be enabled. Beside that driver is working fine. My device is the Netgear A7000 – Nighthawk AC1900 WiFi USB Adapter – USB 3.0, Dual Band. Any tip to make device see all the wifi routers/hotspots or to turn on monitor mode ?

logan001, 2019-09-24:
still download seems to be capped at 100Mbps. with my other driver i can get about 260Mbps.
i have no clue how is the upload. i didnt test it yet. and i think i'll never will. there is no point when driver is so bad

kimocoder, 2019-09-28:
Newer Ubuntu kernels is kinda tricky..

The module signing information may be found below
https://wiki.ubuntu.com/UEFI/SecureBoot/DKMS

kimocoder, 2019-10-30:
We've done some changes. Try to update and report back if any new issues are found. Closing

closed 2019-10-30 by kimocoder