aircrack-ng/rtl8812au #411: Monitor Mode For Both 2.4 & 5?  AWUS1900

bobpiesz, 2019-09-03:
No matter what I try, I'm having issues with monitor mode only working in either 2.4 or 5.  I cannot seem to pin point as to what is causing this.

My read out:

> wlan1     unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=5.22 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=0/100  Signal level=0 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

>

kimocoder, 2019-09-08:
I can.. the 8814AU is a very different driver from the other supported chipsets.. it's old, buggy, dirty..

seeul8er, 2019-09-19:
@bobpiesz What command do you use to set the channel/frequency?
I am using v5.6.4.1 with AWUS1900 and set it up as follows:
```bash
   frequency="2427"
   ip link set "$interface" down
   iw dev "$interface" set type monitor
   ip link set "$interface" up
   iw dev "$interface" set freq $frequency
```

Looking at Wireshark live traces: The adapter always forwards a bunch of packets at once and then pauses for some ~30-60 seconds till some new data arrives. Looking at the channel of the received packets it looks like the adapter is looping through all the frequencies (scan mode?).

[AWUS1900_monitor_mode.pcapng.zip](https://github.com/aircrack-ng/rtl8812au/files/3631174/AWUS1900_monitor_mode.pcapng.zip)

seeul8er, 2019-09-24:
Not sure if is the same issue but running `sudo airmon-ng check kill` killed wpa_supplicant which fixed the channel to what I set it to. Tested on Ubuntu 18.04 Kernel 4.15

kimocoder, 2019-11-11:
@bobpiesz 

Update to v5.6.4.2 branch :+1: 
Patches has been added and bugs has been squashed!

Closing the issue report, but feel free to re-open or open tickets on other issues. Thanks

[image]

closed 2019-11-11 by kimocoder