aircrack-ng/rtl8812au #413: 5.6.4.1 freeze almost everything

logan001, 2019-09-07:
after installing this branch today my wifi adapter stop working together with almost everything.
i couldn't shutdown/reboot computer. terminal open after a few seconds. nano/xed never works. same for iwconfig. and probably more. i didn't try everything.
i had to blacklist 88xxau in order to be able to remove it.

my adapter is Asus USB-AC68 and use 8814 driver

here is what is mostly different in dmesg

> [   11.513974] audit: type=1131 audit(1567868393.547:27): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-rfkill comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   15.109999] usb 1-7: USB disconnect, device number 3
[   15.110682] usbcore: registered new interface driver rtl88xxau
[   15.391852] usb 2-7: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[   15.407722] usb 2-7: Int endpoint with wBytesPerInterval of 512 in config 1 interface 0 altsetting 0 ep 133: setting to 64
[   15.409713] usb 2-7: New USB device found, idVendor=0b05, idProduct=1817, bcdDevice= 0.00
[   15.409714] usb 2-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   15.409715] usb 2-7: Product: 802.11ac NIC
[   15.409716] usb 2-7: Manufacturer: Realtek
[   15.409716] usb 2-7: SerialNumber: 123456
[   17.274492] audit: type=1130 audit(1567868399.309:28): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-rfkill comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   17.274923] rtl88xxau 2-7:1.0 wlp2s0f0u7: renamed from wlan0
[   17.292261] BUG: kernel NULL pointer dereference, address: 00000000000008c0
[   17.292264] #PF: supervisor read access in kernel mode
[   17.292266] #PF: error_code(0x0000) - not-present page
[   17.292267] PGD 0 P4D 0 
[   17.292269] Oops: 0000 [#1] PREEMPT SMP NOPTI
[   17.292271] CPU: 2 PID: 474 Comm: systemd-udevd Tainted: P           OE     5.2.13-27-ck1-tkg-MuQSS #1
[   17.292273] Hardware name: System manufacturer System Product Name/CROSSHAIR VI HERO, BIOS 7403 08/20/2019
[   17.292305] RIP: 0010:rtw_ndev_notifier_call+0x35/0x59 [88XXau]
[   17.292306] Code: 53 48 89 f3 48 8b 2a 48 85 ed 74 16 48 89 ef e8 8f fe ff ff 48 83 fb 0b 75 08 0f b6 c0 66 85 c0 75 08 b8 00 00 00 00 5b 5d c3 <48> 8b 04 25 c0 08 00 00 48 8d b8 48 45 00 00 ba 10 00 00 00 48 89
[   17.292308] RSP: 0018:ffff9c1003ec37c0 EFLAGS: 00010202
[   17.292309] RAX: 0000000000000001 RBX: 000000000000000b RCX: 0000000000000001
[   17.292310] RDX: ffffffffc114b336 RSI: 000000000000000b RDI: ffff8a6ee4aeb000
[   17.292311] RBP: ffff8a6ee4aeb000 R08: 0000000000000000 R09: 0000000000000228
[   17.292312] R10: ffff9c1003ec37d8 R11: ffffffffffffffc0 R12: ffff9c1003ec3818
[   17.292313] R13: 00000000ffffffee R14: ffff8a6ee4aeb010 R15: ffff8a6ee4aeb000
[   17.292314] FS:  00007fd067442840(0000) GS:ffff8a6f0e880000(0000) knlGS:0000000000000000
[   17.292315] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   17.292316] CR2: 00000000000008c0 CR3: 000000080262e000 CR4: 00000000003406e0
[   17.292317] Call Trace:
[   17.292324]  raw_notifier_call_chain+0x46/0x60
[   17.292327]  dev_change_name+0x22b/0x360
[   17.292330]  do_setlink+0xd7f/0xfb0
[   17.292334]  rtnl_setlink+0x15a/0x1d0
[   17.292338]  rtnetlink_rcv_msg+0x164/0x400
[   17.292341]  ? rtnl_calcit.isra.0+0x130/0x130
[   17.292343]  netlink_rcv_skb+0x75/0x150
[   17.292345]  netlink_unicast+0x1ac/0x260
[   17.292347]  netlink_sendmsg+0x202/0x3c0
[   17.292350]  sock_sendmsg+0x4f/0x60
[   17.292352]  __sys_sendto+0x181/0x1c0
[   17.292355]  __x64_sys_sendto+0x25/0x30
[   17.292357]  do_syscall_64+0x5f/0x1d0
[   17.292359]  ? page_fault+0x8/0x30
[   17.292361]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[   17.292363] RIP: 0033:0x7fd068c4164b
[   17.292364] Code: 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 f3 0f 1e fa 48 8d 05 d5 4b 0c 00 41 89 ca 8b 00 85 c0 75 14 b8 2c 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 75 c3 0f 1f 40 00 55 48 83 ec 30 44 89 4c 24
[   17.292365] RSP: 002b:00007fff3c05dca8 EFLAGS: 00000246 ORIG_RAX: 000000000000002c
[   17.292367] RAX: ffffffffffffffda RBX: 00007fff3c05dd14 RCX: 00007fd068c4164b
[   17.292368] RDX: 0000000000000030 RSI: 000055725c33ca00 RDI: 0000000000000006
[   17.292369] RBP: 000055725c344f40 R08: 00007fff3c05dcb0 R09: 0000000000000010
[   17.292369] R10: 0000000000000000 R11: 0000000000000246 R12: 000055725c33c5c0
[   17.292370] R13: 00000000000001da R14: 00000000000001da R15: 000055725c324cf0
[   17.292372] Modules linked in: nvidia_drm(POE) nvidia_modeset(POE) nvidia(POE) nls_iso8859_1 nls_cp437 vfat fat fuse arc4 iwlmvm edac_mce_amd kvm_amd kvm 88XXau(OE) mac80211 btusb drm_kms_helper btrtl btbcm btintel irqbypass bluetooth drm iwlwifi eeepc_wmi asus_wmi sparse_keymap wmi_bmof mxm_wmi igb agpgart crct10dif_pclmul crc32_pclmul mousedev ecdh_generic joydev snd_ctxfi ghash_clmulni_intel ecc input_leds cfg80211 snd_pcm ipmi_devintf snd_timer aesni_intel ipmi_msghandler syscopyarea aes_x86_64 ccp snd r8169 crypto_simd sysfillrect cryptd sysimgblt sp5100_tco glue_helper i2c_algo_bit fb_sys_fops realtek dca i2c_piix4 pcspkr k10temp libphy soundcore rfkill rng_core evdev pinctrl_amd wmi pcc_cpufreq gpio_amdpt mac_hid acpi_cpufreq vboxnetflt(OE) vboxnetadp(OE) vboxpci(OE) vboxdrv(OE) pkcs8_key_parser crypto_user ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 sd_mod uas usb_storage hid_generic usbhid hid ahci libahci libata crc32c_intel xhci_pci xhci_hcd scsi_mod
[   17.292405] CR2: 00000000000008c0
[   17.292406] ---[ end trace 71f4d7bc1b29725a ]---
[   17.292437] RIP: 0010:rtw_ndev_notifier_call+0x35/0x59 [88XXau]
[   17.292438] Code: 53 48 89 f3 48 8b 2a 48 85 ed 74 16 48 89 ef e8 8f fe ff ff 48 83 fb 0b 75 08 0f b6 c0 66 85 c0 75 08 b8 00 00 00 00 5b 5d c3 <48> 8b 04 25 c0 08 00 00 48 8d b8 48 45 00 00 ba 10 00 00 00 48 89
[   17.292440] RSP: 0018:ffff9c1003ec37c0 EFLAGS: 00010202
[   17.292441] RAX: 0000000000000001 RBX: 000000000000000b RCX: 0000000000000001
[   17.292442] RDX: ffffffffc114b336 RSI: 000000000000000b RDI: ffff8a6ee4aeb000
[   17.292443] RBP: ffff8a6ee4aeb000 R08: 0000000000000000 R09: 0000000000000228
[   17.292443] R10: ffff9c1003ec37d8 R11: ffffffffffffffc0 R12: ffff9c1003ec3818
[   17.292444] R13: 00000000ffffffee R14: ffff8a6ee4aeb010 R15: ffff8a6ee4aeb000
[   17.292445] FS:  00007fd067442840(0000) GS:ffff8a6f0e880000(0000) knlGS:0000000000000000
[   17.292446] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   17.292447] CR2: 00000000000008c0 CR3: 000000080262e000 CR4: 00000000003406e0
[   22.280050] audit: type=1131 audit(1567868404.315:29): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-rfkill comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'

novasharper, 2019-09-08:
This looks like it is a duplicate of #409

kimocoder, 2019-09-08:
Dupe. Continue on others issue report

closed 2019-09-08 by kimocoder