aircrack-ng/rtl8812au #416: Recommended version?

botto, 2019-09-18:
Hi, I'm raising this as there it's a little confusing which one is the recommended version to use.  
The confusion comes from this comment https://github.com/aircrack-ng/rtl8812au/issues/399#issuecomment-521881267 that indicates 5.3.4 is not the recommended version and that 5.2.20 should be considered "stable" ?

Why then do the docs indicate 5.3.4 and the default version when landing on the repo 5.3.4?

Thanks

kimocoder, 2019-09-18:
Hei Martin, det kommer av at jeg ikke har vært så veldig aktiv for tiden, v5.3.4 skulle bli standard, men Realtek lanserte 2 kjappe oppdateringer etter v5.3.4 så vi rakk aldri å bli ferdig med den. Nå har jeg rotet bort 8821au chipsetet, så jeg får kontakte Alfa så sender de ny en, så får jeg tested det chipsetet også.

@aircrack-ng could you change default branch to v5.6.4.1 and be sure to enable Travis? Thanks!

kimocoder, 2019-09-18:
v5.2.20(.2) kan ansees som stabil, men den sliter med rekkevidde, det problemet har ikke v5.6.4.1 lengre pga. oppdatert phydm, men ellers er v5.6.4.1 ferdig patchet, sett bort ifra at 8821au (8811) ikke er testet

aircrack-ng, 2019-09-19:
Updated default branch. Looks like Travis is already checking that branch.

kimocoder, 2019-09-19:
Perfect, thanks a lot!

botto, 2019-10-20:
Hei Christian,
takk for forklaringen.

Thank you for updating default branch.

closed 2019-10-20 by botto

kimocoder, 2019-10-31:
@botto bare hyggelig..

Uploaded version 5.6.4.2 today, still needs some work, but working great on 8812au chipset :+1: