aircrack-ng/rtl8812au #417: Virtual interfaces?

jdhau, 2019-09-19:
Hi all,

I guess this is more of a feature request, but I'll provide some context.

Most tools for MitM type attacks require virtual interfaces to work correctly.

I've personally got an rt8812au, and whilst I'm able to create a proto-virtual interface using ifconfig wlan0:ap 192.168.1.1, I'm unable to assign it its own MAC address, and I don't think it's able to perform anything other than the mode that the main interface is on.

`iw phy0 interface add wlan0_ap type managed addr <mac address>`

Fails outright with (-19) no such device, also when I replace phy0 with wlan0.

Is virtual interface support something that's in the works? Asking because it's really a quite essential feature if we want to use this chipset beyond providing password cracking hashes.

kimocoder, 2019-09-19:
I'm sorry, but virtual Interface support is not supported by this as this driver is based on nl80211/cfg80211 and not the newer mac80211 which is required by virtual interfaces.

closed 2019-09-19 by kimocoder

jdhau, 2019-09-19:
Hi,

Thanks for your response.

So since I’m assuming Realtek never developed for mac80211, which in effect
means that the 8812au is useless on its own when it comes to running more
than one mode at once?