aircrack-ng/rtl8812au #418: Need assistance

zurare, 2019-09-19:
So I've spent about 2 hours reading around and trying as many different drivers for the AWUS 1900AC and nothing seems to be working. No matter the driver I install, the card will work, just incapable of being put into monitor mode (yes ik the alternate command to) and it's saying operation not supported -95. Is there anything I can do to fix this? I'd love to be able to use this card but I have no use for it and will return it if I can't get it to monitor/packet injection. I am extremely grateful for any assistance there might be in resolving this issue.
Thank you,
zurare

zurare, 2019-09-19:
Figured this might possibly be important, but I run on 5.2.0-kali2-amd64

zurare, 2019-09-19:
Also every time I try to set the device in monitor with iw dev, it changes to Mode:Auto instead of managed or monitor. Any help is appreciated! -Thanks

0m3rta13, 2019-09-19:
Which branch are you using? 
I’ve had the same device and for the year+ I’ve had it it’s never really worked properly. Its touch and go unfortunately. 
If you have the option to return it, do that rather. Honestly. Because proper driver support is just not there. (from Realtek. Everyone working on it here on Github Are the only reason it even works at all)

Sucks too cos it’s a nice device, when it works.

zurare, 2019-09-20:
Not sure what a branch is, but I ran cat on /etc/*-release, let me know if this is of any use...
DISTRIB_ID=Kali
DISTRIB_RELEASE=kali-rolling
DISTRIB_CODENAME=kali-rolling
DISTRIB_DESCRIPTION="Kali GNU/Linux Rolling"
PRETTY_NAME="Kali GNU/Linux Rolling"
NAME="Kali GNU/Linux"
ID=kali
VERSION="2019.4"
VERSION_ID="2019.4"
VERSION_CODENAME="kali-rolling"
ID_LIKE=debian
ANSI_COLOR="1;31"
HOME_URL="https://www.kali.org/"
SUPPORT_URL="https://forums.kali.org/"
BUG_REPORT_URL="https://bugs.kali.org/"

And damn, I really appreciate the reply, any other 2.4/5GhZ monitor mode devices with similar range you would recommend? Was really looking forward to using this :(
I really appreciate the reply tho,
thanks :)

zurare, 2019-09-20:
I see the AC1200 tends to work well, I just got excited over this card bc it literally has twice the antennas :(

0m3rta13, 2019-09-20:
No worries man. Oh I meant which branch of this driver here in github. Probably safe to assume it’s v5.6.4.1 as that’s now default. I’m going to pull my Alfa out later and ty it out again as I haven’t touched it in a couple updates. 

Which ac1200 device? (There are a few ac1200 devices, it’s a not a model but more of a reference to speed and standard. I’m sure you know or have heard/seen for eg. 802.11a or 802.11n. ac is the latest, more improved standard).  It’s worth reading up on when you have time. 

Most of the common ac1200 capable devices use the rtl8812 which uses this driver too and from what I’ve been reading it works waaay better and seems to only keep improving. @kimocoder uses one himself and says good things. Otherwise get a device with an rtl8821 or rtl8822 as these seem to be the new devices moving forward with realteks new rtw88 driver which looks promising. Hopefully they’ll add support for our devices in time. 

Till then I’m still rocking my TP-Link WN722 v1, an MT7601U and RT3070 USB devices with a Qualcomm Atheros AR9565 as my onboard WiFi in my laptop. 

And for what it’s worth, from what i can tell, our AWUS1900 doesn’t even use all 4 antennas simultaneously at any given time. My WN722 and MT7601 get MUCH better/more stable range with two Cantennas I made up for them. 😆

zurare, 2019-09-21:
Sorry for the late reply, been doing loads of college stuff. But I just purchased/exchanged the AC1900 for the AC1200 with the AWUS036ACH and just tried to install the aircrack-ng driver found online (seems to be the one to install) but it's weird because sometimes it works flawlessly and other times it takes a few tries to get it to work. But my question is, when I remove the driver, it STILL shows the 88XXau driver being used on my card. How do I manually control which driver to use, because a lot of people seem to have luck with the aircrack-ng driver. And I've got the TP-Link one too, but I'd just like something a little stronger, but if I can't get this damn thing to work I'm just gonna return it too and save myself $60 and the hassle xD . Any commands I need to run to choose the driver? It's the same performance/issues when the aircrack-ng one is installed/uninstalled so I'm figuring the driver install is working, just not being touched/used.
Thank you so much man, hopefully this one works lol

kimocoder, 2019-09-21:
Well, whats the hazzle? it's working good over here.. our driver is a straight, fresh Realtek driver, TP-Link is vendor version, running an older version than v5.6.4.1.

Simple as that, but what issues are you having?

kimocoder, 2019-09-21:
And provide "lsusb" & "airmon-ng" output, also "uname -a" and "modinfo 88XXau" output

zurare, 2019-09-22:
lsusb:
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 006: ID 0bda:b009 Realtek Semiconductor Corp. 
Bus 003 Device 005: ID 04f3:250e Elan Microelectronics Corp. USB2.0 Hub
Bus 003 Device 004: ID 04f2:b65d Chicony Electronics Co., Ltd 1351
Bus 003 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 003 Device 002: ID 174c:1351 ASMedia Technology Inc. 1351
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
airmon-ng:
PHY	Interface	Driver		Chipset

phy0	wlan0		rtl8723de	Realtek Semiconductor Co., Ltd. RTL8723DE 802.11b/g/n PCIe Adapter
phy1	wlan1		88XXau		Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
(no issues getting it into monitor mode, just doesn't really show any networks and might take several tries to get it to show any)
Here comes the spam,
modinfo 88XXau:
filename:       /lib/modules/5.2.0-kali2-amd64/updates/88XXau.ko
version:        v5.2.20.2_28373.20180619
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     AA91E592731268BF07B4B3C
alias:          usb:v0846p9054d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v20F4p809Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v20F4p809Ad*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0106d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8813d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA833d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA834d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep400Dd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep400Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0B05p1853d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0B05p1852d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0B05p1817d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p331Ad*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8813d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0120d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0122d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p011Ed*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v3823p6249d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDApA811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep4007d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0411p029Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0846p9052d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2019pAB32d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0411p0242d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep400Fd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep400Ed*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0E66p0023d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3318d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3314d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v04BBp0953d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA813d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA812d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0823d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0BDAp0820d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0BDApA811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8822d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0821d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp881Ad*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2604p0012d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8812d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v148Fp9097d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v050Dp1109d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0411p025Dd*dc*dsc*dp*ic*isc*ip*in*
[... 109 lines cut ...]
parm:           rtw_OffEfuseMask:default open Efuse Mask value:0 (uint)
parm:           rtw_FileMaskEfuse:default drv Mask Efuse value:0 (uint)
parm:           rtw_rxgain_offset_2g:default RF Gain 2G Offset value:0 (uint)
parm:           rtw_rxgain_offset_5gl:default RF Gain 5GL Offset value:0 (uint)
parm:           rtw_rxgain_offset_5gh:uint
parm:           rtw_rxgain_offset_5gm:default RF Gain 5GM Offset value:0 (uint)
parm:           rtw_pll_ref_clk_sel:force pll_ref_clk_sel, 0xF:use autoload value (uint)
parm:           rtw_tx_pwr_idx_override:0-63 int value to force-set all power index values to (int)
parm:           rtw_tx_pwr_by_rate:0:Disable, 1:Enable, 2: Depend on efuse (int)
parm:           rtw_target_tx_pwr_2g_a:2.4G target tx power (unit:dBm) of RF path A for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_b:2.4G target tx power (unit:dBm) of RF path B for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_c:2.4G target tx power (unit:dBm) of RF path C for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_d:2.4G target tx power (unit:dBm) of RF path D for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_5g_a:5G target tx power (unit:dBm) of RF path A for each rate section, should match the real calibrate power, -1: undefined (array of int)
[... cut ...]
parm:           rtw_target_tx_pwr_5g_d:5G target tx power (unit:dBm) of RF path D for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_en_napi:int
parm:           rtw_en_gro:int
parm:           rtw_iqk_fw_offload:int
parm:           rtw_led_enable:Enable status LED (int)

However, all these commands are taking place AFTER the github driver has been uninstalled. It seems I install it and nothing changes.
Thank you for any assistance,
zurare

kimocoder, 2019-09-22:
There you go.. your running driver v5.2.20(.2), not the v5.6.4.1 ..

zurare, 2019-09-22:
So how do I install the correct driver version and get Kali to use the correct one and not the built in one?

kimocoder, 2019-09-22:
"apt-get remove realtek-rtl88xxau-dkms" if you installed with apt-get

Also, a "updatedb" and "locate 88XXau.ko" if not, delete with "rm -rf"

Then follow readme..

"git clone httpd://github.com/aircrack-ng/rtl8812au -b v5.6.4.1"
Then "cd rtl8812au" and "make -j2 && make install"

At last, reload module with "rmmod 88XXau" and "modprobe 88XXau"

zurare, 2019-09-22:
So I followed those exact commands, and I'm still facing the issue of not being able to see any networks :(
Gonna send more readouts to make sure everything went through correctly.
lsusb:
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 006: ID 0bda:b009 Realtek Semiconductor Corp. 
Bus 003 Device 005: ID 04f3:250e Elan Microelectronics Corp. USB2.0 Hub
Bus 003 Device 004: ID 04f2:b65d Chicony Electronics Co., Ltd 1351
Bus 003 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 003 Device 002: ID 174c:1351 ASMedia Technology Inc. 1351
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
airmon-ng:
phy0	wlan0		rtl8723de	Realtek Semiconductor Co., Ltd. RTL8723DE 802.11b/g/n PCIe Adapter
phy1	wlan1		88XXau		Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
modinfo 88XXau:
filename:       /lib/modules/5.2.0-kali2-amd64/updates/88XXau.ko
version:        v5.2.20.2_28373.20180619
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     AA91E592731268BF07B4B3C
alias:          usb:v0846p9054d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v20F4p809Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v20F4p809Ad*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0106d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8813d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA833d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA834d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep400Dd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep400Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0B05p1853d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0B05p1852d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0B05p1817d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p331Ad*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8813d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0120d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0122d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p011Ed*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v3823p6249d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDApA811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep4007d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0411p029Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0846p9052d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2019pAB32d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0411p0242d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep400Fd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep400Ed*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0E66p0023d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3318d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3314d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v04BBp0953d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA813d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA812d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0823d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0BDAp0820d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0BDApA811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8822d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0821d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp881Ad*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2604p0012d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8812d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v148Fp9097d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v050Dp1109d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0411p025Dd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v20F4p805Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0122d*dc*dsc*dp*ic*isc*ip*in*
[... 103 lines cut ...]
parm:           rtw_powertracking_type:default init value:64 (uint)
parm:           rtw_GLNA_type:default init value:0 (uint)
parm:           rtw_TxBBSwing_2G:default init value:0xFF (uint)
parm:           rtw_TxBBSwing_5G:default init value:0xFF (uint)
parm:           rtw_OffEfuseMask:default open Efuse Mask value:0 (uint)
parm:           rtw_FileMaskEfuse:default drv Mask Efuse value:0 (uint)
parm:           rtw_rxgain_offset_2g:default RF Gain 2G Offset value:0 (uint)
parm:           rtw_rxgain_offset_5gl:default RF Gain 5GL Offset value:0 (uint)
parm:           rtw_rxgain_offset_5gh:uint
parm:           rtw_rxgain_offset_5gm:default RF Gain 5GM Offset value:0 (uint)
parm:           rtw_pll_ref_clk_sel:force pll_ref_clk_sel, 0xF:use autoload value (uint)
parm:           rtw_tx_pwr_idx_override:0-63 int value to force-set all power index values to (int)
parm:           rtw_tx_pwr_by_rate:0:Disable, 1:Enable, 2: Depend on efuse (int)
parm:           rtw_target_tx_pwr_2g_a:2.4G target tx power (unit:dBm) of RF path A for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_b:2.4G target tx power (unit:dBm) of RF path B for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_c:2.4G target tx power (unit:dBm) of RF path C for each rate section, should match the real calibrate power, -1: undefined (array of int)
[... cut ...]
parm:           rtw_target_tx_pwr_5g_c:5G target tx power (unit:dBm) of RF path C for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_5g_d:5G target tx power (unit:dBm) of RF path D for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_en_napi:int
parm:           rtw_en_gro:int
parm:           rtw_iqk_fw_offload:int
parm:           rtw_led_enable:Enable status LED (int)

zurare, 2019-09-22:
I now see that even though I followed your exact commands, it's still installed as v5.2.20(.2). What did I do wrong :(

zurare, 2019-09-22:
filename:       /lib/modules/5.2.0-kali2-amd64/kernel/drivers/net/wireless/88XXau.ko
version:        v5.6.4.1_33916.20190619
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     5EBCE397043160009308900
alias:          usb:v0846p9054d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v20F4p809Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v20F4p809Ad*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0106d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8813d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA833d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA834d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep400Dd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep400Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0B05p1853d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0B05p1852d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0B05p1817d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p331Ad*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8813d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p011Fd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0120d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0122d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p011Ed*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v3823p6249d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDApA811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep4007d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0411p029Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0846p9052d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2019pAB32d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0411p0242d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep400Fd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep400Ed*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0E66p0023d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3318d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3314d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v04BBp0953d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pB611d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA813d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA812d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0823d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0BDAp0820d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0BDApA811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8822d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0821d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp881Ad*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2604p0012d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8812d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v148Fp9097d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v050Dp1109d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0411p025Dd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v20F4p805Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0122d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p010Fd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p010Ed*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0115d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p010Dd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0103d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0101d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v13B1p003Fd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3316d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3315d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3313d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p330Ed*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0846p9051d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v07B8p8812d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2019pAB30d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v1740p0100d*dc*dsc*dp*ic*isc*ip*in*
[... 91 lines cut ...]
parm:           rtw_TxBBSwing_2G:default init value:0xFF (uint)
parm:           rtw_TxBBSwing_5G:default init value:0xFF (uint)
parm:           rtw_OffEfuseMask:default open Efuse Mask value:0 (uint)
parm:           rtw_FileMaskEfuse:default drv Mask Efuse value:0 (uint)
parm:           rtw_rxgain_offset_2g:default RF Gain 2G Offset value:0 (uint)
parm:           rtw_rxgain_offset_5gl:default RF Gain 5GL Offset value:0 (uint)
parm:           rtw_rxgain_offset_5gh:uint
parm:           rtw_rxgain_offset_5gm:default RF Gain 5GM Offset value:0 (uint)
parm:           rtw_pll_ref_clk_sel:force pll_ref_clk_sel, 0xF:use autoload value (uint)
parm:           rtw_tx_pwr_by_rate:0:Disable, 1:Enable, 2: Depend on efuse (int)
parm:           rtw_tx_pwr_lmt_enable:0:Disable, 1:Enable, 2: Depend on efuse (int)
parm:           rtw_target_tx_pwr_2g_a:2.4G target tx power (unit:dBm) of RF path A for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_b:2.4G target tx power (unit:dBm) of RF path B for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_c:2.4G target tx power (unit:dBm) of RF path C for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_d:2.4G target tx power (unit:dBm) of RF path D for each rate section, should match the real calibrate power, -1: undefined (array of int)
[... cut ...]
parm:           rtw_target_tx_pwr_5g_d:5G target tx power (unit:dBm) of RF path D for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_tsf_update_pause_factor:num of bcn intervals to stay TSF update pause status (int)
parm:           rtw_tsf_update_restore_factor:num of bcn intervals to stay TSF update restore status (int)
parm:           rtw_en_napi:int
parm:           rtw_en_gro:int
parm:           rtw_iqk_fw_offload:int
parm:           rtw_ch_switch_offload:int

zurare, 2019-09-22:
So after a couple reinstalls of the driver everything seemed to finally be installed properly. Now as long as I have NetworkManager killed (already tried that before lol) it seems to be picking up networks properly. Have yet to try packet injection, just hoping that the card is performing as it should and hopefully not just good luck having it work for a couple minutes now lol. I'll give you an update and I'm really really appreciative of your guys' help. Thank you so much! :)

wice222, 2019-10-01:
Can this work on kernel 5.2.0?
I'm having this error:

root@kali:/rpms/rtl8812au# ./dkms-install.sh
About to run dkms install steps...

Creating symlink /var/lib/dkms/rtl8812au/5.6.4.1/source ->
                 /usr/src/rtl8812au-5.6.4.1

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.2.0-kali2-amd64 KSRC=/lib/modules/5.2.0-kali2-amd64/build.............
cleaning build area...

DKMS: build completed.

88XXau.ko:
Running module version sanity check.
**modinfo: ERROR: missing module or filename.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.2.0-kali2-amd64/updates/**

depmod...

DKMS: install completed.
Finished running dkms install steps.

kimocoder, 2019-10-01:
Could not get much information from that, let me check when I'm back home

wice222, 2019-10-01:
Thanks for reply

Finally getting this: 
root@kali:/rpms/rtl8812au# iwconfig
wlan0     unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Monitor  Frequency=2.427 GHz  Access Point: Not-Associated
          Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=0/100  Signal level=0 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

lo        no wireless extensions.

eth0      no wireless extensions.
==========================
**airodump-ng wlan0**  brings Zero resoult
When AR9271 802.11n connected to the same machine gives dozens of resoults...

kimocoder, 2019-10-01:
@wice222 what chipset are you using?

wice222, 2019-10-01:
root@kali:~# lsusb
Bus 003 Device 008: ID 0bda:8813 Realtek Semiconductor Corp. RTL8814AU 802.11a/b/g/n/ac Wireless Adapter

kimocoder, 2019-11-11:
@zurare @wice222 @0m3rta13 

You should use the v5.6.4.2 branch, added lots of patches to it, Realtek also fixed some issues :+1: 
Tested working, as seen below..

[image]

kimocoder, 2019-11-11:
Problem resolved in branch v5.6.4.2. If any more issues should appear, open a new report with detailed info to work with. Thanks! :+1:

closed 2019-11-11 by kimocoder

MarkCowley1, 2019-11-13:
So I am having a similar problem as @wice222 above.
[image]
[image]
I tried uninstalling the driver and cloning "git clone httpd://github.com/aircrack-ng/rtl8812au -b v5.6.4.1"
Then "cd rtl8812au" and "make -j2 && make install" to try and get it to work but I am stil having the same issue where I get a message saying the device is busy with a previous request.
The model I am using is Dual Band USB Adapter 1200(866+300)Mbps Wireless speed with 802.11ac)

kimocoder, 2019-11-13:
@MarkCowley1 and why are you using the buggy v5.6.4.1, when it's the v5.6.4.2 that's in active development?

MarkCowley1, 2019-11-14:
@kimocoder yes I see that now.

karthikamaravadi, 2019-12-06:
Hello @kimocoder. I tried doing the samething you said above and still not able to discover or perform any analysis. Can you please let me know ur thoughts on this one.
[image]

creited, 2020-03-05:
> Hello @kimocoder. I tried doing the samething you said above and still not able to discover or perform any analysis. Can you please let me know ur thoughts on this one.
> [image]

Try to use: 

wifite --kill

That solved my issue. Network-manager process and wifite for 8814au will not work together in my case.

If you need network manager for any other reason (eth0 for example), you can do the following:

edit /etc/NetworkManager/NetworkManager.conf and add the following:

[keyfile]
unmanaged-devices=interface-name:eth*,except:interface-name:eth0;interface-name:wlan*

That should make wifite work without having to kill network-manager.