aircrack-ng/rtl8812au #419: Installation fails under Linux Mint 19, kernel 4.15

cristianav, 2019-09-21:
Hello,

First of all, many thanks for all your work, much appreciated.

Now, I'm trying to install the drivers but I'm stopped at this step:

```
CC [M]  /var/lib/dkms/rtl8812au/5.6.4.1/build/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.1/build/os_dep/linux/wifi_regd.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.1/build/os_dep/linux/rtw_android.o
/var/lib/dkms/rtl8812au/5.6.4.1/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_indicate_sta_assoc’:
/var/lib/dkms/rtl8812au/5.6.4.1/build/os_dep/linux/ioctl_cfg80211.c:4484:3: error: implicit declaration of function ‘cfg80211_sinfo_alloc_tid_stats’; did you mean ‘cfg80211_inform_bss_width’? [-Werror=implicit-function-declaration]
   cfg80211_sinfo_alloc_tid_stats(&sinfo, GFP_KERNEL);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   cfg80211_inform_bss_width
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.1/build/os_dep/linux/rtw_rhashtable.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.1/build/os_dep/linux/ioctl_mp.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.1/build/hal/hal_intf.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.1/build/hal/hal_com.o
/var/lib/dkms/rtl8812au/5.6.4.1/build/os_dep/linux/ioctl_cfg80211.c: At top level:
cc1: warning: unrecognized command line option ‘-Wno-cast-function-type’
cc1: warning: unrecognized command line option ‘-Wno-cast-function-type’
cc1: some warnings being treated as errors
scripts/Makefile.build:332: recipe for target '/var/lib/dkms/rtl8812au/5.6.4.1/build/os_dep/linux/ioctl_cfg80211.o' failed
make[2]: *** [/var/lib/dkms/rtl8812au/5.6.4.1/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[2]: *** Waiting for unfinished jobs....
Makefile:1552: recipe for target '_module_/var/lib/dkms/rtl8812au/5.6.4.1/build' failed
make[1]: *** [_module_/var/lib/dkms/rtl8812au/5.6.4.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-20-generic'
Makefile:1606: recipe for target 'modules' failed
make: *** [modules] Error 2
```

Linux Mint 19 4.15.0-20-generic #21-Ubuntu SMP x86_64 x86_64 x86_64 GNU/Linux

Any idea about this?
Thanks

kimocoder, 2019-09-21:
Thanks, did you try using "make && make install" instead of DKMS ?

cristianav, 2019-09-21:
Yes, both ways but the same result. 
Is maybe the compiler too old or something like that?

Thanks for your reply.

⁣Get BlueMail for Android ​

kimocoder, 2019-09-21:
@CGarces may you add some Linux Mint checks to Travis? Added you to README.md under credits btw 🥇

kimocoder, 2019-09-21:
I'll fix 1 issue shortly, then Mint check should/will be added to Travis CI for watching. I'll notify you when I push a minor compiler fix

CGarces, 2019-09-22:
"Mint check"?

I can add checks for some scenarios, but not for one specific distribution. Please point me to the commit that fix this issue and I'll try to understand the problem.

kimocoder, 2019-09-22:
I have no idea on Travis 😅 I'll boot up a virtual Mint, it's Ubuntu based so it can't be that serious, just don't have more time right away

romainreignier, 2019-10-01:
I had the exact same error while building this module on a 4.14 kernel.
It appears that the function `cfg80211_sinfo_alloc_tid_stats()` was introduced in 4.18 with [this commit](https://github.com/torvalds/linux/commit/8689c051a20195b228e19acb155c7d6e48a86753).
The v5.3.4 is building on this kernel.

PR #420 (2019-10-22): Fix compilation issue kernels < 4.18 (Ubuntu 18.04 and derived distros)
As requested on #419 

Linux Mint 19 use a EOL kernel (like Ubuntu18.04)
I have made a new Travis build using similar kernel and gcc compiler.
 The fix for the issue just remove the function cfg80211_sinfo_alloc_tid_stats, introduced at https://github.com/torvalds/linux/commit/8689c051a20195b228e19acb155c7d6e48a86753#diff-35454f3f0395271a0e34581e160bea5f

Fix  #419 
Fix  #436
closed 2019-10-22 by kimocoder