aircrack-ng/rtl8812au #421: [v5.6.4.1][linux-5.3.1]: WARNING: CPU: 1 PID: 42 at net/wireless/core.c:868 wiphy_register.cold+0x37/0x435 [cfg80211]

vedantk, 2019-09-22:
Hello,

The v5.6.4.1 88XXau driver appears to be crashing while attempting to register an 8811 device (marketed as the "EDUP 802.11ac USB WiFi Dongle", serial = 00e04c000001). This is on an Arch Linux kernel (version 5.3.1). I installed the driver using dkms-install.sh.

```
WARNING: CPU: 1 PID: 42 at net/wireless/core.c:868 wiphy_register.cold+0x37/0x435 [cfg80211]
...
[  511.028685] Call Trace:                                                      
[  511.028701]  ? preempt_count_add+0x68/0xa0                                   
[  511.028803]  rtw_wiphy_register+0x16/0x18 [88XXau]                           
[  511.028901]  rtw_cfg80211_ndev_res_register+0x11/0x17 [88XXau]               
[  511.028992]  rtw_os_ndev_register+0xe3/0x14c [88XXau]                        
[  511.029085]  rtw_os_ndevs_register+0x9a/0xfb [88XXau]                        
[  511.029176]  rtw_os_ndevs_init+0x24/0x38 [88XXau]                            
[  511.029266]  rtw_drv_init+0x2b3/0x376 [88XXau]                               
[  511.029275]  usb_probe_interface+0xe0/0x280 
```

Additional dmesg output attached -- there are two more (apparently related) call traces:

```
WARNING: CPU: 1 PID: 42 at /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/os_intfs.c:1605 rtw_os_ndev_register+0xaa/0x14c [88XXau]
...
WARNING: CPU: 1 PID: 42 at /var/lib/dkms/rtl8812au/5.3.4/build/os_dep/linux/os_intfs.c:3039 rtw_os_ndevs_register+0xd1/0xfb [88XXau]
```

[latest-dmesg.txt](https://github.com/aircrack-ng/rtl8812au/files/3640281/latest-dmesg.txt)

Thanks for taking a look.

pryre, 2019-09-23:
I'm having the same issue I think. I've attached the matching logs if that helps (unplugged and replugged the USB for the second fault).

```
[    5.974456] WARNING: CPU: 1 PID: 462 at /var/lib/dkms/rtl88xxau/r672.2c3ce70/build/os_dep/linux/os_intfs.c:1605 rtw_os_ndev_register+0xaa/0x14c [88XXau]
[    5.998317] WARNING: CPU: 8 PID: 462 at /var/lib/dkms/rtl88xxau/r672.2c3ce70/build/os_dep/linux/os_intfs.c:3039 rtw_os_ndevs_register+0xd1/0xfb [88XXau]
[    6.268307] usbcore: registered new interface driver rtl88xxau
...
[ 3706.286998] WARNING: CPU: 8 PID: 5643 at /var/lib/dkms/rtl88xxau/r672.2c3ce70/build/os_dep/linux/os_intfs.c:1605 rtw_os_ndev_register+0xaa/0x14c [88XXau]
[ 3706.318096] WARNING: CPU: 8 PID: 5643 at /var/lib/dkms/rtl88xxau/r672.2c3ce70/build/os_dep/linux/os_intfs.c:3039 rtw_os_ndevs_register+0xd1/0xfb [88XXau]
```

[latest-dmesg-pryre.txt](https://github.com/aircrack-ng/rtl8812au/files/3644363/latest-dmesg-pryre.txt)

christosvar, 2019-09-24:
Same issue here. I got the same warnings with both `v5.6.4.1` and `v5.3.4`, in kernel version `v5.3.1` in Antergos/Arch. And finally the driver crashes.

I am happy to help with any logs etc (even though the messages seem identical to the above).

luzat, 2019-09-25:
Can confirm with custom kernel 5.3.0 and v5.2.20 on Debian.

emptysoda, 2019-09-26:
Same issue here with kernel 5.3.1 on arch.

haritonovsb, 2019-10-01:
Can confirm:
linux 5.3.1 on archlinux (current latest available)

christosvar, 2019-10-07:
I can confirm, this issue is resolved

nemasu, 2019-10-17:
I'm having this issue with custom kernel 5.3.6. Driver crashes.

kimocoder, 2019-10-22:
Has this issue been resolved or not? I've got time to look into it..

ValZapod, 2019-10-23:
As said in there https://github.com/aircrack-ng/rtl8812au/pull/424 solved it.

nemasu, 2019-10-23:
I was still having problems after the fix, but ... it just started working, so, not sure what to think anymore.

kimocoder, 2019-11-09:
Move over to v5.6.4.2 instead of v5.6.4.1

PR #432 (2019-10-30): VENDOR_CWD_RAW_DATA policy for kernel >= 5.3.0
Resolves #421
closed 2019-11-09 by kimocoder