aircrack-ng/rtl8812au #422: No scan results

marcingrabda, 2019-09-24:
Hi,

After building and installing the driver, I'm not able to find any networks:
```
$ sudo iwlist wlan0 scan
wlan0     No scan results
```

I'm running Raspbian Buster, kernel: 4.19.66-v7+.
Device: Edimax EW-7811UTC AC600.

Any ideas what's wrong?

Dunklas, 2019-09-30:
I'm experiencing the same problem.

[image]

OS: Arch Linux
Kernel: 5.3.1.
Device: NetGear, Inc. A6100 AC600 DB Wireless Adapter [Realtek RTL8811AU]

kimocoder, 2019-09-30:
Yes, the 8811 (8821au) still got issues.. can't seem to find my own, so I can't debug right away..

marcingrabda, 2019-09-30:
Switching to an older version helped in my case.

kimocoder, 2019-09-30:
Yes, we're aware of the issue . Hopefully will be fixed soon even though I don't have time right away :100:

Dunklas, 2019-09-30:
Thanks for the super quick response! I'll downgrade for the time being.

sumandroid12, 2019-10-01:
I have a device with rtl8811au chipset and I'm on 5.3.1-arch1-1-ARCH kernel. Is it possible to have it working by downgrading to an earlier version? I have tried but couldn't get it to work.

kimocoder, 2019-10-01:
Yes, go down to v5.2.20, that works 🙂

sumandroid12, 2019-10-01:
I have installed v5.2.20 but the device isn't detected. Below is the output from dmesg after inserting the wifi dongle. Need some help with this :face_with_head_bandage:. I think this has happened because of the other modules i tried installing, They may not have been fully removed. should I delete my computer?
<details>
<summary>dmesg output</summary>

```
[  734.584825] WARNING: CPU: 0 PID: 15 at /var/lib/dkms/rtl8812au/5.2.20.2/build/os_dep/linux/os_intfs.c:2977 rtw_os_ndevs_register+0xd1/0xfb [88XXau]
[  734.584827] Modules linked in: ccm algif_aead des_generic arc4 libarc4 algif_skcipher bnep cmac md4 algif_hash af_alg nls_iso8859_1 nls_cp437 vfat fat nvidia_drm(POE) nvidia_modeset(POE) nvidia(POE) intel_rapl_msr intel_rapl_common x86_
pkg_temp_thermal intel_powerclamp coretemp 88XXau(OE) kvm_intel snd_hda_codec_hdmi mei_hdcp ofpart kvm cmdlinepart intel_spi_platform intel_spi spi_nor joydev mtd iTCO_wdt iTCO_vendor_support hp_wmi sparse_keymap wmi_bmof cfg80211 i915 irq
bypass ses enclosure scsi_transport_sas rndis_host cdc_ether usbnet crct10dif_pclmul crc32_pclmul snd_usb_audio ghash_clmulni_intel mii snd_usbmidi_lib i2c_algo_bit snd_rawmidi snd_hda_codec_realtek snd_seq_device drm_kms_helper snd_hda_co
dec_generic ledtrig_audio aesni_intel btusb snd
_hda_intel drm aes_x86_64 btrtl crypto_simd snd_hda_codec btbcm cryptd btintel glue_helper intel_gtt intel_cstate ipmi_devintf snd_hda_core agpgart ipmi_msghandler bluetooth intel_uncore uvcvideo snd_hwdep videobuf2_vmalloc syscopyarea psmouse
[  734.584890]  snd_pcm videobuf2_memops sysfillrect pcspkr intel_rapl_perf r8169 i2c_i801 mei_me snd_timer videobuf2_v4l2 sysimgblt videobuf2_common ecdh_generic rfkill snd realtek ecc input_leds rtsx_pci_ms mousedev mei videodev libphy memstick lpc_ich fb_sys_fops soundcore hp_accel lis3lv02d mc input_polldev wmi hp_wireless evdev battery mac_hid intel_smartconnect acpi_tad ac pkcs8_key_parser crypto_user ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 uas usb_storage hid_generic usbhid hid sd_mod rtsx_pci_sdmmc serio_raw mmc_core ahci atkbd libahci libps2 libata xhci_pci xhci_hcd scsi_mod crc32c_intel ehci_pci ehci_hcd rtsx_pci i8042 serio
[  734.584948] CPU: 0 PID: 15 Comm: kworker/0:1 Tainted: P        W  OE     5.3.1-arch1-1-ARCH #1
[  734.584950] Hardware name: Hewlett-Packard HP Pavilion 15 Notebook PC /2281, BIOS F.11 08/07/2014
[  734.584960] Workqueue: usb_hub_wq hub_event
[  734.585032] RIP: 0010:rtw_os_ndevs_register+0xd1/0xfb [88XXau]
[  734.585036] Code: 83 58 01 00 00 39 e8 7f b2 e9 7a ff ff ff 48 c7 c7 e8 c4 21 c1 e8 84 18 16 df 0f 0b eb dd 48 c7 c7 e8 c4 21 c1 e8 74 18 16 df <0f> 0b 85 ed 74 0d 48 89 df e8 69 33 01 00 e9 4e ff ff ff 48 63 ed
[  734.585038] RSP: 0018:ffffb60fc00c3968 EFLAGS: 00010246
[  734.585041] RAX: 0000000000000024 RBX: ffffa3db354e0000 RCX: 0000000000000000
[  734.585043] RDX: 0000000000000000 RSI: ffffa3db57017708 RDI: 00000000ffffffff
[  734.585045] RBP: 0000000000000000 R08: 00000000000004e7 R09: 0000000000000003
[  734.585046] R10: 0000000000000000 R11: 0000000000000001 R12: 0000000000000001
[  734.585048] R13: 0000000000000000 R14: ffffb60fc0523000 R15: ffffffffc1220a4f
[  734.585051] FS:  0000000000000000(0000) GS:ffffa3db57000000(0000) knlGS:0000000000000000
[  734.585053] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  734.585055] CR2: 00007f752e1bb270 CR3: 0000000190c0a004 CR4: 00000000001606f0
[  734.585058] Call Trace:
[  734.585134]  rtw_os_ndevs_init+0x24/0x38 [88XXau]
[  734.585204]  rtw_drv_init+0x2b3/0x376 [88XXau]
[  734.585211]  usb_probe_interface+0xe0/0x280
[  734.585217]  really_probe+0xf0/0x380
[  734.585222]  driver_probe_device+0xb6/0x100
[  734.585226]  ? driver_allows_async_probing+0x50/0x50
[  734.585229]  bus_for_each_drv+0x8c/0xd0
[  734.585234]  __device_attach+0xf0/0x170
[  734.585238]  bus_probe_device+0x8e/0xa0
[  734.585244]  device_add+0x3cd/0x6b0
[  734.585249]  ? sysfs_merge_group+0xb4/0x110
[  734.585253]  usb_set_configuration+0x67e/0x7e0
[  734.585261]  generic_probe+0x3c/0x57
[  734.585264]  really_probe+0xf0/0x380
[  734.585268]  driver_probe_device+0xb6/0x100
[  734.585272]  ? driver_allows_async_probing+0x50/0x50
[  734.585274]  bus_for_each_drv+0x8c/0xd0
[  734.585278]  __device_attach+0xf0/0x170
[  734.585282]  bus_probe_device+0x8e/0xa0
[  734.585286]  device_add+0x3cd/0x6b0
[  734.585292]  usb_new_device.cold+0x120/0x2e3
[  734.585297]  hub_event+0xbb2/0x15a0
[  734.585307]  process_one_work+0x1d1/0x3a0
[  734.585312]  worker_thread+0x4a/0x3d0
[  734.585316]  kthread+0xfb/0x130
[  734.585320]  ? process_one_work+0x3a0/0x3a0
[  734.585323]  ? kthread_park+0x80/0x80
[  734.585330]  ret_from_fork+0x1f/0x40
[  734.585336] ---[ end trace 05938d215e10f080 ]---
[  734.708309] usb 2-1: reset high-speed USB device number 9 using xhci_hcd
```
</details>

TugaPowerROM, 2019-10-01:
Dont think its the case because for me the same happens when use 5.2.20 branch.
Adaptor its not detected.

kimocoder, 2019-10-01:
@TugaPowerROM I need "lsusb" output..

Btw, I remember that v5.2.20 branch need some newer kernel patches!
Forgot about that this weekend, I will add it in a couple of days

xuv, 2019-10-04:
Same issue. Have also tried previous versions of the driver down to 5.2.20 with the same issue.

System: Ubuntu 18.04 LTS 
Kernel: 5.0.0-23-generic
Device: TP-LINK Archer T2U Plus

TugaPowerROM, 2019-10-05:
> @TugaPowerROM I need "lsusb" output..
> 
> Btw, I remember that v5.2.20 branch need some newer kernel patches!
> Forgot about that this weekend, I will add it in a couple of days

Latest changes on 5.2.20 indeed make the driver works with kernel 5.3.1, altough it find alot less acess points then actual 5.6.4.1 ( it find the ones the driver 5.6.4.1 cannot find )

Led its not blinking, tried to force with modprob.d config file, led keeps off all the time, with  /proc/net/rtl8812au/$(your interface name)/led_enable also does not work ( does not find the file/folder )

Using wpa_supplicant after connection is made after 1/2 seconds the connection its broke.

lsusb:  Bus 004 Device 075: ID 0846:9054 NetGear, Inc.

christosvar, 2019-10-07:
Same case here, no scan results with either 5.6.4.1 or 5.2.20 on Arch kernel 5.3.1.

dbartz, 2019-10-08:
Same case here, no scan results with either 5.6.4.1 or 5.2.20 on Ubuntu 18.04.3.

Device:
TP-LINK Archer T2U v3 [Realtek RTL8811AU]

Revitspace, 2019-10-09:
Same here, device recognised and flashes, however no wifi results when searching (using v.5.6.4.1).

Tried going back to 5.2.20 and device is not recognised at all.

**System:**
Kali Linux 2019.3 (64bit)
**Device:**
TP-LINK Archer T2U

kimocoder, 2019-10-09:
Well guys, v5.2.20 needs som kernel support patches.
Don't have time right away, but I'll give an update here

bl4ckOut, 2019-10-10:
> 
> 
> Switching to an older version helped in my case.

Which version do you use ?

marcingrabda, 2019-10-10:
> > Switching to an older version helped in my case.
> 
> Which version do you use ?

5.3.4 or 5.2.20, but I would have to check at home.

ovlx, 2019-10-11:
applying the changes in this commit https://github.com/gordboy/rtl8812au/commit/30d47a0a3f43ccb19e8fd59fe93d74a955147bf2 to the 5.2.20 branch fixes support for 5.3 kernel

CGarces, 2019-10-11:
> applying the changes in this commit [gordboy/rtl8812au@30d47a0](https://github.com/gordboy/rtl8812au/commit/30d47a0a3f43ccb19e8fd59fe93d74a955147bf2) to the 5.2.20 branch fixes support for 5.3 kernel

But that was allready done at #430 ...

dbartz, 2019-10-11:
> applying the changes in this commit gordboy/rtl8812au@30d47a0 to the 5.2.20 branch fixes support for 5.3 kernel

At least for me the issue is not related to Linux kernel 5.3, as I am using Ubuntu 18.04.03 which runs a 5.0 kernel.

ovlx, 2019-10-11:
> > applying the changes in this commit [gordboy/rtl8812au@30d47a0](https://github.com/gordboy/rtl8812au/commit/30d47a0a3f43ccb19e8fd59fe93d74a955147bf2) to the 5.2.20 branch fixes support for 5.3 kernel
> 
> But that was allready done at #430 ...
 my mistake my local git did not pull that commit for some reason,

TugaPowerROM, 2019-10-11:
For me 5.2.20 its not finding all the acess points, led does not blink, connection its not stable.
The most stable connection/speed wise is 5.6.4.1 , the only issue its missing acess points, with 5.2.20 it finds the acess points that 5.6.4.1 does not find.

cabez0n, 2019-10-15:
Same here. No scan results. Tested both 5.6.4.1 or 5.2.20.2 with the patches (else it crashes)
and still doesn't work. 

Ubuntu 19.10 
5.3.0-17-generic
TP-LINK Archer T2U PLUS

cabez0n, 2019-10-15:
Possible work around:

I had No scan results. Tested both 5.6.4.1 or 5.2.20.2 with the patches (else it crashes)
and still didn't work. 
Ubuntu 19.10 
5.3.0-17-generic
TP-LINK Archer T2U PLUS

Tested a few options in /etc/modprobe.d/88XXau.conf and the ones below seem to make it return scan results, and I can connect to 2.4 GHz networks. It didn't work for 5 GHz aps. 

options 88XXau rtw_power_mgnt=0 rtw_enusbss=0 rtw_ips_mode=1 rtw_vht_enable=2 \
rtw_ht_enable=1 rtw_led_enable=1 rtw_full_ch_in_p2p_handshake=1

christosvar, 2019-10-15:
Update: v5.2.20 (99ee31b) worked for me! Very few results while scanning (yes 5 GHz wifi is missing), but it worked! THANKS for the hard work!

CGarces, 2019-10-16:
> Possible work around:
> 
> I had No scan results. Tested both 5.6.4.1 or 5.2.20.2 with the patches (else it crashes)

Hi @cabez0n , what means "with the patches", which patch is missing?

Joermungand, 2019-10-17:
‘Good’ news, guys: v5.2.20 (rtl8812) *used to* work with kernel 5.3.5 (both plain Arch *and* Manjaro). No more scan results after kernel update to 5.3.6 (in either distro).

llccd, 2019-10-18:
I reverted back to commit c215cafcaa7731cd79e5236f4c85a3ddfb5e6355 and scan worked.

TugaPowerROM, 2019-10-19:
> Possible work around:
>
> I had No scan results. Tested both 5.6.4.1 or 5.2.20.2 with the patches (else it crashes)
> and still didn't work.
> Ubuntu 19.10
> 5.3.0-17-generic
> TP-LINK Archer T2U PLUS
>
> Tested [...]

for me it wont find alot of acess points on 2.4Ghz using v5.6.4.1 , also the 88XXau.conf seems it have no effect because tried to disabled the led and it wont disable. Using 5.2.20 it find the missing acess points but driver is not stable on 5.3.x kernel. Where you saw 5.2.20.2 version ?

marcingrabda, 2019-10-20:
v5.2.20 works fine on Ubuntu 19.10, kernel 5.3.0-1007-raspi2.

Kasami-s, 2019-10-22:
Seem to be having same issue as the guys above.
Latest v5.6.4.1 - list of networks is always empty
Latest v5.2.20 - no interface

reverting to commit c215caf mentioned earlier did not help. 

Kernel: 5.3.7-arch1-1-ARCH (x86_64)
Chipset: RTL8811AU

Andy-Chen-740629, 2019-10-24:
I also get the same issue,no scan results

ubuntu 18.04 LTS
kernel 4.15
driver 5.6.4.1
TP-Link T2U Nano

Which version driver is correct ?

thanks!

Andy-Chen-740629, 2019-10-24:
I download Edimax EW7811 rtl8811au Vendor driver
and compiler build rtl8811au.ko file

scan function is normal

I think your src code have some issue
-:)

CGarces, 2019-10-24:
> I download Edimax EW7811 rtl8811au Vendor driver
> and compiler build rtl8811au.ko file
> 
> scan function is normal
> 
> I think your src code have some issue
> -:)

Hi @Andy-Chen-740629 
Please execute lsusb to check you device id. A quick search on google give me that your device is 7392:7811

`7392:7811 Edimax Technology Co., Ltd EW-7811Un 802.11n Wireless Adapter`

This is not the correct driver, looks like you must use rtl8192
https://github.com/pvaret/rtl8192cu-fixes/blob/master/os_dep/linux/usb_intf.c#L98

Andy-Chen-740629, 2019-10-24:
update
EW-7811UTC base rtl8811au chipset
you can download src code from EDIMAX official web site

My TP-link T2U nano use rtl8811au chipset 
The VID:PID = 2357:011E

It need add above ID on os_dep/linux/usb_intf.c file 

and disable BT Function
modified the Makefile file
CONFIG_BT_COEXIST = n
the run Make 
you can get rtl8811au.ko driver

insmod rtl8811au.ko 

iwlist wlan0 scan
you can see the AP SSID information

Andy-Chen-740629, 2019-10-24:
EW-7811UTC_UAC_DAC_USC_Linux_Driver_1.0.2.4.zip

This is driver name !!

6bytes, 2019-10-25:
> update
> EW-7811UTC base rtl8811au chipset
> you can download src code from EDIMAX official web site
>
> My TP-link T2U nano use rtl8811au chipset
> The VID:PID = 2357:011E
>
> It need add above ID on [...]

I have the same tp-link adapter with the same VID:PID and I'm having the same issue as others where scan give no results.

@Andy-Chen-740629 can you please share more details on your fix? I downloaded the EDIMAX src code (EW-7811UTC_GPL_1.0.1.6.zip) and driver (EW-7811UTC_UAC_DAC_USC_Linux_Driver_1.0.2.4.zip) but **make** throws a bunch of erros

I'm on kernel fedora kernel 5.3.6

Andy-Chen-740629, 2019-10-26:
> > update
> > EW-7811UTC base rtl8811au chipset
> > you can download src code from EDIMAX official web site
> > My TP-link T2U nano use rtl8811au chipset
> > The VID:PID = 2357:011E
> > It need add above ID on os_dep/linux/usb_intf.c file
> > and disable BT Function
> > modified the Makefile file
> > CONFIG_BT_COEXIST = n
> > the run Make
> > you can get rtl8811au.ko driver
> > insmod rtl8811au.ko
> > iwlist wlan0 scan
> > you can see the AP SSID information
> 
> I have the same tp-link adapter with the same VID:PID and I'm having the same issue as others where scan give no results.
> 
> @Andy-Chen-740629 can you please share more details on your fix? I downloaded the EDIMAX src code (EW-7811UTC_GPL_1.0.1.6.zip) and driver (EW-7811UTC_UAC_DAC_USC_Linux_Driver_1.0.2.4.zip) but **make** throws a bunch of erros
> 
> I'm on fedora kernel 5.3.6

hi
edimax vendor driver only work kernel 4.x
if work and compile done under kernel 5.x , it need some fix src code
but I do not know how to do fix it !

gustavovendramini, 2019-10-27:
Same issue on 5.6.4.1 and 5.2.20

System: Ubuntu 19.10
Kernel: 5.3.0-19-generic
Device: TP-LINK Archer T2U V3

natete, 2019-10-29:
Same issue on 5.6.4.1 and 5.2.20

System: Ubuntu 19.10
Kernel: 5.3.0-19-generic
Device: TP-LINK Archer T2U plus

marcingrabda, 2019-10-30:
> v5.2.20 works fine on Ubuntu 19.10, kernel 5.3.0-1007-raspi2.

Correction. The kernel module compiles, loads fine, finds networks and connects to them, but after a few minutes device disconnects with an error and never reconnects again.

kimocoder, 2019-10-30:
Found the solution to scanning..

Somehow we need to do "ifconfig <Interface> up" before "iwlist <Interface> scanning" :1st_place_medal:

Kasami-s, 2019-10-30:
> Found the solution to scanning..
> Tested on v5.6.4.1
> 
> Somehow we need to do "ifconfig up" before "iwlist scanning"

Doesn't seem to work for me.
`sudo ifconfig wlp0s20f0u1 up` runs, state remains the same
`6: wlp0s20f0u1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
`
networks still won't appear during scan (on latest v5.6.4.1)

kimocoder, 2019-10-30:
I'll show when im back Home, very soon

TugaPowerROM, 2019-10-30:
> I'll show when im back Home, very soon

Also does not fix for me, ( the issue you closed ) my issue its not no scan results, its the scan missing alot of hotspots, it shows some hotspots but its missing also alot of hotspots, kinda its stuck over the channel/band search.

TugaPowerROM, 2019-10-30:
sudo airmon-ng start "interface" rest frezzed trying to enable monitor mode.

> [   11.167867] Call Trace:
> [   11.167894]  rtw_cfg80211_ch_switch_notify+0xd4/0x123 [88XXau]
> [   11.167907]  rtw_chk_start_clnt_join+0x5c/0x90 [88XXau]
> [   11.167919]  join_cmd_hdl+0x26d/0x36e [88XXau]
> [   11.167926]  rtw_cmd_thread+0x191/0x3c1 [88XXau]
> [   11.167937]  ? rtw_chk_start_clnt_join+0x90/0x90 [88XXau]
> [   11.167939]  kthread+0x104/0x140
> [   11.167946]  ? rtw_stop_cmd_thread+0x3f/0x3f [88XXau]
> [   11.167947]  ? kthread_park+0x80/0x80
> [   11.167949]  ret_from_fork+0x35/0x40
> [   11.167950] ---[ end trace ccc73c295aad483d ]---

Only way it works good its first:

> sudo ip link set "interface" down
> sudo iw dev "interface" set type monitor
> sudo airmon-ng start "interface"

 (this time it will go fine )

TugaPowerROM, 2019-10-30:
As reference, this is working fine finding all the hotspots/routers but does not support monitor mode

https://github.com/olegbuevich/rtl8814au/tree/kernel-v5.3-support

kimocoder, 2019-10-31:
Take a look at this below .. this is what's strange!

https://raw.githubusercontent.com/kimocoder/rtl8812au/v5.6.4.1/rtl8812au.webm

homerogf8, 2019-10-31:
Same issue on 5.6.4.1, 5.6.4.2, 5.2.20.

System: Fedora 30
Kernel: 5.3.7-200.fc30.x86_64
Device: TP-Link Archer T2U V3

kimocoder, 2019-10-31:
I see that countless other Realtek drivers on newer kernels have the same issue, but wifite2 scans just fine.. I also noticed that iw was updated to v5.3 recently.. needs more testing, but it's working fine over here, as long as i put the Interface "up" manually

kimocoder, 2019-10-31:
[image]

kimocoder, 2019-10-31:
Injection / wifie2 test

[image]

TugaPowerROM, 2019-10-31:
Using 5.6.4.2 my adaptor its not even detected, tried to add manually the adaptor to
https://github.com/aircrack-ng/rtl8812au/blob/v5.6.4.2/os_dep/linux/usb_intf.c#L219

Still a no go, its not detected.

Bus 001 Device 002: ID 0846:9054 NetGear, Inc.

kimocoder, 2019-10-31:
I need to add a bunch of devices. I'll do it tomorrow :+1:

ortegahernandes, 2019-11-01:
Here is still a problem.

[image]
[image]

Ubuntu 19.10
Kernel 5.3-19
Device: d-link ac600 (DWA-171)
Bus 011 Device 002: ID 2001:3314 D-Link Corp. 802.11n WLAN Adapter

kimocoder, 2019-11-01:
@ortegahernandes which branch?

ortegahernandes, 2019-11-01:
5.6.4.1

KirkBushman, 2019-11-02:
same, showing no wifi Fedora 30, kernel 5.3.7, DWA-171, 5.6.4.1

phandox, 2019-11-03:
Hi, it started to work for me when using driver from this repository -> PR is opened - https://github.com/abperiasamy/rtl8812AU_8821AU_linux/pull/316 

Fedora 31, kernel 5.3.7, Edimax EW-7811UAC

[48 more replies]