aircrack-ng/rtl8812au #423: Pi 4 - Buster 4.19.66  no /proc/ net/rtl* Directory Structure

cariad61, 2019-09-24:
I have successfully complied and installed version 5.6.4.1 of this driver and have it working with my AWUS036ACH adapter on a clean install of Buster on a Pi 4.

However in /proc/net  there are no  rtl directories i.e. /proc/net/rtl88au, does not exist.

Please can anyone give any advise.

Thank you

Paul

cariad61, 2019-09-29:
After installing the same driver from [https://github.com/diederikdehaas/rtl8812AU/tree/driver-5.6.4] the /proc entries were found.

Please note an update to my original post, and the reason for installing the new driver set was because the drivers installed here caused the pi to lockup. When monitoring on a serial connection this gave kernel panic messages before crashing completely.

This is my second AWUS036ACH adapter, and what was really interesting was when I tired the second adapter on an existing stable system that the first AWUS036ACH adapter was used on the second AWUS036ACH adapter caused that system to crash.

From this I assume there is a difference in the two AWUS036ACH adapters.

So for anyone experiencing the same problem and who does not require anything fancy the install  [here](https://github.com/diederikdehaas/rtl8812AU/tree/driver-5.6.4) gave me a solution on my pi model 4.  My specific requirement was maximum power output - and monitoring the usb hub gave me a peak of 650 mili amps.   This was achieved by using the line rtw_power_mgnt=0 rtw_enusbss=0 rtw_tx_pwr_idx_override=63, in the /etc/modprobe.d/8812au.conf file.

closed 2019-09-29 by cariad61

kimocoder, 2019-09-29:
@cariad61 debug via /proc has been removed, however, just pushed a PR that fixes mfloat and arm64 fixes in general.