aircrack-ng/rtl8812au #427: Poor Signal with AWUS1900

manofftoday, 2019-09-29:
Hello,

Thank you so much for your work in this interface, really aprecciated.

I recently bought an AWUS1900. I'm considering to return it.

I need some help since I've not idea what version to use with this interface.

I installed in my host the 5.6.4 and I'm getting high signal wifi. It's the only version I tryed that it gets good wifi signal. (host.jpg)

I'm trying different versions on Kali Linux VBox.

I read that using an AWUS1900 the recommendable version to use is 5.2.20 since 5.6.4 is not supported and 5.6.4.1 is not stable. 

This is the wifi signal with the rest of the versions (5.2.20,5.6.4.1 ....) (kali.jpg)

v5.3.4 won't compile when I try to install with my kernel: 5.2.0-kali2-amd64

Could you help me?
Would you recommend me to return this interface and to get an wus036ach?

I couldn't do deauth with aireplay-ng either.

Thank you so much.

[image]
[image]

0m3rta13, 2019-09-29:
Crap device. Or rather crap driver support thanks to Realtek  (I’ve had mine barely working for just over a year). They seem to have forgotten they made the rtl8814au chipset. Pity cos it has great potential (and looks pretty cool). 

As a simple WiFi receiver it does alright, however, temperamental. 
Unfortunately in a virtual machine, don’t bother, you’ll age before your time. 

My advice, if you can, Return it 

Sorry to be the bearer of bad news

manofftoday, 2019-09-29:
> Crap device. Or rather crap driver support thanks to Realtek (I’ve had mine barely working for just over a year). They seem to have forgotten they made the rtl8814au chipset. Pity cos it has great [...]

Yes, I can return it...
Should I order the awus036ach then?
I'm starting an audit course and I need a proper device to work with 2,5ghz and 5ghz...

Thank you so much for your reply.

kimocoder, 2019-09-29:
there arebig differences between them.. @0m3rta13 said it. 8814au runs on a oooooold HAL/phydm

0m3rta13, 2019-09-29:
I can’t speak from personal experience as I don’t have one but @kimocoder is using one (his tests are shown in the reader) and he’s been putting a ton of time into this driver so yeah. That’s what I wouldn’t go with. 

OT/Side note: I still do my audits using a TP-link WN722 v1 (when 5Ghz isn’t a factor, which I see you need). So if you can still find a V1 (only V1 due to chipset) it’s still an amazing device for auditing 2.4Ghz

Good luck!!!

Mister-X-, 2019-09-29:
That chipset is using ath9k_htc and there are plenty of cards still available with it. Wikidevi has a [whole list](https://wikidevi.com/wiki/Ath9k_htc). TL;DR: An easy to find one is Alfa AWUS036NHA.

0m3rta13, 2019-09-29:
Indeed. Very common. But in my checks that one had the least power consumption, I like the form factor and it can be modified to add a second antenna. Either way. If you’re starting a course just grab the 036ach. The 5Ghz and ac will be useful

manofftoday, 2019-10-01:
Thank you so much for your assistance.
I returned the AWUS1900 and ordered a 036ACH. It will arrive tomorrow.

You helped me a lot.

cedricbambooza, 2020-12-31:
https://github.com/aircrack-ng/rtl8814au 

maybe consider closing the Issue ^^

closed 2024-05-08 by manofftoday