aircrack-ng/rtl8812au #428: Missing acess points during scan

TugaPowerROM, 2019-09-29:
Using driver 5.6.4.1 with the latest commits when typing this issue its all partially working except one small thing.
Under Mint 19.2 Cinnamon and kernel 5.3.1 build process and install its fine, no errors. Upon boot, driver does not hang the system anymore.

The problem is Network Manager / Wicd / and wpa_supplicant 2.9 its not hable to find some acess points, if adaptor is on Monitor Mode running aircrack-ng / wifite this programs can see the missing acess points.

My adaptor is A7000 – Nighthawk AC1900 WiFi USB Adapter – USB 3.0, Dual Band , it uses RTL8814AU identified as device wlx0836c90a77fb under the system.

`iwconfig wlx0836c90a77fb
wlx0836c90a77fb  unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=0/100  Signal level=0 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

iwconfig wlx98ded016334c
wlx98ded016334c  IEEE 802.11  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=20 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off

`

pryre, 2019-10-23:
I think I have the same issue. I cannot scan for APs (no results) using wifi-menu on Arch, and trying to connect to a known AP gives a WPA error (WPA association/authentication failed for interface 'wlan0'). I have no idea what mode the adaptor is in, I haven't used `wifite`.

iwconfig
```
wlan0     unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated
          Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

Dmesg (on USB plug in)
```
[22053.706391] usb 3-1: new high-speed USB device number 9 using xhci_hcd
[22053.846732] usb 3-1: New USB device found, idVendor=7392, idProduct=a812, bcdDevice= 2.00
[22053.846734] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[22053.846735] usb 3-1: Product: Edimax AC600 USB
[22053.846736] usb 3-1: Manufacturer: Realtek
[22053.846737] usb 3-1: SerialNumber: 00e04c000001
```

kimocoder, 2019-10-30:
Duplicate of #422 
The solution is actually over there..

closed 2019-10-30 by kimocoder

TugaPowerROM, 2019-10-30:
Dont think its a duplicate beause my issue is not no scan results, its the scan showing only few hotspots, the fix you pushed does not fix the missing hotspots that is the bug reported.