aircrack-ng/rtl8812au #429: 5.6.4.1 (5.3.1-arch1-1-ARCH) platform regulatory.0: Direct firmware load for regulatory.db failed with error -2

WataruMatsuoka, 2019-10-01:
I installed 1.6.4.1 driver with latest commit.
There is no problem while booting, but device doesn't wakes up.
Under 5.3.1-arch1-1-ARCH kernel and no additional other wireless driver installed.
My device is Elecom WDC-433DU2HBK(056e 4007). Here is my dmesg output.

> [   66.921110] usb 2-4: Product: WDC-433DU2HBK
[   66.921112] usb 2-4: Manufacturer: Elecom
[   66.921114] usb 2-4: SerialNumber: 00e04c000001
[   67.120160] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   67.130492] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   67.130573] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
[   67.130575] cfg80211: failed to load regulatory.db
[   67.283110] 88XXau: loading out-of-tree module taints kernel.
[   67.286359] 88XXau: module verification failed: signature and/or required key missing - tainting kernel
[   67.516934] usbcore: registered new interface driver rtl88xxau
[   67.569962] rtl88xxau 2-4:1.0 wlp0s29f7u4: renamed from wlan0
[   67.586864] BUG: kernel NULL pointer dereference, address: 00000000000008c0
[   67.586926] #PF: supervisor read access in kernel mode
[   67.586966] #PF: error_code(0x0000) - not-present page
[   67.587004] PGD 0 P4D 0 
[   67.587030] Oops: 0000 [#1] PREEMPT SMP PTI
[   67.587061] CPU: 0 PID: 350 Comm: systemd-udevd Tainted: G           OE     5.3.1-arch1-1-ARCH #1
[   67.587114] Hardware name: EPSON DIRECT CORP. Endeavor NJ3100E/NJ3100E, BIOS AM8-0215-DM5S        10/07/2009
[   67.587256] RIP: 0010:rtw_ndev_notifier_call+0x31/0x55 [88XXau]
[   67.587299] Code: d2 74 45 55 53 48 89 f3 48 8b 2a 48 85 ed 74 12 48 89 ef e8 99 fe ff ff 48 83 fb 0b 75 04 84 c0 75 08 b8 00 00 00 00 5b 5d c3 <48> 8b 04 25 c0 08 00 00 48 8d b8 48 45 00 00 ba 10 00 00 00 48 89
[   67.587407] RSP: 0018:ffffb8d8c039b7b0 EFLAGS: 00010202
[   67.587441] RAX: 0000000000000001 RBX: 000000000000000b RCX: 0000000000000001
[   67.587485] RDX: ffffffffc0ab1e80 RSI: 000000000000000b RDI: ffff9abab7b06000
[   67.587528] RBP: ffff9abab7b06000 R08: 0000000000800000 R09: ffffffffa189c660
[   67.587570] R10: ffffb8d8c039b7c8 R11: ffffb8d9c039b6b6 R12: 000000000000000b
[   67.587613] R13: ffffb8d8c039b818 R14: ffffffffa1907d20 R15: 0000000000000000
[   67.587658] FS:  00007fa6b7a1b840(0000) GS:ffff9ababba00000(0000) knlGS:0000000000000000
[   67.587711] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   67.587747] CR2: 00000000000008c0 CR3: 0000000139178000 CR4: 00000000000406f0
[   67.587790] Call Trace:
[   67.587819]  notifier_call_chain+0x4c/0x70
[   67.587854]  dev_change_name+0x219/0x320
[   67.587886]  do_setlink+0xd45/0xf10
[   67.587917]  ? prep_new_page+0x46/0xd0
[   67.587949]  ? get_page_from_freelist+0xc03/0x1240
[   67.587989]  rtnl_setlink+0x105/0x170
[   67.588023]  rtnetlink_rcv_msg+0x137/0x3c0
[   67.588054]  ? rtnl_calcit.isra.0+0x120/0x120
[   67.588086]  netlink_rcv_skb+0x75/0x140
[   67.588115]  netlink_unicast+0x177/0x1f0
[   67.588148]  netlink_sendmsg+0x204/0x3d0
[   67.588185]  sock_sendmsg+0x5e/0x60
[   67.588212]  __sys_sendto+0x120/0x190
[   67.588242]  __x64_sys_sendto+0x25/0x30
[   67.588270]  do_syscall_64+0x5f/0x1c0
[   67.588298]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[   67.588333] RIP: 0033:0x7fa6b923664b
[   67.588360] Code: 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 f3 0f 1e fa 48 8d 05 d5 4b 0c 00 41 89 ca 8b 00 85 c0 75 14 b8 2c 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 75 c3 0f 1f 40 00 55 48 83 ec 30 44 89 4c 24
[   67.588464] RSP: 002b:00007fff00f3c838 EFLAGS: 00000246 ORIG_RAX: 000000000000002c
[   67.588512] RAX: ffffffffffffffda RBX: 00007fff00f3c8a4 RCX: 00007fa6b923664b
[   67.588554] RDX: 0000000000000030 RSI: 000055f372ed2640 RDI: 000000000000000f
[   67.588597] RBP: 000055f372e8e910 R08: 00007fff00f3c840 R09: 0000000000000010
[   67.588603] audit: type=1130 audit(1569848547.106:19): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-rfkill comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   67.588643] R10: 0000000000000000 R11: 0000000000000246 R12: 000055f372f48f40
[   67.593023] R13: 000000000000015e R14: 000000000000015e R15: 000055f372ec6b40
[   67.593023] Modules linked in: 88XXau(OE) cfg80211 i915 snd_hda_codec_hdmi snd_hda_codec_realtek joydev mousedev snd_hda_codec_generic i2c_algo_bit ledtrig_audio drm_kms_helper snd_hda_intel snd_hda_codec tpm_infineon snd_hda_core snd_hwdep snd_pcm r8169 asus_laptop coretemp r592 iTCO_wdt snd_timer iTCO_vendor_support drm memstick snd lpc_ich realtek psmouse sparse_keymap soundcore syscopyarea pcspkr libphy sysfillrect tpm_tis input_leds sysimgblt rfkill fb_sys_fops tpm_tis_core tpm input_polldev mac_hid evdev battery intel_agp intel_gtt ac agpgart rng_core ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 sd_mod sr_mod cdrom serio_raw atkbd libps2 ahci libahci sdhci_pci cqhci uhci_hcd libata sdhci scsi_mod mmc_core firewire_ohci firewire_core crc_itu_t ehci_pci ehci_hcd i8042 serio
[   67.593023] CR2: 00000000000008c0
[   67.609589] ---[ end trace 79e26bd8b09cf9b4 ]---

henrywallace, 2019-10-03:
I'm seemingly experiencing this bug too on the exact same kernel version 5.3.1-arch1-1-ARCH. The `BUG: kernel NULL pointer dereference, address: 00000000000008c0` seems to be the pertinent part, or at least is what is highlighted in red in dmesg.

I've also get a seemingly similar stacktrace when installing https://github.com/zebulon2/rtl8814au. Speaking of what's the difference between those projects?

Here is my stack trace: https://gist.github.com/henrywallace/cb8bb75e1be21abaa71ad13e62264850
It also looks like there are repeated segments? I'm curious if that was just the kernel module attempting to be loaded multiple times. I'm a kernel module noob. Any pointers on how to read or debug these kinds of things?

WataruMatsuoka, 2019-10-03:
Oh, I experienced this bug with another version or developer's driver too.
I got another stack trace with v5.3.4 driver of this repository.
https://gist.github.com/WataruMatsuoka/e9212d04d2b40969f00dbde57ed3acfb

WataruMatsuoka, 2019-10-03:
This issue and #421 may be same.
Your stack trace and [mine](https://gist.github.com/WataruMatsuoka/e9212d04d2b40969f00dbde57ed3acfb) has `WARNING: CPU: 1 PID: 42 at net/wireless/core.c:868 wiphy_register.cold+0x37/0x435 [cfg80211]`.

christosvar, 2019-10-07:
I can confirm, this issue is resolved.

terencode, 2019-10-21:
`v5.6.4.1` has the fix merge (#424). Please rebuild from this branch and try again.

henrywallace, 2019-10-23:
Can confirm, fixed for me.

kimocoder, 2019-11-09:
Move over to v5.6.4.2 instead of v5.6.4.1

closed 2019-11-09 by kimocoder