aircrack-ng/rtl8812au #431: Really slow AC speedtest with RTL8814AU

dmarkey, 2019-10-03:
Speedtest is giving me 25 mbit/s but the physical link is 

```
Connected to ca:19:07:12:34:56 (on wlp7s0f3u2)
        SSID: MarkeyAC
        freq: 5180
        RX: 37348388 bytes (24682 packets)
        TX: 1052512 bytes (10485 packets)
        signal: -65 dBm
        tx bitrate: 1300.0 MBit/s

        bss flags:      short-slot-time
        dtim period:    0
        beacon int:     0
```

Tried the driver in both usb2 and usb3 mode.

dmarkey, 2019-10-03:
Should be giving over 100 easily.

dmarkey, 2019-10-03:
Switching to the LTS kernel gives much better results
```
➜  ~ speedtest-cli --server 1041 --no-upload
Retrieving speedtest.net configuration...
Testing from Virgin Media Ireland (46.7.68.164)...
Retrieving speedtest.net server list...
Retrieving information for the selected server...
Hosted by Digiweb (Dublin) [1.72 km]: 13.858 ms
Testing download speed................................................................................
Download: 108.15 Mbit/s
Skipping upload test
➜  ~ uname -a
Linux xxx 4.19.76-1-lts #1 SMP Tue Oct 1 18:47:17 CEST 2019 x86_64 GNU/Linux
```

logan001, 2019-10-09:
you should go with 5.1.5 but that has a bug of not shutting down the power for the device when you shut down the computer.
i have the same problem with every branch except 5.1.5.
use to work with 5.3.4 too but this one was "nerfed" at the beginning of this year

llccd, 2019-10-18:
Try this patch (for 5.6.4.1)
```
--- a/hal/rtl8814a/usb/rtl8814au_xmit.c
+++ b/hal/rtl8814a/usb/rtl8814au_xmit.c
@@ -132,7 +132,6 @@ static s32 update_txdesc(struct xmit_frame *pxmitframe, u8 *pmem, s32 sz ,u8 bag
                rtl8814a_fill_txdesc_sectype(pattrib, ptxdesc);

                //offset 20
-               }
 #ifdef CONFIG_USB_TX_AGGREGATION
                if (pxmitframe->agg_num > 1){
                        //RTW_INFO("%s agg_num:%d\n",__FUNCTION__,pxmitframe->agg_num );
@@ -224,7 +223,8 @@ static s32 update_txdesc(struct xmit_frame *pxmitframe, u8 *pmem, s32 sz ,u8 bag
                        SET_TX_DESC_TX_RATE_8814A(ptxdesc, MRateToHwRate(pmlmeext->tx_rate));
                }

-               if ((pxmitframe->frame_tag&0x0f)== MGNT_FRAMETAG) {
+               }
+               else if ((pxmitframe->frame_tag&0x0f)== MGNT_FRAMETAG) {
                //RTW_INFO("pxmitframe->frame_tag == MGNT_FRAMETAG\n");

                SET_TX_DESC_USE_RATE_8814A(ptxdesc, 1);
```

dmarkey, 2019-10-24:
Sorry that patch doesnt apply.

CGarces, 2019-10-24:
> Try this patch (for 5.6.4.1)
> 
> ```
> --- a/hal/rtl8814a/usb/rtl8814au_xmit.c
> +++ b/hal/rtl8814a/usb/rtl8814au_xmit.c
> ```
Hi @llccd fell free to send a PR. 
You are the author of that patch or comes from another repo?

kimocoder, 2019-10-30:
Remember, the 8814AU is a v5.1.5 version, not like 8812AU and 8821AUA, which are way newer.. nothing we could do about it right now