aircrack-ng/rtl8812au #438: Make Error

mehmetyldz87, 2019-10-10:
Hi, 

When I try to compile and install driver. I encounter with this error.  What is this error ? How can I fix it? Thanks for your helps

OS: Ubuntu 16.04 LTS

/home/gorkes/rtl8812au/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_indicate_sta_assoc’:
/home/gorkes/rtl8812au/os_dep/linux/ioctl_cfg80211.c:4484:3: error: implicit declaration of function ‘cfg80211_sinfo_alloc_tid_stats’ [-Werror=implicit-function-declaration]
   cfg80211_sinfo_alloc_tid_stats(&sinfo, GFP_KERNEL);
   ^
/home/gorkes/rtl8812au/os_dep/linux/ioctl_cfg80211.c: At top level:
cc1: warning: unrecognized command line option ‘-Wno-misleading-indentation’
cc1: warning: unrecognized command line option ‘-Wno-cast-function-type’
cc1: warning: unrecognized command line option ‘-Wno-cast-function-type’
cc1: some warnings being treated as errors
scripts/Makefile.build:330: recipe for target '/home/gorkes/rtl8812au/os_dep/linux/ioctl_cfg80211.o' failed
make[2]: *** [/home/gorkes/rtl8812au/os_dep/linux/ioctl_cfg80211.o] Error 1
Makefile:1571: recipe for target '_module_/home/gorkes/rtl8812au' failed
make[1]: *** [_module_/home/gorkes/rtl8812au] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-65-generic'
Makefile:1608: recipe for target 'modules' failed
make: *** [modules] Error 2

kimocoder, 2019-10-10:
Yes I think I know where to look, I will do some tests soon. I'll leave a note here soon

lupa18, 2019-10-13:
Hello ! here same problem :pray:

mehmetyldz87, 2019-10-18:
hi @kimocoder  ,did you get a chance to look at the problem? Thanks

Trivedi-dot, 2019-10-19:
Hello sir,am at Ubuntu 18.04.3 LTS and having the same issue.!

kimocoder, 2019-11-11:
@mehmetyldz87 @Trivedi-dot @lupa18 

Done lots of changes. Update to v5.6.4.2, clone it and compile/build/install/reboot and see if the problem is fixed, report back if it's resolved or the problem is persistent :+1:

lupa18, 2019-11-11:
it seem is working with dkms. thanks!

kimocoder, 2019-11-11:
Awesome! Another issue smashed.

If you'd encounter more issues, please open tickets and share details :1st_place_medal:

closed 2019-11-11 by kimocoder

mehmetyldz87, 2019-12-06:
Thanks , problem solved for me