aircrack-ng/rtl8812au #439: hostapd doesn't work correctly after put one of rtl8812au device in monitor mode

NicholasFalcon, 2019-10-10:
I have a problem most likely in the driver rtl8812au version 5.6.4.1. I have 2 AWUS036AC one is used for hostapd v 2.9 the other is put into monitor mode. After reboot OS (debian, ubuntu, kali) hostapd works fine clients can connect. I put second awus036ac in monitor mode the clients can connect, then if I will restart hostapd the client won't be able to connect AP. But in this case 2 AWUS036NHR (rtl8192cu) works fine

kimocoder, 2019-11-11:
@NicholasFalcon 

Use branch v5.6.4.2 instead of v5.6.4.1. Also I've just added support for hostapd to the branch.
Report back if both the problem is resolved or persistent :+1:

NicholasFalcon, 2019-11-14:
@kimocoder 

Hello, I've installed the new driver version (5.6.4.2) in kali linux, downloaded all the dependencies.
Downloaded hostapd `v 2.9`

> apt install hostapd

Downloaded dnsmasq `v 2.80`

> apt install dnsmasq

edited configs:

/etc/hostapd/hostapd.conf
`interface=wlan0`
`driver=nl80211`
`ssid=alphaAP`
`hw_mode=g`
`channel=6`

/etc/dnsmasq.conf
`bogus-priv`
`address=/#/192.0.2.1`
`server=/captive/192.0.2.1`
`local=/captive/`
`interface=wlan0`
`domain=captive`
`dhcp-range=192.0.2.50,192.0.2.250,2h`
`dhcp-option=3,192.0.2.1`
`dhcp-option=6,192.0.2.1`
`dhcp-authoritative`

/etc/network/interfaces
`auto wlan0`
`face wlan0 inet static`
`address 192.0.2.1 `
`netmask 255.255.255.0 `

after it:

> ifup wlan0
> service dnsmasq start
> service hostapd start

all worked fine, but when i put second interface in monitor mode `wlan1`

> ip link set wlan1 up
> iw dev wlan1 set type monitor
> ip link set wlan1 up
> airodump-ng wlan1

and restart hostapd
> ifup wlan0
> service hostapd stop
> service hostapd start

ap in wifi list not found
hostapd status is not different when the launch is successful and not successful

LeeTiK, 2019-11-23:
similar problem = (

cedricbambooza, 2021-05-11:
Hey, I'm afraid I can't give you the right support on this one. 
Just wanted to make sure old solved or unrelevant issues don't stay open too long.
But this seems relevant and unsolved to you, so maybe someone in the community can help you :)

sir-lh, 2023-05-29:
I wonder if this question has been solved? hostapd doesn't work after one in monitor mode.

sir-lh, 2023-05-29:
I wonder if this question has been solved? Although hostapd is running. it doesn't seem to receive the frames.
For example, auth frames are not received in the wps_pbc.