aircrack-ng/rtl8812au #440: Make fails - Error 2 - 4.19.75-v7+ Raspian

picameratk, 2019-10-10:
I am running Rasbian Buster 4.19.75-v7+ and am attempting to run make.  I just performed:
1. make uninstall
2. make clean
3 rm -rf rtl8812au
4. git clone -b v5.6.4.1 https://github.com/aircrack-ng/rtl8812au.git
5.  Changed Makefile using nano
   CONFIG_PLATFORM_I386_PC = n
   CONFIG_PLATFORM_ARM_RPI = y
6.  Ran make
root@YE4nGYi8:/home/tom/edimax/rtl8812au# make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.19.75-v7+/build M=/home/tom/edimax
/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.75-v7+'
  CC [M]  /home/tom/edimax/rtl8812au/core/rtw_cmd.o
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[2]: *** [scripts/Makefile.build:304: /home/tom/edimax/rtl8812au/core/rtw_cmd.o] Error 2
make[1]: *** [Makefile:1519: _module_/home/tom/edimax/rtl8812au] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.75-v7+'
make: *** [Makefile:1608: modules] Error 2

Any suggestions:?

Environment: Raspberry Pi 3B+, Edimax EW-7811UTC, 
Raspian Buster - Linux YE4nGYi8 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux

root@YE4nGYi8:/home/tom/edimax/rtl8812au# dpkg -l | grep raspberrypi
ii  raspberrypi-bootloader                1.20190925-2                              armhf        Raspberry Pi bootloader
ii  raspberrypi-kernel                    1.20190925-2                              armhf        Raspberry Pi bootloader
ii  raspberrypi-kernel-headers            1.20190925-2                              armhf        Header files for the Raspberry Pi Linux kernel

 Note that I used the driver this process creates with the my Edmax previous.  I updated the kernel and need to rerun make.

0unknwn, 2019-10-11:
I get the same result when trying to build with this kernel

picameratk, 2019-10-11:
I just ran an apt update and an apt upgrade and received different versions of this software
ii  raspberrypi-bootloader                1.20190925+1-1                            armhf        Raspberry Pi bootloader
ii  raspberrypi-kernel                    1.20190925+1-1                            armhf        Raspberry Pi bootloader
ii  raspberrypi-kernel-headers            1.20190925+1-1                            armhf        Header files for the Raspberry Pi Linux kernel

With these versions I can now run make Successfully but other problems have surfaced and I will open another issue for those new issues

closed 2019-10-11 by picameratk