aircrack-ng/rtl8812au #441: AWUS1900 - SIOCGIWPRIV: Argument list too long

FireFart, 2019-10-10:
Using the latest driver from `v5.6.4.1` and an AWUS1900 (RTL8814AU) I get the following error in bettercap when it tries to activate the device via libpcap:
```text
SIOCGIWPRIV: Argument list too long
```
Is there a way to enable debug logging to provide you more details on this error?

FireFart, 2019-10-15:
Looks like this happens somewhere around here https://github.com/the-tcpdump-group/libpcap/blob/19236989728539214cf0cf1897495bac9bb3abb7/pcap-linux.c#L5812 when libpcap tries to get a list of private ioctls

kimocoder, 2019-10-30:
I'll take a look at this shortly

stepen, 2020-01-15:
(For INFO) I see the same error when using libtins library v4.3 (https://github.com/mfontanini/libtins) with this driver.

m00k12, 2020-06-08:
I think this is a kernel bug.  The function [iw_handler_get_private()](https://elixir.bootlin.com/linux/v5.7.1/source/net/wireless/wext-priv.c#L16) contains the following code:

```C
	/* Check if there is enough buffer up there */
	if (wrqu->data.length < dev->wireless_handlers->num_private_args) {
		/* User space can't know in advance how large the buffer
		 * needs to be. Give it a hint, so that we can support
		 * any size buffer we want somewhat efficiently... */
		wrqu->data.length = dev->wireless_handlers->num_private_args;
		return -E2BIG;
	}
````

The top level kernel function for handling ioctls calls is [wext_handle_ioctl()](https://elixir.bootlin.com/linux/v5.7.1/source/net/wireless/wext-core.c#L1037), but that guards the call to copy_to_user() with a check for ret >= 0.  Since we are returning -E2BIG, the copy back to user-space never happens and the size 'hint' isn't seen.  The code in libpcap then goes on to allocate no space and call ioctl(SIOCGIWPRIV) again, which fails again with -E2BIG and forces it down an error path.

libpcap can be patched with something like the following hack before the second call to ioctl(SIOCGIWPRIV) to make things work with this (and other) drivers:

```C
if(ireq.u.data.length == 0) ireq.u.data.length = 64;
```

I don't believe anything can be changed in this driver to fix this issue.  I could be wrong however.

alexkrycek, 2020-06-18:
In case anyone else is using libtins and receiving this error message, I fixed it by following the following advice (according to [this link](https://github.com/mfontanini/libtins/issues/356#issuecomment-626311038)):

> Removing `set_rfmon(true)` from SnifferConfiguration object helps to solve this issue. You need to set monitoring mode by yourself. For example: `iw dev wlan1 set type monitor`.
Also will help with raw pcap handler by removing `pcap_set_rfmon()` function.

chambers-web, 2020-09-22:
I am facing the same issue with the same driver and AWUS036ACH. 

@alexkrycek: where did you remove `set_rfmon(true)` from?

alexkrycek, 2020-09-22:
If you use the libtins library, you create a SnifferConfiguration object and set certain parameters with it. One of those is set with the command "config.set_rfmon(true)". That's the command that you should omit. See https://libtins.github.io/examples/beacon-display/

chambers-web, 2020-09-23:
I see. I get the error from tcpdump. And tcpdump doesn't use libtins so I don't know where to change then.. I'll look at libpcap.

m00k12, 2020-09-23:
> I'll look at libpcap.

There's a patch I provided on this old ticket, but newer versions fixed the issue in libpcap, see here: https://github.com/the-tcpdump-group/libpcap/issues/941

chambers-web, 2020-09-28:
Thanks @m00k12, I installed libpcap from the source as the latest release 1.9.1 is older than the patch. It still doesn't work but maybe it's a progress?

```
kali@kali:~$ sudo iw dev
...
phy#0
	Interface wlan0
...
		type monitor
		txpower 20.00 dBm
kali@kali:~$ sudo /sbin/tcpdump -I -i wlan0  # tcpdump 1.9.1
tcpdump: wlan0: SIOCGIWPRIV: Argument list too long
kali@kali:~$ sudo /usr/local/bin/tcpdump -I -i wlan0  # tcpdump from the source
tcpdump: wlan0: That device doesn't support monitor mode  # <--- This does not make sense to me
kali@kali:~$
```

m00k12, 2020-09-28:
@chambers-web - make sure you compiled libpcap with libnl support.  Their configure script won't complain if it doesn't find any libnl, but then the built library lacks a lot of device control stuff which could be your problem.  If not, strace and printf() might help you track down the cause.

chambers-web, 2020-09-28:
Oh wow, @m00k12, you are fantastic! Thank you!

Indeed the previous libpcap was missing libnl support so I compiled it again. Now almost good - except that 1) tcpdump captures no packets 2) It brings down the host eventually. At this point I'm not sure if it's a libpcap issue - probably on the driver side? Maybe I should try another wifi adapter with a different chipset?

And I can reproduce 2) with `airmon-ng start wlan0`. 

```
kali@kali:~$ sudo /usr/local/bin/tcpdump -I -i wlan0
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on wlan0, link-type IEEE802_11_RADIO (802.11 plus radiotap header), snapshot length 262144 bytes
^C
0 packets captured
0 packets received by filter
0 packets dropped by kernel

Message from syslogd@kali at Sep 28 17:10:59 ...
 kernel:[  445.621307] Internal error: Oops - BUG: 0 [#1] SMP ARM

Message from syslogd@kali at Sep 28 17:10:59 ...
 kernel:[  445.621544] Process tcpdump (pid: 1209, stack limit = 0xf2196af9)
```

FireFart, 2020-09-28:
Guys please keep the discussion to the original issue. This is no help forum

cedricbambooza, 2021-05-11:
pls consider closing the issue, when it's solved by now :)