aircrack-ng/rtl8812au #442: Edimax EW-7811UTC no longer works in AP or client mode with latest Raspian

picameratk, 2019-10-13:
After upgrading to Linux 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux and running a make && make install I can no longer get the Edimax to work using hostapd or as a client (i.e., wpa_supplicant)

The Makefile has been changed so that 
CONFIG_PLATFORM_I386_PC = n and
CONFIG_PLATFORM_ARM_RPI = y (if I try to use ARM64 version it throws an error)

While hostapd 2.9 will start successfully and iwconfig shows an SSID I can use iw xxxx scan on another wlan adapter and will not see the SSID being broadcast.   Additionally I cannot see the SSID on my iPhone either.

When running in client mode dhclient will never find a router to get an ip address from.

One abnormal item is that the txpower will always show as zero.  Of course, I run 
iw wlan2 set txpower fixed 2000 and then txpower shows.

I used git clone -b v5.6.4.1 https://github.com/aircrack-ng/rtl8812au.git to get rtl8812au.  make and make install run without errors.  I have also run make uninstall and make clean a few times as well. 

Note that I have two identical Edimax devices and both of them are giving similar results.  I also have a Raspberry Pi 3 B+ that I have not updated and the Edimax devices operate correctly.

Any idea what might be going wrong?

ETC:
dpkg -l  info:
ii  raspberrypi-bootloader                1.20190925+1-1                            armhf        Raspberry Pi bootloader

ii  raspberrypi-kernel                    1.20190925+1-1                            armhf        Raspberry Pi bootloader

ii  raspberrypi-kernel-headers            1.20190925+1-1                            armhf        Header files for the Raspberry Pi Linux kernel

Environment: 
Raspberry Pi 3 B+ with a 2.5A power capability.
The builtin Raspberry Pi wifi will operate correctly in both hostapd and in client mode.
An additional Hawkings wifi adapter works as well.

picameratk, 2019-10-14:
After using a v5.3.4 version I had left laying around I am able to get the Edimax to run either hostapd or client mode.  In other words at the v5.3.4 level everything is working again.

cedricbambooza, 2021-05-11:
pls consider closing the issue, when it's solved by now :)