aircrack-ng/rtl8812au #445: [Kernel 5.3.5, Fedora 30] Empty visible wifi networks on v5.2.20 and v5.6.4.1

KirkBushman, 2019-10-16:
I was on v5.3.4 driver before, I updated today to the linux kernel 5.3.5, and it doesn't show any visible network in the panel, I've then tried all the versions, even the ones that have the fix for kernel 5.3+, but nothing... 
Can't see any network.

jhoward321, 2019-10-17:
I can report having the same problem. Same kernel on Fedora 30.

CLiNTPELiX, 2019-10-18:
I'm having nearly same problems on latest Ubuntu 19.10 and Fedora 31 Beta.
I can see all channel in 2.4 GHz(b,g,n) and 5 GHz(ac) (but only channels below 56)

felubra, 2019-10-19:
I have the same problem in the 5.2.15-200.fc30.x86_64 kernel too.

ortegahernandes, 2019-10-21:
Here also does not appear wifi networks
Ubuntu 19.10 (kernel 5.3.0-19)

kimocoder, 2019-10-30:
Follow #422 the solution is there

closed 2019-10-30 by kimocoder