aircrack-ng/rtl8812au #448: Impossible to put monitor mode Ubuntu 19

acdcjavier, 2019-10-18:
Hello
Recently I've installed a fresh Ubuntu 19, I installed all aircrack-ng features but I can't put monitor mode my wireless card RTL8812AU

Got this error:

> 
> PHY	Interface	Driver		Chipset
> 
> phy0	wlan0		ath9k		Qualcomm Atheros QCA9565 / AR9565 Wireless Network Adapter (rev 01)
> phy2	wlan1		88XXau		Linksys WUSB6300 802.11a/b/g/n/ac [Realtek RTL8812AU]
>  (monitor mode enabled)
> Error setting channel: command failed: Device or resource busy (-16)
> Error -16 likely means your card was set back to station mode by something.
> Removing non-monitor wlan1 interface...
>  (monitor mode disabled)

How can I solve that?

ZeroChaos-, 2019-10-18:
I've also tried the latest 5.6.4.1 and can't put the device in monitor mode via iwconfig or iw.

ZeroChaos-, 2019-10-21:
based on some feedback it seems this may be limited to 5.3.x kernels

ZeroChaos-, 2019-10-21:
positive confirmation this works in kernel 5.2.x but does not on 5.3.x.

kimocoder, 2019-10-24:
Yes,had some time to look into several things.. this is the case..

* Scanning isn't working on newer kernels (by iw), BUT.. when running wifite2 everything works perfect! ?? (recommend v5.6.4.1 over v5.2.20 atm. not for 8821au though)

* I actually struggle to get packet injection working using "iw" now, but when using "airmon-ng start wlan1" it works like a charm.. ??

Got some more time this weekend :+1:

ZeroChaos-, 2019-10-24:
airmon-ng start uses iwconfig for this driver, not iw

acdcjavier, 2019-10-28:
> Yes,had some time to look into several things.. this is the case..
>
>     * Scanning isn't working on newer kernels (by iw), BUT.. when running wifite2 everything works perfect! ?? (recommend v5.6.4.1 [...]

Ok, thanks, but how can you add a new "monitor interface" for rtl8812AU?
I have tried using wifite2 but the program cannot set the monitor mode, nor make a new monitor interface

kimocoder, 2019-10-30:
Found solution for scanning.. need to do "ifconfig <Interface> up" before "iwlist <Interface> scanning"

ZeroChaos-, 2019-10-30:
You know "iwlist scanning" and monitor mode are different things, right? This bug is about not being able to set monitor mode.

kimocoder, 2019-10-30:
Uploading v5.6.4.2 base from Realtek.. stay put, I'll start right away

kimocoder, 2019-11-11:
@acdcjavier The v5.6.4.2 branch is working fine for 8812AU now, update, build, install and reboot to load new module.

Open a new issue report if you got more issues, or this doesn't work for you in any way. :+1:

closed 2019-11-11 by kimocoder