aircrack-ng/rtl8812au #449: dmesg error

Elleni57, 2019-10-18:
This version (net-wireless/rtl8812au_aircrack-ng-5.6.4.1) enables usb ac68 wifi stick with kernel 5.3. but I see the following in dmesg: 
Call Trace: rtw_cfg80211_ch_switch_notify+0xd5/0x134 [88XXau] rtw_chk_start_clnt_join+0x66/0x95 [88XXau] join_cmd_hdl+0x276/0x384 [88XXau] ? rtw_chk_start_clnt_join+0x95/0x95 [88XXau] rtw_cmd_thread+0x261/0x3bc [88XXau] ? rtw_stop_cmd_thread+0x34/0x34 [88XXau] kthread+0x113/0x130 ? kthread_park+0x80/0x80 ret_from_fork+0x22/0x40 ---[ end trace b69a0943f723e7b9 ]---

uname -a
Linux gentoo1 5.3.7-gentoo #1 SMP PREEMPT Fri Oct 18 21:00:31 CEST 2019 x86_64 AMD Ryzen 5 1600 Six-Core Processor AuthenticAMD GNU/Linux

Thought, I should report that here, I hope this is the right place.

ZerBea, 2019-10-20:
You are possible hit by this kernel issue:
https://bugzilla.kernel.org/show_bug.cgi?id=202541
Please check dmesg for "WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state".

Elleni57, 2019-10-20:
Did not find mentioned WARN, as dmesg |grep WARN gave only

WARNING: CPU: 9 PID: 3383 at net/wireless/nl80211.c:16129 cfg80211_ch_switch_notify.cold.86+0x15/0x25

Following the output of my dmesg: 
http://dpaste.com/0KNSVWG

Elleni57, 2019-10-21:
To be more precisely: compiled rtl8812au_aircrack-ng-5.6.4.1 version with 5.2.20 kernel, and it shows the same call trace in dmesg. So I went back to rtl8812au_aircrack-ng-5.3.4_p20190622 which works fine with 5.2 kernel. It even builds with 5.3.X kernel, and the module shows loaded with lsmod, but there is no internet connectivity. Thats why I was trying to update to more recent version.

terencode, 2019-10-21:
If I'm not mistaken this was already reported numerous time like #421 and #383.

kimocoder, 2019-10-30:
Duplicate

closed 2019-10-30 by kimocoder

CGarces, 2019-12-26:
I think that not is a duplicate.
The stack trace `[ 16.035713] rtw_cfg80211_ch_switch_notify+0xd5/0x134 [88XXau]`
not is part of  #421 
#383 shows the trace at the beginning of the issue but after test several versions of kernel/drivers is closed due #424 that solve one null pointer but not the initial issue.