aircrack-ng/rtl8812au #450: Trouble getting my T9UH (RTL8814AU) to run under USB3

aaa3k1, 2019-10-19:
- Running Xubuntu 19.10
- Cloned from `https://github.com/aircrack-ng/rtl8812au.git`
- Tried installing under DKMS
- Tried installing just using make + make install 
- Under both separate installation attempts, I tried:
`rmmod 88XXau
modprobe 88XXau rtw_switch_usb_mode:1`
- Tried creating `/etc/modprobe.d/88XXau.conf` and adding the following line:
`options 88XXau rtw_switch_usb_mode=1`

Output of `lsusb -v` and the speedtests indicate that it is not running under USB 3:

`Bus 001 Device 007: ID 2357:0106 TP-Link 802.11ac NIC
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  **bcdUSB               2.10**
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x2357 TP-Link
  idProduct          0x0106 
  bcdDevice            0.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0035
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           5
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1

`

Feel like I am running around in circles now, any assistance would be greatly appreciated, thank you!

ZerBea, 2019-10-20:
You are possible hit by this kernel issue:
https://bugzilla.kernel.org/show_bug.cgi?id=202541
Please check dmesg.

aaa3k1, 2019-10-20:
I did not find that "WARN SET TR Deq Ptr" message, but I did see that
`88XXau rtw_switch_usb_mode:1` was not valid, set it to `=1` instead of `:1` and it worked.

My apologies.

aaa3k1, 2019-10-20:
Nevermind, it was displaying as USB 3.0 in `lsusb -v` for a moment but seems to have reverted to 2.0. It seems that rtw_switch_usb_mode:1 is stuck in my module when I check `dmesg | grep 88XXau` even when I have performed a reinstall.

`[b@b-pc modprobe.d]$ sudo dmesg | grep 88XXau
[    2.648773] 88XXau: loading out-of-tree module taints kernel.
[    2.649186] 88XXau: module verification failed: signature and/or required key missing - tainting kernel
[   13.787354] Modules linked in: rfcomm fuse cmac algif_hash algif_skcipher af_alg snd_hda_codec_hdmi 88XXau(OE) intel_rapl_msr intel_rapl_common bnep x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel btusb btrtl btbcm joydev hid_roccat_koneplus btintel hid_roccat input_leds mousedev hid_roccat_common kvm bluetooth nouveau cfg80211 snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio i2c_algo_bit ecdh_generic irqbypass squashfs hid_generic rfkill ecc ttm snd_hda_intel loop snd_hda_codec crct10dif_pclmul drm_kms_helper crc32_pclmul usbhid snd_hda_core hid ghash_clmulni_intel snd_hwdep aesni_intel snd_pcm aes_x86_64 crypto_simd iTCO_wdt drm r8169 snd_timer cryptd iTCO_vendor_support glue_helper agpgart snd syscopyarea realtek intel_cstate sysfillrect mxm_wmi intel_uncore mei_me psmouse intel_rapl_perf libphy pcspkr sysimgblt mei soundcore fb_sys_fops i2c_i801 evdev mac_hid wmi uinput crypto_user ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 sd_mod ahci serio_raw atkbd
[   13.787432]  rtw_cfg80211_ch_switch_notify+0xf4/0x146 [88XXau]
[   13.787447]  ? rtw_chk_start_clnt_join+0x8c/0x8c [88XXau]
[   13.787460]  rtw_chk_start_clnt_join+0x5d/0x8c [88XXau]
[   13.787474]  join_cmd_hdl+0x27b/0x387 [88XXau]
[   13.787483]  rtw_cmd_thread+0x253/0x3a9 [88XXau]
[   13.787497]  ? rtw_stop_cmd_thread+0x39/0x39 [88XXau]
[   68.486547] Modules linked in: 88XXau(OE) rfcomm fuse cmac algif_hash algif_skcipher af_alg snd_hda_codec_hdmi intel_rapl_msr intel_rapl_common bnep x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel btusb btrtl btbcm joydev hid_roccat_koneplus btintel hid_roccat input_leds mousedev hid_roccat_common kvm bluetooth nouveau cfg80211 snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio i2c_algo_bit ecdh_generic irqbypass squashfs hid_generic rfkill ecc ttm snd_hda_intel loop snd_hda_codec crct10dif_pclmul drm_kms_helper crc32_pclmul usbhid snd_hda_core hid ghash_clmulni_intel snd_hwdep aesni_intel snd_pcm aes_x86_64 crypto_simd iTCO_wdt drm r8169 snd_timer cryptd iTCO_vendor_support glue_helper agpgart snd syscopyarea realtek intel_cstate sysfillrect mxm_wmi intel_uncore mei_me psmouse intel_rapl_perf libphy pcspkr sysimgblt mei soundcore fb_sys_fops i2c_i801 evdev mac_hid wmi uinput crypto_user ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 sd_mod ahci serio_raw atkbd
[   68.486570]  libahci libps2 libata xhci_pci crc32c_intel scsi_mod xhci_hcd i8042 serio [last unloaded: 88XXau]
[   68.486629]  rtw_cfg80211_ch_switch_notify+0xf4/0x146 [88XXau]
[   68.486647]  ? rtw_chk_start_clnt_join+0x8c/0x8c [88XXau]
[   68.486663]  rtw_chk_start_clnt_join+0x5d/0x8c [88XXau]
[   68.486679]  join_cmd_hdl+0x27b/0x387 [88XXau]
[   68.486689]  rtw_cmd_thread+0x253/0x3a9 [88XXau]
[   68.486704]  ? rtw_stop_cmd_thread+0x39/0x39 [88XXau]
[  163.737122] Modules linked in: 88XXau(OE) rfcomm fuse cmac algif_hash algif_skcipher af_alg snd_hda_codec_hdmi intel_rapl_msr intel_rapl_common bnep x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel btusb btrtl btbcm joydev hid_roccat_koneplus btintel hid_roccat input_leds mousedev hid_roccat_common kvm bluetooth nouveau cfg80211 snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio i2c_algo_bit ecdh_generic irqbypass squashfs hid_generic rfkill ecc ttm snd_hda_intel loop snd_hda_codec crct10dif_pclmul drm_kms_helper crc32_pclmul usbhid snd_hda_core hid ghash_clmulni_intel snd_hwdep aesni_intel snd_pcm aes_x86_64 crypto_simd iTCO_wdt drm r8169 snd_timer cryptd iTCO_vendor_support glue_helper agpgart snd syscopyarea realtek intel_cstate sysfillrect mxm_wmi intel_uncore mei_me psmouse intel_rapl_perf libphy pcspkr sysimgblt mei soundcore fb_sys_fops i2c_i801 evdev mac_hid wmi uinput crypto_user ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 sd_mod ahci serio_raw atkbd
[  163.737155]  libahci libps2 libata xhci_pci crc32c_intel scsi_mod xhci_hcd i8042 serio [last unloaded: 88XXau]
[  163.737217]  rtw_cfg80211_ch_switch_notify+0xf4/0x146 [88XXau]
[  163.737232]  ? rtw_chk_start_clnt_join+0x8c/0x8c [88XXau]
[  163.737244]  rtw_chk_start_clnt_join+0x5d/0x8c [88XXau]
[  163.737257]  join_cmd_hdl+0x27b/0x387 [88XXau]
[  163.737265]  rtw_cmd_thread+0x253/0x3a9 [88XXau]
[  163.737278]  ? rtw_stop_cmd_thread+0x39/0x39 [88XXau]
**[  202.650857] 88XXau: unknown parameter 'rtw_switch_usb_mode:1' ignored**
[  210.080301] Modules linked in: 88XXau(OE) rfcomm fuse cmac algif_hash algif_skcipher af_alg snd_hda_codec_hdmi intel_rapl_msr intel_rapl_common bnep x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel btusb btrtl btbcm joydev hid_roccat_koneplus btintel hid_roccat input_leds mousedev hid_roccat_common kvm bluetooth nouveau cfg80211 snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio i2c_algo_bit ecdh_generic irqbypass squashfs hid_generic rfkill ecc ttm snd_hda_intel loop snd_hda_codec crct10dif_pclmul drm_kms_helper crc32_pclmul usbhid snd_hda_core hid ghash_clmulni_intel snd_hwdep aesni_intel snd_pcm aes_x86_64 crypto_simd iTCO_wdt drm r8169 snd_timer cryptd iTCO [... cut ...] pcm aes_x86_64 crypto_simd iTCO_wdt drm r8169 snd_timer cryptd iTCO_vendor_support glue_helper agpgart snd syscopyarea realtek intel_cstate sysfillrect mxm_wmi intel_uncore mei_me psmouse intel_rapl_perf libphy pcspkr sysimgblt mei soundcore fb_sys_fops i2c_i801 evdev mac_hid wmi uinput crypto_user ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 sd_mod ahci serio_raw atkbd
[  737.910093]  libahci libps2 libata xhci_pci crc32c_intel scsi_mod xhci_hcd i8042 serio [last unloaded: 88XXau]`

aaa3k1, 2019-10-20:
OK, I think I figured out the trick for me to get it working.

Here are the steps that worked for me:

1. `rmmod 88XXau`
2. Disconnect the device (important).
3. modprobe 88XXau rtw_switch_usb_mode=1

Not disconnecting the device first even though the module was disabled made me stuck in USB2.0.

closed 2019-10-20 by aaa3k1

ValZapod, 2019-10-24:
@aaa3k1 At least USB 3.0 works propely on linux, while on windows there is a service RunSwUSB which on boot loads runSW.exe which switches from usb 2.0 to USB 3.0! Pathetic.