aircrack-ng/rtl8812au #451: rtl8812au: packet injection not working in Kali Linux VirtualBox

ZaidOsman15, 2019-10-22:
I got the VM to connect my wireless adapter to it, put the adapter into monitor mode and ran “Airodump-ng wlan0” without any issues. But when i try to test packet injection using “Aireplay-ng -9 wlan0” it says “no answer” and found 0 APs. I should also mention that I’m a total beginner to Kali, so am I doing something wrong or?

yrjo, 2019-10-29:
i have seen this before, switch to something else than virtualbox helped (try vmware player instead), as there has been some issue with usb passthrough between host and VM

kimocoder, 2019-10-30:
We know. This issue is also in our README.md

kimocoder, 2019-11-11:
There's nothing we can do here right now, so the issue is closing. It's a known one.

closed 2019-11-11 by kimocoder