aircrack-ng/rtl8812au #452: [ASUS USB-AC68 / RTL8814AU] 3 Mbps connection speed

slavafomin, 2019-10-22:
Hello!

First of all I want to thank you for your effort and your hard work on this driver.

However, I've tried to use it for my newly bought [ASUS USB-AC68](https://wikidevi.com/wiki/ASUS_USB-AC68) and I've got a terrible connection speed. The network, where I was testing it supported around 30 Mbps (on my built-in wireless adapter), but this adapter gave me only 2-3 Mbps.

I was hoping that you will be able to help me in resolving this issue.

```
$ cat /etc/issue
Ubuntu 18.04.3 LTS
```

```
$ uname -a
Linux interpid 5.0.0-32-generic #34~18.04.2-Ubuntu SMP Thu Oct 10 10:36:02 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

```
$ lsusb

Bus 003 Device 005: ID 0b05:1853 ASUSTek Computer, Inc.
```

When connecting the adapter via USB with driver already installed:

```
$ dmesg

[ +10,199574] usb 3-2: new high-speed USB device number 6 using xhci_hcd
[  +0,148407] usb 3-2: New USB device found, idVendor=0b05, idProduct=1853, bcdDevice= 0.00
[  +0,000004] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0,000003] usb 3-2: Product: 802.11ac NIC
[  +0,000002] usb 3-2: Manufacturer: Realtek
[  +0,000002] usb 3-2: SerialNumber: 123456
[  +0,036396] PKCS#7 signature not signed with a trusted key
[  +0,128377] usbcore: registered new interface driver rtl88xxau
[  +0,015077] rtl88xxau 3-2:1.0 wlx04d9f513015c: renamed from wlan0
[  +3,921847] ------------[ cut here ]------------
[  +0,000051] WARNING: CPU: 5 PID: 8949 at /build/linux-hwe-iAAoxd/linux-hwe-5.0.0/net/wireless/nl80211.c:15676 cfg80211_ch_switch_notify+0x113/0x120 [cfg80211]
[  +0,000001] Modules linked in: 88XXau(OE) xt_conntrack ipt_MASQUERADE nf_conntrack_netlink nfnetlink xfrm_user xfrm_algo xt_addrtype iptable_filter iptable_nat nf_nat_ipv4 nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c bpfilter br_netfilter bridge stp llc ccm aufs overlay nls_iso8859_1 snd_hda_codec_hdmi intel_rapl snd_hda_codec_realtek snd_hda_codec_generic x86_pkg_temp_thermal arc4 ledtrig_audio intel_powerclamp snd_hda_intel snd_hda_codec coretemp snd_hda_core iwldvm snd_hwdep uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 snd_pcm mac80211 kvm_intel videobuf2_common snd_seq_midi snd_seq_midi_event videodev intel_cstate snd_rawmidi joydev intel_rapl_perf asus_nb_wmi snd_seq media input_leds iwlwifi serio_raw asus_wmi snd_seq_device mxm_wmi sparse_keymap snd_timer wmi_bmof mei_me snd lpc_ich cfg80211 mei soundcore nvidia_uvm(POE) mac_hid asus_wireless sch_fq_codel parport_pc ppdev lp parport ip_tables x_tables autofs4 algif_skcipher af_alg dm_crypt
[  +0,000019]  hid_logitech_hidpp hid_logitech_dj hid_generic usbhid hid i915 nvidia_drm(POE) nvidia_modeset(POE) crct10dif_pclmul kvmgt vfio_mdev crc32_pclmul mdev vfio_iommu_type1 ghash_clmulni_intel nvidia(POE) vfio kvm aesni_intel irqbypass i2c_algo_bit drm_kms_helper aes_x86_64 syscopyarea sysfillrect crypto_simd sysimgblt cryptd fb_sys_fops glue_helper drm psmouse ahci alx ipmi_devintf libahci mdio ipmi_msghandler wmi video
[  +0,000013] CPU: 5 PID: 8949 Comm: RTW_CMD_THREAD Tainted: P           OE     5.0.0-32-generic #34~18.04.2-Ubuntu
[  +0,000001] Hardware name: ASUSTeK COMPUTER INC. N76VB/N76VB, BIOS N76VB.205 05/24/2013
[  +0,000013] RIP: 0010:cfg80211_ch_switch_notify+0x113/0x120 [cfg80211]
[  +0,000001] Code: e4 e9 49 ff ff ff 49 8b 84 24 f0 00 00 00 48 85 c0 74 0b 48 8b 13 48 89 50 68 eb 81 0f 0b 48 c7 c7 80 59 5c c1 e8 60 3c 78 d0 <0f> 0b e9 6c ff ff ff 66 0f 1f 44 00 00 0f 1f 44 00 00 55 48 89 e5
[  +0,000001] RSP: 0018:ffffbc78c3187db8 EFLAGS: 00010286
[  +0,000001] RAX: 0000000000000024 RBX: ffffbc78c3187df8 RCX: 0000000000000000
[  +0,000000] RDX: 0000000000000000 RSI: ffff96ce9ed56448 RDI: ffff96ce9ed56448
[  +0,000001] RBP: ffffbc78c3187de0 R08: 00000000000004a7 R09: 0000000000000004
[  +0,000000] R10: ffffbc78c3187e00 R11: 0000000000000001 R12: ffff96ce1b49f400
[  +0,000001] R13: 000000000000099e R14: ffff96cdfb246000 R15: ffff96ce99ecc000
[  +0,000001] FS:  0000000000000000(0000) GS:ffff96ce9ed40000(0000) knlGS:0000000000000000
[  +0,000001] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  +0,000000] CR2: 000055723ebb14c0 CR3: 00000003cec0e002 CR4: 00000000001606e0
[  +0,000001] Call Trace:
[  +0,000040]  rtw_cfg80211_ch_switch_notify+0xd4/0x123 [88XXau]
[  +0,000025]  rtw_chk_start_clnt_join+0x5c/0x90 [88XXau]
[  +0,000023]  join_cmd_hdl+0x26d/0x36e [88XXau]
[  +0,000014]  rtw_cmd_thread+0x191/0x3c1 [88XXau]
[  +0,000022]  ? rtw_chk_start_clnt_join+0x90/0x90 [88XXau]
[  +0,000003]  kthread+0x121/0x140
[  +0,000015]  ? rtw_stop_cmd_thread+0x3f/0x3f [88XXau]
[  +0,000002]  ? kthread_park+0xb0/0xb0
[  +0,000002]  ret_from_fork+0x35/0x40
[  +0,000001] ---[ end trace 2a995d9636bce2e9 ]---
[  +0,209707] IPv6: ADDRCONF(NETDEV_CHANGE): wlx04d9f513015c: link becomes ready
```

```
[... 200 lines cut ...]
parm:           rtw_rxgain_offset_5gl:default RF Gain 5GL Offset value:0 (uint)
parm:           rtw_rxgain_offset_5gh:uint
parm:           rtw_rxgain_offset_5gm:default RF Gain 5GM Offset value:0 (uint)
parm:           rtw_pll_ref_clk_sel:force pll_ref_clk_sel, 0xF:use autoload value (uint)
parm:           rtw_tx_pwr_by_rate:0:Disable, 1:Enable, 2: Depend on efuse (int)
parm:           rtw_tx_pwr_lmt_enable:0:Disable, 1:Enable, 2: Depend on efuse (int)
parm:           rtw_target_tx_pwr_2g_a:2.4G target tx power (unit:dBm) of RF path A for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_b:2.4G target tx power (unit:dBm) of RF path B for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_c:2.4G target tx power (unit:dBm) of RF path C for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_d:2.4G target tx power (unit:dBm) of RF path D for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_5g_a:5G target tx power (unit:dBm) of RF path A for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_5g_b:5G target tx power (unit:dBm) of RF path B for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_5g_c:5G target tx power (unit:dBm) of RF path C for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_5g_d:5G target tx power (unit:dBm) of RF path D for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_tsf_update_pause_factor:num of bcn intervals to stay TSF update pause status (int)
parm:           rtw_tsf_update_restore_factor:num of bcn intervals to stay TSF update restore status (int)
parm:           rtw_en_napi:int
parm:           rtw_en_gro:int
parm:           rtw_iqk_fw_offload:int
parm:           rtw_ch_switch_offload:int
```

I've tried to use `v5.1.5` branch as was suggested in some issues, but it didn't help.

I would be glad to provide more relevant information. Any help will be highly appreciated!

kimocoder, 2019-10-24:
@slavafomin 

v5.1.5 and v5.6.4.1 is basicly the same on the 8814au. the HAL is from v5.1.5 on that chipset, old and buggy. however, a new PR was just merged to fix the speed issues..

source https://github.com/aircrack-ng/rtl8812au/pull/454

could you do a git pull / build / install / restart and test again + report back?

slavafomin, 2019-10-25:
@kimocoder oh, that's great to hear. I will try to do this on first occasion and will report back, thanks!

cedricbambooza, 2020-12-31:
there's also a new package here https://github.com/aircrack-ng/rtl8814au

slavafomin, 2021-05-11:
Thanks! I'm not using the adapter at the moment, so I won't be able to provide any more details on this. I think we can close the issue now.

closed 2021-05-11 by slavafomin