aircrack-ng/rtl8812au #458: v5.6.4.1 monitor mode not transmitting

webbbn, 2019-10-30:
I'm trying to switch from version 5.2.20 to 5.6.4.1 with the hopes of fixing some issues with txpower and latency that I'm seeing in 5.2.20. Unfortunately I can't get the driver to transmit any data with the newer driver.

I normally use iw and friends (very similar to the process described in the README) to configure the interface, and that works great with the 5.2.20 version. I've tried to use the same on the newer version, and it receives great, but does not appear to be transmitting anything. I don't see any power on the antenna line. I've also tried configuring the interface with airmon-ng and that doesn't seem to have any effect.

The code that I'm using is here:

https://github.com/webbbn/wifibroadcast_bridge/blob/master/src/raw_socket.cc

I've tried sending using both legacy modes and MCS modes in case there might be something wrong in one of my radiotap headers, but neither work.

The script that I normally use to configure the interface is here:

https://github.com/webbbn/wifibroadcast_bridge/blob/master/python/wifi_config.py

Any suggestions that I could try would be appreciated. Is there a relatively simple test program that could be used as an example for sending using this driver in monitor mode? If that exists and works, that could help me track down the issue.

kimocoder, 2019-10-30:
I will look into it

webbbn, 2019-11-05:
Any ideas on this? Is anyone having any success with this version of the driver in monitor mode? If so, could you please describe how you installed it, and tested it?

I've installed it on half a dozen systems now, and it doesn't work on any of them, so it seems like I must be missing something. I tried configuring with airmon-ng and testing with "aireplay-ng -9", and it completely fails.

kimocoder, 2019-11-05:
Yeah, it works for me on latest Kali.
All i have to do after loading the module is "ifconfig wlan1 up". Uploaded a video in another issue report.

webbbn, 2019-11-05:
You are correct. It does work on Kali.

So I wonder why it's not working for me on anything else.

In any case, that gives me something to work with.

kimocoder, 2019-11-09:
Move over to v5.6.4.2 instead of v5.6.4.1

closed 2019-11-09 by kimocoder