aircrack-ng/rtl8812au #460: Building against Clang?

etna, 2019-11-10:
Now that Clang is capable of building the Linux kernel, will it eventually be possible to build this driver with Clang if the kernel itself was already built against Clang?

Currently attempting to build this driver against Clang returns the following errors
[clangErrors.txt](https://github.com/aircrack-ng/rtl8812au/files/3828499/clangErrors.txt)

Thanks for the consideration.

kimocoder, 2019-11-11:
@etna what command did you run, to be specific?

etna, 2019-11-12:
> @etna what command did you run prior to that log, to be specific?

Hello there.

Initially tried the standard `CC=<path to clang> CXX=<path to clang++> make` but the Makefile would not accept those values; build still looks for GCC and uses it.

So I tried putting `CC=<path to clang> CXX=<path to clang++>` inside the Makefile to force the build to use Clang, which resulted in the errors you saw in the log.