aircrack-ng/rtl8812au #461: compilation error on 5.6.4.2

logan001, 2019-11-10:
Title
Arch / Kernel 5.3.9-20-tkg-pds / gcc (GCC) 9.2.0 / driver commin a1788aa1d7287f7408391981e6f9974cb5918194
```

DKMS make.log for rtl8812au-5.6.4.2 for kernel 5.3.9-20-tkg-pds (x86_64)
Sun 10 Nov 2019 07:02:39 PM EET
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.3.9-20-tkg-pds/build M=/var/lib/dkms/rtl8812au/5.6.4.2/build  modules
make[1]: Entering directory '/usr/lib/modules/5.3.9-20-tkg-pds/build'
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_recv.o
/var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_security.c: In function ‘getcrc32’:
/var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_security.c:224:1: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
  224 | void rtw_wep_encrypt(_adapter *padapter, u8 *pxmitframe)
      | ^~~~
/var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_security.c:344:12: error: invalid storage class for function ‘secmicgetuint32’
  344 | static u32 secmicgetuint32(u8 *p)
      |            ^~~~~~~~~~~~~~~
/var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_security.c:354:13: error: invalid storage class for function ‘secmicputuint32’
  354 | static void secmicputuint32(u8 *p, u32 val)
      |             ^~~~~~~~~~~~~~~
/var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_security.c:364:13: error: invalid storage class for function ‘secmicclear’
  364 | static void secmicclear(struct mic_data *pmicdata)
      |             ^~~~~~~~~~~
/var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_security.c:579:13: error: invalid storage class for function ‘phase1’
  579 | static void phase1(u16 *p1k, const u8 *tk, const u8 *ta, u32 iv32)
      |             ^~~~~~
/var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_security.c:626:13: error: invalid storage class for function ‘phase2’
  626 | static void phase2(u8 *rc4key, const u8 *tk, const u16 *p1k, u16 iv16)
      |             ^~~~~~
/var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_security.c:947:13: error: invalid storage class for function ‘bitwise_xor’
  947 | static void bitwise_xor(u8 *ina, u8 *inb, u8 *out);
      |             ^~~~~~~~~~~
/var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_security.c:948:13: error: invalid storage class for function ‘construct_mic_iv’
  948 | static void construct_mic_iv(
      |             ^~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_security.c:956:13: error: invalid storage class for function ‘construct_mic_header1’
  956 | static void construct_mic_header1(
      |             ^~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_security.c:961:13: error: invalid storage class for function ‘construct_mic_header2’
  961 | static void construct_mic_header2(
      |             ^~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_security.c:966:13: error: invalid storage class for function ‘construct_ctr_preload’
  966 | static void construct_ctr_preload(
      |             ^~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_security.c:974:13: error: invalid storage class for function ‘xor_128’
  974 | static void xor_128(u8 *a, u8 *b, u8 *out);
      |             ^~~~~~~
/var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_security.c:975:13: error: invalid storage class for function ‘xor_32’
  975 | static void xor_32(u8 *a, u8 *b, u8 *out);
      |             ^~~~~~
/var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_security.c:976:11: error: invalid storage class for function ‘sbox’
  976 | static u8 sbox(u8 a);
      |           ^~~~
/var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_security.c:977:13: error: invalid storage class for function ‘next_key’
  977 | static void next_key(u8 *key, sint round);
      |             ^~~~~~~~
[... 76 lines cut ...]
 2777 | static int omac1_aes_128_vector(const u8 *key, size_t num_elem,
      |            ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_security.c:3338:1: error: expected declaration or statement at end of input
 3338 | }
      | ^
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_security.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_mlme_ext.c: In function ‘rtw_txpwr_init_regd’:
/var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_mlme_ext.c:260:6: warning: this statement may fall through [-Wimplicit-fallthrough=]
  260 |   if (rfctl->regd_name)
      |      ^
/var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_mlme_ext.c:262:2: note: here
  262 |  default:
      |  ^~~~~~~
/var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_mlme_ext.c:1345:17: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1345 |    ptable->func = &OnAuthClient;
      |    ~~~~~~~~~~~~~^~~~~~~~~~~~~~~
/var/lib/dkms/rtl8812au/5.6.4.2/build/core/rtw_mlme_ext.c:1347:2: note: here
 1347 |  case WIFI_ASSOCREQ:
      |  ^~~~
make[1]: *** [Makefile:1630: _module_/var/lib/dkms/rtl8812au/5.6.4.2/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.3.9-20-tkg-pds/build'
make: *** [Makefile:2255: modules] Error 2
```

Nairod785, 2019-11-10:
Same issue here... Awus1900 stuck in a non working state

kimocoder, 2019-11-10:
I know, will fix it in a few hours

kimocoder, 2019-11-11:
@logan001 @Nairod785 

Problem is resolved. Also added some more patches :1st_place_medal: 
Tested compiling successfully on kernel v5.2 @ Kali linux

kimocoder, 2019-11-11:
[image]

kimocoder, 2019-11-11:
iwlist scanning works..

[image]

kimocoder, 2019-11-11:
Issue resolved. And improved a lot!

closed 2019-11-11 by kimocoder