aircrack-ng/rtl8812au #462: Compilation Issues for Raspberry Pi 3B

TotallyAProgrammer, 2019-11-11:
Every time I (attempt to) install the driver I get the following error from make/gcc:
`gcc: error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together`

I have attempted to find and solve the issue myself, but to no avail.

kimocoder, 2019-11-11:
Yes, I'm familier with that one, but I don't have more time right away :1234:

kimocoder, 2019-11-11:
The function is below, try to manually "nano Makefile" and goto line 1211 and switch "hard" with "soft" and then compile/build again. Please report back!

https://github.com/aircrack-ng/rtl8812au/blob/v5.6.4.2/Makefile#L1211

kimocoder, 2019-11-11:
@TotallyAProgrammer 

It should be fixed now, try "git pull" (update) and build again.
Please report back here

TotallyAProgrammer, 2019-11-11:
After the latest commit ( #ae31db1 ), it compiled and runs great. The chip I'n using is the rtl8811au on raspbian buster. I'm glad this got fixed and works. Thank you

closed 2019-11-11 by TotallyAProgrammer

kimocoder, 2019-11-11:
Awesome sir! Thanks for the info, happy it worked out :1st_place_medal:

solsticedhiver, 2019-11-24:
I still have the problem on a **rpi1** (using also *distcc* but I don't think that's the issue here):

```
DKMS make.log for rtl88xxau-r970.3b0b7b9 for kernel 4.19.85-1-ARCH (armv6l)
Sun Nov 24 20:12:08 CET 2019
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.19.85-1-ARCH/build M=/var/lib/dkms/rtl88xxau/r970.3b0b7b9/build  modules
make[1]: Entering directory '/usr/lib/modules/4.19.85-1-ARCH/build'
  CC [M]  /var/lib/dkms/rtl88xxau/r970.3b0b7b9/build/core/rtw_cmd.o
armv6l-unknown-linux-gnueabihf-gcc: error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together
distcc[11172] ERROR: compile /var/lib/dkms/rtl88xxau/r970.3b0b7b9/build/core/rtw_cmd.c on 192.168.44.115:3634 failed
distcc[11172] (dcc_build_somewhere) Warning: remote compilation of '/var/lib/dkms/rtl88xxau/r970.3b0b7b9/build/core/rtw_cmd.c' failed, retrying locally
distcc[11172] Warning: failed to distribute /var/lib/dkms/rtl88xxau/r970.3b0b7b9/build/core/rtw_cmd.c to 192.168.44.115:3634, running locally instead
armv6l-unknown-linux-gnueabihf-gcc: error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together
distcc[11172] ERROR: compile /var/lib/dkms/rtl88xxau/r970.3b0b7b9/build/core/rtw_cmd.c on localhost failed
make[2]: *** [scripts/Makefile.build:304: /var/lib/dkms/rtl88xxau/r970.3b0b7b9/build/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1525: _module_/var/lib/dkms/rtl88xxau/r970.3b0b7b9/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.19.85-1-ARCH/build'
make: *** [Makefile:2245: modules] Error 2
```

the fix from 23368c9 seems to be gone from the `Makefile`