aircrack-ng/rtl8812au #463: Kernel config option inconsistent between Kconfig and Makefile

Kawashima-Azumi, 2019-11-11:
Hi,

Since a few days ago, the kernel config for this driver has been changed to 88XXAU in Kconfig, but remains RTL8812AU in Makefile.
Is it possible to change both to RTL88XXAU for consistency?

Best regards,

kimocoder, 2019-11-11:
And you suppose to change it to?

Is this better then?
`EXTRA_CFLAGS += -DDRV_NAME=\"88XXau\"
ifeq ($(CONFIG_USB_HCI), y)
USER_MODULE_NAME = 88XXau
endif
else
EXTRA_CFLAGS += -DDRV_NAME=\"rtl8812au\"
endif`

kimocoder, 2019-11-11:
Changing those settings would make things go bad, I've looked at it, but i only changed from 88XXAU to 88XXau which is the module name. These params for DRV_NAME is also set in "/include/autoconf.h"

closed 2019-11-11 by kimocoder

Kawashima-Azumi, 2019-11-12:
> And you suppose to change it to?
> 
> Is this better then?
> 
> ```
> EXTRA_CFLAGS += -DDRV_NAME=\"88XXau\"
> ifeq ($(CONFIG_USB_HCI), y)
> USER_MODULE_NAME = 88XXau
> endif
> else
> EXTRA_CFLAGS += -DDRV_NAME=\"rtl8812au\"
> endif
> ```

Well, anything consistent between Kconfig and Makefile would be ok.