aircrack-ng/rtl8812au #464: Makefile missing RPI option in V5.6.4.2

botto, 2019-11-11:
Hi,

I tried 5.6.4.2 but noticed the Makefile is missing options for ARM64_RPI and ARM_RPI.  
Example: https://github.com/aircrack-ng/rtl8812au/blob/v5.6.4.1/Makefile#L150  
I tried adding it manually but the compilation fails.

Thanks

kimocoder, 2019-11-11:
I'll add it shortly, give me the errors you've got

kimocoder, 2019-11-11:
@botto 

Added RPI & RPI64 platform support
https://github.com/aircrack-ng/rtl8812au/commit/b6c538d9080bde982a1a8c44ecca57381e56b406

Also tried to fix the -mhard-float issue related to the new VHT IO handler.
https://github.com/aircrack-ng/rtl8812au/commit/23368c91c21ac82dbe2ae670fe529c640d805950

Please report any issues back here, if any :+1:

botto, 2019-11-11:
Excellent, I can see them now:
So when using `CONFIG_PLATFORM_ARM64_RPI = y` on a RPI 4 I get
```
rtl8812au_aircrack git:(v5.6.4.2) ✗ make -j 4
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/4.19.75-v7l+/build M=/home/pi/rtl8812au_aircrack  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.75-v7l+'
  CC [M]  /home/pi/rtl8812au_aircrack/core/rtw_cmd.o
  CC [M]  /home/pi/rtl8812au_aircrack/core/rtw_security.o
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
make[2]: *** [scripts/Makefile.build:303: /home/pi/rtl8812au_aircrack/core/rtw_cmd.o] Error 1
make[2]: *** Waiting for unfinished jobs....
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
make[2]: *** [scripts/Makefile.build:303: /home/pi/rtl8812au_aircrack/core/rtw_security.o] Error 1
make[1]: *** [Makefile:1519: _module_/home/pi/rtl8812au_aircrack] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.75-v7l+'
make: *** [Makefile:2197: modules] Error 2
```

However using  `CONFIG_PLATFORM_ARM_RPI = y` works fine on the RPI 4

Uname:  
`Linux pi-4 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux`

solsticedhiver, 2019-11-25:
I can't compile on rpi1. I reported the problem in issue #462 with https://github.com/aircrack-ng/rtl8812au/issues/462#issuecomment-557919640

kimocoder, 2019-11-26:
This issue should be resolved

closed 2019-11-26 by kimocoder

solsticedhiver, 2019-11-27:
@kimocoder it is not