aircrack-ng/rtl8812au #465: Module not built when integrated into kernel source tree

Kawashima-Azumi, 2019-11-12:
> And you suppose to change it to?
> 
> Is this better then?
> 
> ```
> EXTRA_CFLAGS += -DDRV_NAME=\"88XXau\"
> ifeq ($(CONFIG_USB_HCI), y)
> USER_MODULE_NAME = 88XXau
> endif
> else
> EXTRA_CFLAGS += -DDRV_NAME=\"rtl8812au\"
> endif
> ```

Well, anything consistent between Kconfig and Makefile would be ok.

Otherwise, the module doesn't get built when integrated into the kernel source tree.

_Originally posted by @Kawashima-Azumi in https://github.com/aircrack-ng/rtl8812au/issues/463#issuecomment-552682473_

kimocoder, 2019-11-13:
@Kawashima-Azumi I did a mistake the other day, try updating and build again :+1: 
And please report back if that fixes it

Kawashima-Azumi, 2019-11-14:
Well, I meant that the new kernel config option in Kconfig should also be used in the Makefile:

--- Makefile.orig       2019-11-14 13:45:58.900285978 +0800
+++ Makefile    2019-11-14 16:09:58.783016668 +0800
@@ -2207,11 +2207,11 @@
 $(MODULE_NAME)-$(CONFIG_MP_INCLUDED)+= core/rtw_bt_mp.o
 endif
 
-obj-$(CONFIG_RTL8812AU) := $(MODULE_NAME).o
+obj-$(CONFIG_88XXAU) := $(MODULE_NAME).o
 
 else
 
-export CONFIG_RTL8812AU = m
+export CONFIG_88XXAU = m
 
 all: modules

kimocoder, 2019-11-14:
Done. Pushing shortly, thanks!

closed 2019-11-14 by kimocoder