aircrack-ng/rtl8812au #466: RTL8811AU: ERROR rtw_hal_init: fail

Kawashima-Azumi, 2019-11-12:
lsusb shows:
Bus 001 Device 004: ID 0bda:a811 Realtek Semiconductor Corp. 

# ifconfig wlan1 up
SIOCSIFFLAGS: Operation not permitted

dmesg shows:
RTW: ERROR rtw_hal_init: fail

kimocoder, 2019-11-14:
@Kawashima-Azumi I just pushed a firmware fix for 8821au (8811). Download, compile, install & reboot and please report back. I don't have this chipset, but I think it should be working now

closed 2019-11-14 by kimocoder

Kawashima-Azumi, 2019-11-27:
> @Kawashima-Azumi I just pushed a firmware fix for 8821au (8811). Download, compile, install & reboot and please report back. I don't have this chipset, but I think it should be working now

Thank you for the fix! But bringing up (ifconfig ... up) 8811 still fails.

lsusb.py shows
 1-4  0bda:a811 00  2.10  480MBit/s 500mA 1IF  (Realtek  802.11ac WLAN Adapter  00e04c000001)

I have only 8811, 8812, and 8814, so didn't check if 8821 works.
Anything that I can do to help with debugging?

kimocoder, 2019-11-27:
8811 is 8821

commit b5f97c16 (2019-11-14): Fixed 8821au (8811au) support, fixed fw bug. fixes #466